aws/amazon-freertos #2580: OTA updates fail intermittently due to TLS frames being larger than 8192 bytes

shreyasbharath, 2020-10-15:
We've been been investigating an issue where a lot of our devices are failing to update firmware (they show `queued` state forever on the AWS IoT job console).

We took a deep dive into the issue and ruled out:
- bugs in our custom firmware application code
- any environmental issue or networking issue within our internal control

We've narrowed it down to the TLS socket connection our device establishes with the AWS MQTT broker to download the OTA update. It appears the broker (server) sends us TLS frames/packets that are much bigger than our device's internal buffer (8192 bytes) during the block transfer process, which is configured by AWS for the mbedTLS library. We think something has changed on the AWS side for this to become an issue now as this has always been the same since day one. I also noticed that the limit is still 8192 in the current release of AWS FreeRTOS.

Note that this doesn't happen all the time (some OTA updates succeed), but it is still a very high failure rate.

**System information**
- ESP32
- v1.3.2-246-gf69f40d5
- MQTT|Shadow|OTA|

**Expected behavior**
The server should not be sending us TLS frames larger than the device's internal buffer.

We've looked into the MFL extension, which can be used to negotiate maximum incoming frame size with the server. But this doesn't seem to be turned on in `aws_mqtt_agent.c`.

Has something changed on the server side for this to become an issue in the past few weeks/days? What can we do to get around this?

lundinc2, 2020-10-16:
Hey @shreyasbharath,

Can you share a log with mbed TLS debugging turned on?

Thanks,

Carl

pvyawaha, 2020-10-16:
Hello Shreyasbharath,

Closing this issue as this is resolved with service side changes, please reopen for the same issue or create a new one.

closed 2020-10-16 by pvyawaha