aws/amazon-freertos #2586: why not include GAP service in aws_demos and aws_tests BLE sample code

wingwu2379, 2020-10-16:
Hi theres,
From Bluetooth spec, Vol 3, Part C, Section 12 "GAP SERVICE AND CHARACTERISTICS FOR GATT SERVER"
it said LE Peripheral/Central should include  "Device NAME" & "Appearance" of GAP service
but from AWS sample, AWS_DEMOS and AWS_TESTS both not include GAP SERVICE

ravibhagavandas, 2020-10-16:
Hi @wingwu2379 
The GAP service is included by default from the underlying BLE stack by the vendor when `IotBle_On()` function is invoked. Device name can be configured for peripheral either statically through the config `IOT_BLE_DEVICE_COMPLETE_LOCAL_NAME` defined in `vendors/<vendor>/boards/<board>/aws_demos/config_files/iot_ble_config.h` or at runtime using `IotBle_SetDeviceName()` API.

wingwu2379, 2020-10-16:
Hi @ravibhagavandas ,
Thanks for your quick reply
Could you please point out where to set GAP SERVICE UUID(0x1800) and Device name UUID(0x2A00) ?
As I check where to set GATT database,
IotBle_On() --> _startGATTServices() --> and it seems add 3 service, but I can't find GAP service
Please let me know if I misunderstood, Thanks

ravibhagavandas, 2020-10-17:
`_startGATTServices()` are used to create the  GATT services defined by Amazon FreeRTOS as well as any customer user GATT services through `IotBle_AddCustomServicesCb()` hook.
As I said before, the GAP service is a standard service and its created internally by the vendor ble stack who implements the porting layer. For example for one of the ports, ESP32 Nimble stack the device name is set by calling the API [ble_svc_gap_device_name_set](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/ports/ble/nimble/iot_ble_hal_common_gap.c#L827). So user applications need not create the GAP service as they are created and started automatically.

wingwu2379, 2020-10-19:
Hi @ravibhagavandas 

As you said, the device name can be set by eBTpropertyBdname of  prvBTSetDeviceProperty()
but how to set "Appearance" characteristics of GAP service, according SPEC, it also mandatory for LE Peripheral/Central

So in my view, the GAP service should add in application side or Amazon can provide how to set all of characteristics of GAP service (SPEC Table12.2)

please let me know, if you get another view, Thanks

ravibhagavandas, 2020-10-19:
Hi,
To set the appearance characteristic and to advertise the appearance value you can use the [ulAppearance field](https://github.com/aws/amazon-freertos/blob/master/libraries/abstractions/ble_hal/include/bt_hal_manager_adapter_ble.h#L122) included in `pxSetAdvData` API. 

If you are using BLE middleware as opposed to BLE HAL, you can set the value in config by defining following configuration in `vendors/<vendor>/boards/<board/aws_demos/config_files/iot_ble_config.h`:

```#define IOT_BLE_ADVERTISING_APPEARANCE    <32 bit Appearance Value>```

wingwu2379, 2020-10-20:
Hi @ravibhagavandas,

Thanks for information,
Could you also please guide me how to get below characterisitc?
1. Peripheral Preferred Connection Pararmeters
2. Central Address Resolution
3. Resolvable Private Address Only

Thanks

ravibhagavandas, 2020-10-23:
Hi @wingwu2379
Peripheral preffered connection parameters can be set via advertisement params as per the documentation [here](https://github.com/aws/amazon-freertos/blob/master/libraries/abstractions/ble_hal/include/bt_hal_manager_adapter_ble.h#L123)

Central Address Resolution characteristic and Resolvable Private Address Only are set by the stack based on what privacy mode is configured. These configs are not accessed by application.

ravibhagavandas, 2020-11-01:
Hi @wingwu2379 

I'm going closing this issue for now. Please feel free to reopen this if you have any further issues or questions.

closed 2020-11-01 by ravibhagavandas