aws/amazon-freertos #2606: BLE advertisement from device does not come out for infinite time, but it stops after some time, Even though usTimeout is set to 0, in _setAdvData() function.

kumarfirmware, 2020-10-20:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

Even though usTimeout is set to 0, in _setAdvData() function,  the BLE advertisement from device does not come out for infinite time, but it stops after some time.

I want to send the BLE advertisement continuously, advertisement should stop when central connects to the peripheral device and when central disconnects, peripheral again should send BLE advertisement continuously.

But this is not happening, can you please tell me what can be the reason.

Thanks.

pvyawaha, 2020-10-21:
Hello,

We are looking into this issue and any default configs that might need to be updated, @ravibhagavandas will help in this.

kumarfirmware, 2020-10-21:
Hi Team,

Thanks for the response.

I am using the following configuration, my assumption is with the following configuration, there should not be any pairing or bonding between central and peripheral. Please let me know if the following configuration is correct or not? Do I need to do anything to disable automatic pairing and bonding.

Central just need to connect and exchange the data and disconnect. Central need to connect when ever it wants, that means peripheral should be advertising all the time.

#define IOT_BLE_ENABLE_NUMERIC_COMPARISON        ( 1 )
#define IOT_BLE_ENABLE_SECURE_CONNECTION         ( 0 )
#define IOT_BLE_INPUT_OUTPUT                     ( eBTIONone )
#define IOT_BLE_ENCRYPTION_REQUIRED               ( 0 )
#define IOT_BLE_ENABLE_BONDING   ( 0)

Thanks.

ravibhagavandas, 2020-10-26:
Hello,

> I am using the following configuration, my assumption is with the following configuration, there should not be any pairing or bonding between central and peripheral. Please let me know if the [...]
#define IOT_BLE_ENABLE_NUMERIC_COMPARISON ( 1 )
#define IOT_BLE_ENABLE_SECURE_CONNECTION ( 0 )
#define IOT_BLE_INPUT_OUTPUT ( eBTIONone )
#define IOT_BLE_ENCRYPTION_REQUIRED ( 0 )
#define IOT_BLE_ENABLE_BONDING ( 0)

Yes the configuration disables pairing and bonding and will create an open connection. Since you disable pairing you can also set `IOT_BLE_ENABLE_NUMERIC_COMPARISON ` to `0`.

>But this is not happening, can you please tell me what can be the reason.
Two failure scenarios,
    On power up, it fails to advertise continuously, it stops after some time.
    After central connects and disconnects, then also it fails to advertise continuously.

This is because stack uses Resolvable Private Address resolution (RPA) for security. By default RPA has
 a timeout  of 15 minutes for the random address generated, as per Bluetooth specification. You can change the duration of RPA by changing the config value `CONFIG_NIMBLE_RPA_TIMEOUT` [here](CONFIG_NIMBLE_RPA_TIMEOUT). Alternatively you can choose to restart advertisement after timeout by registering a callback `IotBle_StartAdvCallback_t ` and passing it to the function [IotBle_StartAdv()](https://github.com/aws/amazon-freertos/blob/202007.00/libraries/c_sdk/standard/ble/include/iot_ble.h#L490). The callback will be invoked whenever advertisement status changes.

Please let us know if you have further questions.

Sunilpuli188, 2020-10-28:
Hi Ravi,

Thanks for the quick response.

I noted your suggestions.

1.How do I know advertisement is timed out in the callback IotBle_StartAdvCallback_t?  What condition should I check in the callback function?

2. Which file do I need to configure the value CONFIG_NIMBLE_RPA_TIMEOUT?

3. Also I do not want to use RPA, I want to use my standard BLE MAC address for advertisements. How do I need to do this? How do I change this? what are the settings required for this? I guess this should be the public address.

Kindly let me know these things as soon as possible.

Thanks.

kumarfirmware, 2020-10-28:
Hi Ravi,

Thanks for the quick response.

I noted your suggestions.

1.How do I know advertisement is timed out in the callback IotBle_StartAdvCallback_t? What condition should I check in the callback function?

Which file do I need to configure the value CONFIG_NIMBLE_RPA_TIMEOUT?

Also I do not want to use RPA, I want to use my standard BLE MAC address for advertisements. How do I need to do this? How do I change this? what are the settings required for this? I guess this should be the public address.

Kindly let me know these things as soon as possible.

Thanks.

ravibhagavandas, 2020-11-01:
Hi @kumarfirmware ,

> 1.How do I know advertisement is timed out in the callback IotBle_StartAdvCallback_t? What condition should I check in the callback function?

The GAP advertisement gets preempted whenever RPA timer expires, as per the BLE specification all GAP procedures should be stopped while updating the privacy list. So you will still get the `IotBle_StartAdvCallback_t` callback with status as `eBTStatusFail` as the advertisement gets preempted.

> Which file do I need to configure the value CONFIG_NIMBLE_RPA_TIMEOUT?

The value can be changed through the `sdkconfig` for ESP, in `<build folder>/sdkconfig`. Alternatively you can add the value to `./vendors/espressif/boards/esp32/aws_demos/sdkconfig.defaults` so that it is picked up every time you build.

> Also I do not want to use RPA, I want to use my standard BLE MAC address for advertisements. How do I need to do this? How do I change this? what are the settings required for this? I guess this should be the public address.

RPA is highly recommended for privacy as it keeps BLE device not traceable. Also this was recommended by ESP team, without RPA we had connectivity and pairing issues with RPA enabled devices like IOS. You can disable RPA by changing this [line](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/ports/ble/nimble/iot_ble_hal_common_gap.c#L532) to call `ble_hs_pvcy_rpa_config(0)`.

kumarfirmware, 2020-11-02:
Hi Ravi,

Thanks for the response.

1.I did not do any config change mentioned in the above. But I see now advertisement coming out more than 15 mins. I am not sure what is the default configuration as is? I do not think by default it is RPA.

2. How do I disable pairing and bonding and auto connection in the iOS app? what is the configuration or API?

3. Based on what iOS app finds out to connect to ESP? I understand ESP advertising some service. what is the service ESP is advertising for MQTT over BLE? How is the iOS app detecting it and establishing connection? Can you please explain this connection process in detail?

4. What is the difference between Nimble and Bluedroid? which is good to use?

Thanks.

ravibhagavandas, 2020-11-07:
@kumarfirmware 

1. If you set the advertising timeout param to zero, ESP32 port uses the default configuration which is advertising for ever as shown [here](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/ports/ble/nimble/iot_ble_hal_gap.c#L722). It set to default value 0 [here](https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/standard/ble/src/iot_ble_gap.c#L423). Only exception I have seen from my testing is when using RPA, based on RPA timeout the advertising process gets preempted. Could you measure the time from start of advertisement to timeout and provide the values? That will help us identify what is causing the timeout.

2. IOS app is in central role, so it honors the configuration on device side. To disable paring and bonding on device side you need to set following configs as mentioned above:
```
#define IOT_BLE_ENABLE_NUMERIC_COMPARISON ( 1 )
#define IOT_BLE_ENABLE_SECURE_CONNECTION ( 0 )
#define IOT_BLE_INPUT_OUTPUT ( eBTIONone )
#define IOT_BLE_ENCRYPTION_REQUIRED ( 0 )
#define IOT_BLE_ENABLE_BONDING ( 0)
```

3. The UUID used for advertisement is configurable, by using the config `IOT_BLE_ADVERTISING_UUID` ad `IOT_BLE_ADVERTISING_UUID_SIZE`. By default they are set to 128 bit UUID of device information service as [here](https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/standard/ble/include/iot_ble_config_defaults.h#L75). IOS app uses this uuid to discover and connect to a FreeRTOS device. You can choose to change this config at both device and app side.

4. Nimble is newer BLE stack from ESP32 over the bluedroid stack. Feature wise both supports our required features and have passed our qualification tests.  From our testing, switching to Nimble in AFR has reduced the RAM consumption in BLE.   So we have set default stack as Nimble.

ravibhagavandas, 2020-11-18:
@kumarfirmware 

I am closing this issue, if you don't have further queries. Please re-open this if you have any more questions.

closed 2020-11-18 by ravibhagavandas