aws/amazon-freertos #2621: Attempting to run Shadow Demo and OTA Demo Simultaneously

AlnurElberier, 2020-10-23:
In an attempt to run the shadow demo and the OTA demo simultaneously I have created another detached thread in iot_demo_runner.c as seen below:
 
void DEMO_RUNNER_RunDemos( void )
{
    /* These demos are shared with the C SDK and perform their own initialization and cleanup. */

    static demoContext_t mqttDemoContext2 =
    {
        .networkTypes                = democonfigNETWORK_TYPES,
        .demoFunction                = DEMO_entryFUNCTION2,
        .networkConnectedCallback    = DEMO_networkConnectedCallback,
        .networkDisconnectedCallback = DEMO_networkDisconnectedCallback
    };

    Iot_CreateDetachedThread( runDemoTask,
                              &mqttDemoContext2,
                              democonfigDEMO_PRIORITY,
                              democonfigDEMO_STACKSIZE );
    
    
    static demoContext_t mqttDemoContext =
    {
        .networkTypes                = democonfigNETWORK_TYPES,
        .demoFunction                = DEMO_entryFUNCTION,
        .networkConnectedCallback    = DEMO_networkConnectedCallback,
        .networkDisconnectedCallback = DEMO_networkDisconnectedCallback
    };

    Iot_CreateDetachedThread( runDemoTask,
                              &mqttDemoContext,
                              democonfigDEMO_PRIORITY,
                              democonfigDEMO_STACKSIZE );
}

In aws_demo_config.h I have created my own demo define as follows: 

#if (defined( CONFIG_COMBINED_SHADOW_OTA_ENABLED ))
    #define CONFIG_SHADOW_DEMO_ENABLED
    #define CONFIG_OTA_UPDATE_DEMO_ENABLED
#endif

And in iot_demo_runner .h I added this as well to define my second demo entry function

#if defined( CONFIG_COMBINED_SHADOW_OTA_ENABLED )
    #define DEMO_entryFUNCTION              RunShadowDemo
    #if defined( democonfigSHADOW_DEMO_TASK_STACK_SIZE )
        #undef democonfigDEMO_STACKSIZE
        #define democonfigDEMO_STACKSIZE    democonfigSHADOW_DEMO_TASK_STACK_SIZE
    #endif
    #if defined( democonfigSHADOW_DEMO_TASK_PRIORITY )
        #undef democonfigDEMO_PRIORITY
        #define democonfigDEMO_PRIORITY     democonfigSHADOW_DEMO_TASK_PRIORITY
    #endif
    #define DEMO_entryFUNCTION2              vStartOTAUpdateDemoTask
#elif defined( CONFIG_MQTT_DEMO_ENABLED )

If I leave the priorities as show in the code above then the Shadow Demo is the only thing that runs (Terminal Output):  

6 20678 [iot_thread] [INFO ][DEMO][20678] ---------STARTING DEMO---------

7 20685 [iot_thread] [INFO ][DEMO][20685] ---------STARTING DEMO---------

8 20692 [iot_thread] [INFO ][INIT][20692] SDK successfully initialized.
...9 27115 [iot_thread] [INFO ][DEMO][27114] Successfully initialized the demo. Network type for the demo: 1
10 27124 [iot_thread] Shadow Demo
11 27127 [iot_thread] [INFO ][MQTT][27127] MQTT library successfully initialized.
12 27135 [iot_thread] [INFO ][Shadow][27135] Shadow library successfully initialized.
13 27143 [iot_thread] [INFO ][DEMO][27142] Shadow Thing Name is C201300004 (length 10).
.14 31025 [iot_thread] [INFO ][MQTT][31025] Establishing new MQTT connection.
15 31033 [iot_thread] [INFO ][MQTT][31033] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
[... 91 lines cut ...]
13 38689 [iot_thread] [INFO ][Shadow][38689] Shadow library successfully initialized.
14 38697 [iot_thread] [INFO ][DEMO][38697] Shadow Thing Name is C201300004 (length 10).
......15 48884 [iot_thread] ERROR: Handshake failed with error code -29312
16 48891 [iot_thread] [ERROR][NET][48891] Failed to establish new connection. Socket status: -1002.
17 48931 [iot_thread] [INFO ][DEMO][48931] Successfully initialized the demo. Network type for the demo: 1
18 48940 [iot_thread] OTA Demo
19 48943 [iot_thread] [ERROR][MQTT][48943] Failed to establish new MQTT connection, error NETWORK ERROR.
20 48953 [iot_thread] [ERROR][DEMO][48953] MQTT CONNECT returned error NETWORK ERROR.
21 48961 [iot_thread] [INFO ][Shadow][48961] Shadow library cleanup done.
22 48967 [iot_thread] [INFO ][MQTT][48967] MQTT library cleanup done.
23 48974 [iot_thread] [INFO ][DEMO][48974] memory_metrics::freertos_heap::before::bytes::193000
24 48983 [iot_thread] [INFO ][DEMO][48982] memory_metrics::freertos_heap::after::bytes::157176
25 48991 [iot_thread] [INFO ][DEMO][48991] memory_metrics::demo_task_stack::before::bytes::0
26 49000 [iot_thread] [INFO ][DEMO][49000] memory_metrics::demo_task_stack::after::bytes::0
27 49008 [iot_thread] [ERROR][DEMO][49008] Error running demo.
28 49020 [iot_thread] [INFO ][INIT][49020] SDK cleanup done.
29 49026 [iot_thread] [INFO ][DEMO][49026] -------DEMO FINISHED-------

30 49114 [iot_thread] [INFO ][MQTT][49114] MQTT library successfully initialized.
31 49121 [iot_thread] OTA demo version 0.9.2
32 49125 [iot_thread] Creating MQTT Client...
33 49129 [iot_thread] Waiting for a network connection.
.........................................

Please let me know if you have any incite as for how I should continue.

AlnurElberier, 2020-10-23:
So there seems to be a little inconsistency with the output in our trials. Without modifying the firmware There are different outcomes seeming to happen randomly. In some cases we are able to get a successful execution of both demos, and in others it fails as shown above. 

Successful terminal output:  

6 5602 [iot_thread] [INFO ][DEMO][5602] ---------STARTING DEMO---------

7 5609 [iot_thread] [INFO ][INIT][5609] SDK successfully initialized.
8 6127 [iot_thread] [INFO ][DEMO][6127] ---------STARTING DEMO---------

9 6134 [iot_thread] [INFO ][INIT][6134] SDK successfully initialized.
.....................10 46871 [iot_thread] [INFO ][DEMO][46870] Successfully ini
11 46880 [iot_thread] Shadow Demo
12 46883 [iot_thread] [INFO ][MQTT][46883] MQTT library successfully initialized
13 46891 [iot_thread] [INFO ][Shadow][46891] Shadow library successfully initial
14 46899 [iot_thread] [INFO ][DEMO][46899] Shadow Thing Name is C201300004 (leng
..15 52167 [iot_thread] [INFO ][DEMO][52167] Successfully initialized the demo.
16 52176 [iot_thread] OTA Demo
17 52279 [iot_thread] [INFO ][MQTT][52279] MQTT library successfully initialized
18 52286 [iot_thread] OTA demo version 0.9.2
19 52290 [iot_thread] Creating MQTT Client...
.20 56169 [iot_thread] Connecting to broker...
21 56173 [iot_thread] [INFO ][MQTT][56173] Establishing new MQTT connection.
22 56181 [iot_thread] [INFO ][MQTT][56180] Anonymous metrics (SDK language, SDK                                                                                                                                                              ICS set to 0 to disable.
23 56199 [iot_thread] [INFO ][MQTT][56198] (MQTT connection 2004ef00, CONNECT op
24 56341 [iot_thread] [INFO ][MQTT][56341] (MQTT connection 2004ef00, CONNECT op
25 56354 [iot_thread] [INFO ][MQTT][56354] New MQTT connection 20042bdc establis
26 56361 [iot_thread] Connected to broker.
27 56367 [iot_thread] [OTA_AgentInit_internal] OTA Task is Ready.
28 56373 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [Ready
29 56386 [OTA Agent Task] [INFO ][MQTT][56385] (MQTT connection 2004ef00) SUBSCR
30 56395 [OTA Agent Task] [INFO ][MQTT][56395] (MQTT connection 2004ef00, SUBSCR
.31 56535 [OTA Agent Task] [INFO ][MQTT][56535] (MQTT connection 2004ef00, SUBSC
32 56547 [OTA Agent Task] [prvSubscribeToJobNotificationTopics] OK: $aws/things/
33 56559 [OTA Agent Task] [INFO ][MQTT][56559] (MQTT connection 2004ef00) SUBSCR
34 56569 [OTA Agent Task] [INFO ][MQTT][56569] (MQTT connection 2004ef00, SUBSCR
35 56694 [OTA Agent Task] [INFO ][MQTT][56694] (MQTT connection 2004ef00, SUBSCR
36 56706 [OTA Agent Task] [prvSubscribeToJobNotificationTopics] OK: $aws/things/
37 56716 [OTA Agent Task] [prvRequestJob_Mqtt] Request #0
38 56723 [OTA Agent Task] [INFO ][MQTT][56723] (MQTT connection 2004ef00) MQTT P
39 56733 [OTA Agent Task] [INFO ][MQTT][56733] (MQTT connection 2004ef00, PUBLIS
40 58580 [iot_thread] State: Ready  Received: 0   Queued: 0   Processed: 0   Dro
.41 59855 [iot_thread] State: RequestingJob  Received: 0   Queued: 0   Processed
42 59950 [OTA Agent Task] [INFO ][MQTT][59950] (MQTT connection 2004ef00, PUBLIS
43 59962 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [Reque
44 59975 [OTA Agent Task] [prvParseJobDoc] Size of OTA_FileContext_t [64]
45 59982 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ clientToke
46 59991 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: execution
47 59999 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: jobId
48 60006 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: jobDocume
49 60014 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: afr_ota
50 60021 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: protocols
51 60028 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: files
52 60035 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: filepath
53 60043 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: filesize
54 60050 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: fileid
55 60057 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: certfile
56 60066 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: sig-sha25
57 60074 [OTA Agent Task] [prvDefaultCustomJobCallback] Received Custom Job insi
58 60085 [OTA Agent Task] [prvParseJobDoc] Ignoring job without ID.
59 60091 [OTA Agent Task] [prvOTA_Close] Context->0x20058888
60 60097 [OTA Agent Task] [prvPAL_SetPlatformImageState] Current state (SE_FwSta
61 60105 [OTA Agent Task] [prvPAL_SetPlatformImageState] Request to change state
62 60115 [OTA Agent Task] [prvOTAAgentTask] Handler failed. Current State [Waiti
63 60128 [OTA Agent Task] [prvParseJobDoc] Size of OTA_FileContext_t [64]
64 60136 [iot_thread] [INFO ][MQTT][60136] Establishing new MQTT connection.
65 60144 [iot_thread] [INFO ][MQTT][60143] Anonymous metrics (SDK language, SDK                                                                                                                                                              ICS set to 0 to disable.
66 60160 [iot_thread] [INFO ][MQTT][60160] (MQTT connection 200487a8, CONNECT op
67 60174 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ clientToke
68 60183 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: execution
69 60190 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: jobId
[... 422 lines cut ...]
[... cut ...]
430 130121 [iot_thread] [INFO ][Shadow][130121] Shadow library cleanup done.
431 130128 [iot_thread] [INFO ][MQTT][130128] MQTT library cleanup done.
432 130135 [iot_thread] [INFO ][DEMO][130135] memory_metrics::freertos_heap::bef
433 130144 [iot_thread] [INFO ][DEMO][130144] memory_metrics::freertos_heap::aft
434 130153 [iot_thread] [INFO ][DEMO][130152] memory_metrics::demo_task_stack::b
435 130161 [iot_thread] [INFO ][DEMO][130161] memory_metrics::demo_task_stack::a
436 130170 [iot_thread] [INFO ][DEMO][130170] Demo completed successfully.

yanjos-dev, 2020-10-23:
Hi @AlnurElberier,

What version of the code are you using? Can you also share what platform you are running the code on?

AlnurElberier, 2020-10-23:
I believe I am using FreeRtos Version 202002.00, as it is included in the X-CUBE-AWS package for STM32 devices ( https://www.st.com/en/embedded-software/x-cube-aws.html ). 

I am using the STMicroelectronics  B-L4S5I-IOT01A Discovery Kit, and IAR is my IDE.

AlnurElberier, 2020-10-23:
I was able to get a more predictable output  when I set a 20 second delay in aws_iot_demo_shadow.c:

int RunShadowDemo( bool awsIotMqttMode,
                   const char * pIdentifier,
                   void * pNetworkServerInfo,
                   void * pNetworkCredentialInfo,
                   const IotNetworkInterface_t * pNetworkInterface )
{
    
    /* Return value of this function and the exit status of this program. */
    int status = 0;
    configPRINTF( ( "Shadow Demo\r\n" ) );
    
    vTaskDelay( 20000 ); //ADDED DELAY HERE
    /* Handle of the MQTT connection used in this demo. */
    IotMqttConnection_t mqttConnection = IOT_MQTT_CONNECTION_INITIALIZER;

    /* Length of Shadow Thing Name. */
    size_t thingNameLength = 0;

    /* Allows the Shadow update function to wait for the delta callback to complete
     * a state change before continuing. */
    IotSemaphore_t deltaSemaphore;

    /* Flags for tracking which cleanup functions must be called. */
    bool librariesInitialized = false, connectionEstablished = false, deltaSemaphoreCreated = false;

    /* The first parameter of this demo function is not used. Shadows are specific
     * to AWS IoT, so this value is hardcoded to true whenever needed. */
    ( void ) awsIotMqttMode;

    /* Determine the length of the Thing Name. */
    if( pIdentifier != NULL )
    {
        thingNameLength = strlen( pIdentifier );

        if( thingNameLength == 0 )
        {
            IotLogError( "The length of the Thing Name (identifier) must be nonzero." );

            status = EXIT_FAILURE;
        }
    }
    else
    {
        IotLogError( "A Thing Name (identifier) must be provided for the Shadow demo." );

        status = EXIT_FAILURE;
    }

    /* Initialize the libraries required for this demo. */
    if( status == EXIT_SUCCESS )
    {
        status = _initializeDemo();
    }

    if( status == EXIT_SUCCESS )
    {
        /* Mark the libraries as initialized. */
        librariesInitialized = true;

        /* Establish a new MQTT connection. */
        status = _establishMqttConnection( pIdentifier,
                                           pNetworkServerInfo,
                                           pNetworkCredentialInfo,
                                           pNetworkInterface,
                                           &mqttConnection );
    }

    if( status == EXIT_SUCCESS )
    {
        /* Mark the MQTT connection as established. */
        connectionEstablished = true;

        /* Create the semaphore that synchronizes with the delta callback. */
        deltaSemaphoreCreated = IotSemaphore_Create( &deltaSemaphore, 0, 1 );

        if( deltaSemaphoreCreated == false )
        {
            status = EXIT_FAILURE;
        }
    }

    if( status == EXIT_SUCCESS )
    {
        /* Set the Shadow callbacks for this demo. */
        status = _setShadowCallbacks( &deltaSemaphore,
                                      mqttConnection,
                                      pIdentifier,
                                      thingNameLength );
    }

    if( status == EXIT_SUCCESS )
    {
        /* Clear the Shadow document so that this demo starts with no existing
         * Shadow. */
        _clearShadowDocument( mqttConnection, pIdentifier, thingNameLength );

        /* Send Shadow updates. */
        status = _sendShadowUpdates( &deltaSemaphore,
                                     mqttConnection,
                                     pIdentifier,
                                     thingNameLength );

        /* Delete the Shadow document created by this demo to clean up. */
        _clearShadowDocument( mqttConnection, pIdentifier, thingNameLength );
    }

    /* Disconnect the MQTT connection if it was established. */
    if( connectionEstablished == true )
    {
        IotMqtt_Disconnect( mqttConnection, 0 );
    }

    /* Clean up libraries if they were initialized. */
    if( librariesInitialized == true )
    {
        _cleanupDemo();
    }

    /* Destroy the delta semaphore if it was created. */
    if( deltaSemaphoreCreated == true )
    {
        IotSemaphore_Destroy( &deltaSemaphore );
    }

    return status;
}

But when I try and push an actual OTA, it fails to actually update the firmware. I'll attach the terminal output for the OTA in a file below. 
[OTA_terminalOutput.txt](https://github.com/aws/amazon-freertos/files/5431173/OTA_terminalOutput.txt)

yanjos-dev, 2020-10-26:
Thank you for providing these logs. It looks like the failure is related to the TLS send functionality. Do you know if you have TLS offloaded or not?

AlnurElberier, 2020-10-26:
I am not sure. How would I check?

AlnurElberier, 2020-10-26:
also thanks for getting back in such a timely manner

AlnurElberier, 2020-10-27:
Just following up, please let me know if you have any suggestions. Thanks

leegeth, 2020-11-11:
@AlnurElberier, 

Apologize for the delay in response. You will be able to check if the TLS stack is offloaded by verifying if the config `USE_OFFLOAD_SSL` is defined. If it is defined, ST example applications in [amazon-freertos](https://github.com/aws/amazon-freertos), offloaded stack will be used by ST. For example, it is defined in [CMakeLists.txt](https://github.com/aws/amazon-freertos/blob/master/vendors/st/boards/stm32l475_discovery/CMakeLists.txt#L39) for ST in amazon-freertos. Please verify if it is defined in your IAR IDE project files to check if TLS stack is offloaded or not.

Thanks

yanjos-dev, 2021-03-11:
I am closing this issue due to inactivity. Please reopen this issue if you have any further questions.

closed 2021-03-11 by yanjos-dev