aws/amazon-freertos #2625: Add LICENSE file of third party libraries directly into amazon-freertos repository

DavidJurajdaNXP, 2020-10-26:
**Is your feature request related to a problem? Please describe.**  
I am missing license file of third party libraries in form of standalone LICENSE file.
It is difficult to discuss license issues without specific license file to reference.

**Describe the solution you would like.**  
I would like to have specific LICENSE file in third party library folder for direct referencing.
Specific case is https://github.com/zserge/jsmn/blob/master/LICENSE

Thank you!

leegeth, 2020-11-09:
@DavidJurajdaNXP ,

Thanks for bringing this issue to our attention and apologies for the delay in our response.  We are considering this suggestion and will update on any progress on addressing this issue.

Thanks

leegeth, 2020-11-11:
@DavidJurajdaNXP,

Addressed this issue in the PR #2737. Thanks for bringing this issue to us. Closing this issue. Please open a new issue if you have any further questions.

Thanks

closed 2020-11-11 by leegeth