aws/amazon-freertos #2638: Compilation warnings in tool-chain IAR and armgcc

lylezhu2012, 2020-10-27:
**Describe the bug**
Compilation warnings in tool-chain IAR (8.50.6) and armgcc(9-2020-q2).

**System information**
- Board: MIMXRT1060-EVK
- IDE: IAR (version 8.50.6) and ARM GCC (9-2020-q2)
- Operating System [Windows]
- Version: 202007.00 release

**Expected behavior**

**Screenshots or console output**
The Warnings in IAR as following,
aws_iot_demo_network.c 
aws_ble_gatt_server_demo.c 
C:\c\pkg\rtos\freertos\demos\ble\aws_ble_gatt_server_demo.c(254) : Warning[Pe111]: statement is unreachable
C:\c\pkg\rtos\freertos\demos\ble\aws_ble_gatt_server_demo.c(272) : Warning[Pe111]: statement is unreachable
C:\c\pkg\rtos\freertos\demos\ble\aws_ble_gatt_server_demo.c(275) : Warning[Pe177]: function "vGattDemoSvcHook" was declared but never referenced
aws_wifi_connect_task.c 
C:\c\pkg\rtos\freertos\demos\wifi_provisioning\aws_wifi_connect_task.c(128) : Warning[Pe111]: statement is unreachable
cborvalidation.c 
C:\c\pkg\rtos\freertos\libraries\3rdparty\tinycbor\src\cborvalidation.c(659) : Warning[Pe068]: integer conversion resulted in a change of sign
iot_ble_gap.c 
C:\c\pkg\rtos\freertos\libraries\c_sdk\standard\ble\src\iot_ble_gap.c(405) : Warning[Pe188]: enumerated type mixed with another type
iot_ble_numericComparison.c 
C:\c\pkg\rtos\freertos\demos\ble\iot_ble_numericComparison.c(110) : Warning[Pe111]: statement is unreachable
iot_ble_wifi_provisioning.c 
C:\c\pkg\rtos\freertos\libraries\c_sdk\standard\ble\src\services\wifi_provisioning\iot_ble_wifi_provisioning.c(252) : Warning[Pe188]: enumerated type mixed with another type
C:\c\pkg\rtos\freertos\libraries\c_sdk\standard\ble\src\services\wifi_provisioning\iot_ble_wifi_provisioning.c(252) : Warning[Pe188]: enumerated type mixed with another type
C:\c\pkg\rtos\freertos\libraries\c_sdk\standard\ble\src\services\wifi_provisioning\iot_ble_wifi_provisioning.c(333) : Warning[Pe188]: enumerated type mixed with another type
C:\c\pkg\rtos\freertos\libraries\c_sdk\standard\ble\src\services\wifi_provisioning\iot_ble_wifi_provisioning.c(333) : Warning[Pe188]: enumerated type mixed with another type
C:\c\pkg\rtos\freertos\libraries\c_sdk\standard\ble\src\services\wifi_provisioning\iot_ble_wifi_provisioning.c(456) : Warning[Pe188]: enumerated type mixed with another type
C:\c\pkg\rtos\freertos\libraries\c_sdk\standard\ble\src\services\wifi_provisioning\iot_ble_wifi_provisioning.c(456) : Warning[Pe188]: enumerated type mixed with another type
C:\c\pkg\rtos\freertos\libraries\c_sdk\standard\ble\src\services\wifi_provisioning\iot_ble_wifi_provisioning.c(541) : Warning[Pe188]: enumerated type mixed with another type
C:\c\pkg\rtos\freertos\libraries\c_sdk\standard\ble\src\services\wifi_provisioning\iot_ble_wifi_provisioning.c(690) : Warning[Pe188]: enumerated type mixed with another type
C:\c\pkg\rtos\freertos\libraries\c_sdk\standard\ble\src\services\wifi_provisioning\iot_ble_wifi_provisioning.c(690) : Warning[Pe188]: enumerated type mixed with another type
C:\c\pkg\rtos\freertos\libraries\c_sdk\standard\ble\src\services\wifi_provisioning\iot_ble_wifi_provisioning.c(819) : Warning[Pe188]: enumerated type mixed with another type
C:\c\pkg\rtos\freertos\libraries\c_sdk\standard\ble\src\services\wifi_provisioning\iot_ble_wifi_provisioning.c(819) : Warning[Pe188]: enumerated type mixed with another type
C:\c\pkg\rtos\freertos\libraries\c_sdk\standard\ble\src\services\wifi_provisioning\iot_ble_wifi_provisioning.c(926) : Warning[Pe188]: enumerated type mixed with another type
C:\c\pkg\rtos\freertos\libraries\c_sdk\standard\ble\src\services\wifi_provisioning\iot_ble_wifi_provisioning.c(1032) : Warning[Pe188]: enumerated type mixed with another type

Warnings on ARMGCC as following,
[ 93%] Building C object CMakeFiles/ethermind_wifi_provisioning.elf.dir/C_/c/pkg/rtos/freertos/demos/ble/iot_ble_numericComparison.c.obj
C:\c\pkg\rtos\freertos\demos\ble\aws_ble_gatt_server_demo.c:275:19: warning: 'vGattDemoSvcHook' defined but not used [-Wunused-function]
  275 | static BaseType_t vGattDemoSvcHook( void )
      |                   ^~~~~~~~~~~~~~~~

Thank you!

ravibhagavandas, 2020-10-29:
Hello,
Thanks for reporting this.  We will look into fixing these warnings.

lylezhu2012, 2020-11-09:
Hi,
Thanks.

ravibhagavandas, 2020-12-22:
Here is the PR to fix the warnings #2908 .

ravibhagavandas, 2021-01-04:
Closing this issue as the related PR is merged now.

closed 2021-01-04 by ravibhagavandas