aws/amazon-freertos #2649: IDT v3.3.0 doesn't work with latest Amazon FreeRTOS master branch

shubhamkulkarni97, 2020-10-28:
None of the IDT test cases work with latest master.
IDT outputs following error log:

```
time="2020-10-28T17:02:53+05:30" level=info msg=Using product version "202007.00"
time="2020-10-28T17:02:54+05:30" level=info msg=Using test suite "FRQ_1.2.0" for execution
time="2020-10-28T17:02:54+05:30" level=info msg=Execution failed due to error: open /Users/shubhamkulkarni/forkedafr/amazon-freertos/libraries/coreJSON/test/cbmc/proofs/Makefile.common: no such file or directory deviceId=ESP32 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets
time="2020-10-28T17:02:54+05:30" level=error msg=panic: runtime error: invalid memory address or nil pointer dereference executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32
time="2020-10-28T17:02:54+05:30" level=error msg=[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x1a061f5] suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04
time="2020-10-28T17:02:54+05:30" level=error msg= executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32
time="2020-10-28T17:02:54+05:30" level=error msg=goroutine 1 [running]: executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32
time="2020-10-28T17:02:54+05:30" level=error msg=go.amzn.com/bridgekeeper/afr/reporter.(*reportGenerator).ReportTestGroupResultWithError(0x0, 0x20f7400, 0xc00008ea80, 0x0, 0x0, 0x0, 0x0, 0x0) testCaseId=Full_SecureSockets deviceId=ESP32 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ groupId=FullSecureSockets
time="2020-10-28T17:02:54+05:30" level=error msg=	/local/p4clients/pkgbuild-TUazn/workspace/src/GoAmzn-BridgekeeperAFRQualificationSuite/src/go.amzn.com/bridgekeeper/afr/reporter/reporter.go:280 +0x45 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04
time="2020-10-28T17:02:54+05:30" level=error msg=go.amzn.com/bridgekeeper/afr/executor.executor.generateReportWithError(...) executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32
time="2020-10-28T17:02:54+05:30" level=error msg=	/local/p4clients/pkgbuild-TUazn/workspace/src/GoAmzn-BridgekeeperAFRQualificationSuite/src/go.amzn.com/bridgekeeper/afr/executor/executor.go:172 groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ
time="2020-10-28T17:02:54+05:30" level=error msg=go.amzn.com/bridgekeeper/afr/executor.executor.Execute.func1(0xc00051b318, 0xc00017cfe0) executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32
time="2020-10-28T17:02:54+05:30" level=error msg=	/local/p4clients/pkgbuild-TUazn/workspace/src/GoAmzn-BridgekeeperAFRQualificationSuite/src/go.amzn.com/bridgekeeper/afr/executor/executor.go:69 +0x208 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04
time="2020-10-28T17:02:54+05:30" level=error msg=go.amzn.com/bridgekeeper/afr/executor.executor.Execute(0x7ffeefbfec88, 0x8, 0xc00031a0e0, 0x5, 0xc00031a0f0, 0x4, 0x0, 0x0, 0xc00031a0e8, 0x6, ...) deviceId=ESP32 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets
time="2020-10-28T17:02:54+05:30" level=error msg=	/local/p4clients/pkgbuild-TUazn/workspace/src/GoAmzn-BridgekeeperAFRQualificationSuite/src/go.amzn.com/bridgekeeper/afr/executor/executor.go:95 +0xf2e deviceId=ESP32 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets
time="2020-10-28T17:02:54+05:30" level=error msg=go.amzn.com/bridgekeeper/afr/command.GetRunTestCommand.func1(0xc000305180, 0xc00020d080, 0x0, 0x18, 0x0, 0x0) groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ
time="2020-10-28T17:02:54+05:30" level=error msg=	/local/p4clients/pkgbuild-TUazn/workspace/src/GoAmzn-BridgekeeperAFRQualificationSuite/src/go.amzn.com/bridgekeeper/afr/command/runTestCommand.go:64 +0x2d5 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32
time="2020-10-28T17:02:54+05:30" level=error msg=github.com/spf13/cobra.(*Command).execute(0xc000305180, 0xc00020cf00, 0x18, 0x18, 0xc000305180, 0xc00020cf00) executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32
time="2020-10-28T17:02:54+05:30" level=error msg=	/local/p4clients/pkgbuild-TUazn/workspace/build/GoAmzn-BridgekeeperAFRQualificationSuite/GoAmzn-BridgekeeperAFRQualificationSuite-1.0.2906.0/RHEL5_64/DEV.STD.PTHREAD/build/private/bgospace/Go3p-Github-Spf13-Cobra/src/github.com/spf13/cobra/command.go:826 +0x465 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04
time="2020-10-28T17:02:54+05:30" level=error msg=github.com/spf13/cobra.(*Command).ExecuteC(0xc000304f00, 0xc00017df70, 0x1, 0x1) groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ
time="2020-10-28T17:02:54+05:30" level=error msg=	/local/p4clients/pkgbuild-TUazn/workspace/build/GoAmzn-BridgekeeperAFRQualificationSuite/GoAmzn-BridgekeeperAFRQualificationSuite-1.0.2906.0/RHEL5_64/DEV.STD.PTHREAD/build/private/bgospace/Go3p-Github-Spf13-Cobra/src/github.com/spf13/cobra/command.go:914 +0x2fc suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04
time="2020-10-28T17:02:54+05:30" level=error msg=github.com/spf13/cobra.(*Command).Execute(...) groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ
time="2020-10-28T17:02:54+05:30" level=error msg=	/local/p4clients/pkgbuild-TUazn/workspace/build/GoAmzn-BridgekeeperAFRQualificationSuite/GoAmzn-BridgekeeperAFRQualificationSuite-1.0.2906.0/RHEL5_64/DEV.STD.PTHREAD/build/private/bgospace/Go3p-Github-Spf13-Cobra/src/github.com/spf13/cobra/command.go:864 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32
time="2020-10-28T17:02:54+05:30" level=error msg=main.main() groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ
time="2020-10-28T17:02:54+05:30" level=error msg=	/local/p4clients/pkgbuild-TUazn/workspace/src/GoAmzn-BridgekeeperAFRQualificationSuite/src/go.amzn.com/bridgekeeper/afr/handler/commandHandler.go:32 +0x181 executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 suiteId=FRQ groupId=FullSecureSockets testCaseId=Full_SecureSockets deviceId=ESP32
time="2020-10-28T17:02:54+05:30" level=error msg=Test exited unsuccessfully executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 testCaseId=Full_SecureSockets error=exit status 2
time="2020-10-28T17:02:54+05:30" level=error msg=Unable to add validation result to report executionId=4f62c8f0-1911-11eb-af5c-8c85904b7c04 testCaseId=Full_SecureSockets error=unable to get validation result from file /Users/shubhamkulkarni/gitrepo/afr_DQP/3.3.0/devicetester_freertos_mac/results/4f62c8f0-1911-11eb-af5c-8c85904b7c04/FRQ/FullSecureSockets/Full_SecureSockets/validationResult.json: validation result file does not exist: stat /Users/shubhamkulkarni/gitrepo/afr_DQP/3.3.0/devicetester_freertos_mac/results/4f62c8f0-1911-11eb-af5c-8c85904b7c04/FRQ/FullSecureSockets/Full_SecureSockets/validationResult.json: no such file or directory
```

However IDT works with other branches forked from older master branch. Some of the commits in master branch must have created an issue.

**To reproduce**
Run IDT v3.3.0 on latest master branch

Thank you!

ravibhagavandas, 2020-10-29:
@shubhamkulkarni97 
A particular IDT version is tagged to work against a FreeRTOS version tag. IDTv3.3.0 supports  FreeRTOS 202007.00 version.
For a list of supported versions with IDT you can refer to this page:
https://docs.aws.amazon.com/freertos/latest/userguide/dev-test-versions-afr.html

An IDT version is not guaranteed to work against latest master branch.

shubhamkulkarni97, 2020-10-30:
@ravibhagavandas,
What is expected date for next release tag?

This requirement is in context of qualification for ESP32-S2 SOC (Previous PR at https://github.com/aws/amazon-freertos/pull/2285)
Qualification tests need to be run on top of latest master.

Thanks,
Shubham

ravibhagavandas, 2020-11-02:
@shubhamkulkarni97 

We are working on our next release for Amazon FreeRTOS.  It should have an IDT release tag as well, which runs with the released version of Amazon FreeRTOS.

lundinc2, 2020-11-11:
Hi @shubhamkulkarni97 ,

Please see https://docs.aws.amazon.com/freertos/latest/userguide/dev-test-versions-afr.html for new version of IDT that works with https://github.com/aws/amazon-freertos/releases/tag/202011.00

shubhamkulkarni97, 2020-11-12:
Hi @ravibhagavandas @lundinc2,

I checked with latest version of IDT and 202011.00 tag of Amazon FreeRTOS, still I'm unable to run IDT.

On checking for the issue, I found [Makefile.common](https://github.com/FreeRTOS/coreJSON/blob/53895278e0b8a8dfb8f381ad42825200a41377a9/test/cbmc/proofs/Makefile.common) points to `../aws-templates-for-cbmc-proofs/template-for-repository/proofs/Makefile.common`. But `aws-templates-for-cbmc-proofs` submodule is not updated even after running `git submodule update --init --recursive`.

It says `Skipping submodule coreJSON` and aws-templates-for-cbmc-proofs directory is empty.

lundinc2, 2020-11-12:
Thanks for the report @shubhamkulkarni97 

I am taking a look.

lundinc2, 2020-11-12:
Hi @shubhamkulkarni97 I was able to reproduce. The new core* repositories make use of `update=none` in the git submodule configuration, to avoid writing extra libraries to the filesystem. IDT is expecting that all the submodules are checked out. by adding the `--checkout` flag to your git submodule command you can add these nested submodules to your file system.

For example `git submodule update --checkout --recursive --init`.

This is now documented here https://docs.aws.amazon.com/freertos/latest/userguide/dev-tester-prereqs.html

shubhamkulkarni97, 2020-11-17:
Thanks @lundinc2 for your response. Adding `--checkout` flag fixed the issue.

closed 2020-11-17 by shubhamkulkarni97