aws/amazon-freertos #2651: OTA not working if device was offline during OTA job creation in Console

AbdeenM, 2020-10-28:
Greetings,

So i have build a custom application based on the demos provided and now i want to add OTA feature to the firmware.
The flow is after my application connects to AWS via MQTT i pass the same connection to a function that starts the OTA agent on via a task that listens to any updates as per the demo then my main thread resumes normal.
You can find a sample of my ota_agent.c [here](https://gist.github.com/AbdeenM/c1471dd2bc1304571fbdb33a7703e84d).

Everything works well and the firmware upgrades and is task starts perfectly and listens to any update event on the background while the main app is running as expected.

However, if i create a OTA job from AWS console while the device is offline and turn on the device after the job is created the device boots up normally but never receives the OTA request or if i restart the device during the download time the app never resumes the download or attempts to upgrade via job again which am not sure why, any help would be great.
Also if you notice on line 196 the task is created, but am not sure if my approach is correct and what i did above is correct, if it is not can you point me on the right direction.

Thank you
Best Regards,
Abdeen

pvyawaha, 2020-10-29:
Hello ,

It will be really helpful if you can share the console logs for the changes you have in OTA demo for both cases when device is offline and when it resets at middle of download.

AbdeenM, 2020-11-01:
Hello @pvyawaha 

Thank you for your prompt response.
Okay i tired to regenerate the error but i cant seem to be able to get the case scenario that was working to work again (The scenario was when the OTA Job is started while the device is online, this used to work fine).
Now when i try to make an OTA Job while the device is online, this is what happens:
1. Device is booted and after everything is initialized is ready and listening for jobs. Job is created on console.
2. Device receives job and begins download.
3. Device downloads updated and the OTA callback triggers with an activate  event, which restarts the device in its cycle.
4. Device reboots with the new firmware then a self test timer begins.
5. after the time frame it states `Self test failed to complete within 16000ms` then reboots, deletes the passive firmware and then reverts to the old firmware.

Please note only difference is that i updated to the master branch release of aws-freertos, and am not really sure if this is something from my end or not. The code is exactly the same as the one i attached originally above and the logs are [here](https://gist.github.com/AbdeenM/aaecbb2b2ea813d0b4c5538c4caaf0af).

Thanks,
Best Regards,
Abdeen

pvyawaha, 2020-11-03:
Hello, 

Thank you for sharing the logs, it seems the device is receiving the complete update file , verified and reboots but during self-test it doesn't receive the job document within the self-test time.  I see some shadow operations added and bunch of unsubscribe but it is not clear which topics from the logs. Can you please check if device is not unsubscribing from the job topics?

Also, is it correct that you were able to run the OTA demo but this issue started after modifying ?

AbdeenM, 2020-11-03:
Hello,

You are most welcome. 
Yes correct the device receives the update, validates it and reboots just fine however the self test is what doesn’t go through. 
The shadow logs you see are because am running a custom application and now adding the ota as a feature to it, I will disable the shadow and have only ota for easier debugging my bad. 

About the ota demo I have not ran it before as I am working on the custom app, however the ota was indeed working just fine the issue in question was not there. 

I suspect that it is because on my ota callback function am not receiving the eOTA_JobEvent_StartTest event so OTA_SetImageState( eOTA_ImageState_Accepted ); is never really called, I never got any log from that event. 

During the self test and after reboot does the ota callback function fire with a eOTA_JobEvent_StartTest event?

Thanks

AbdeenM, 2020-11-05:
Hello @pvyawaha hope you are doing well.
Did you get a chance to go through my previous message? 

Thanks

pvyawaha, 2020-11-06:
Hello,

Thank you for sharing the logs , it is clear that when the device reboots the new firmware it is not receiving the job document within the default 16 seconds timeout. Can you please check the job status after you reboot and provide me status details which has the error codes?

You can follow instructions here https://docs.aws.amazon.com/freertos/latest/userguide/ota-failure-codes.html

Also the thing names that you are using is more than the default limit , so is the config otaconfigMAX_THINGNAME_LEN increased? This can be increased to max 128.

AbdeenM, 2020-11-06:
Hello,

You are most welcome.
The job status for all my failed attempts is always stuck on `IN_PROCESS` it does not update to failed or anything else. as soon as the job is created it is in queue once the download begins it changes to in process and stays that way.
I still have the job from the previous logs shared so following the guide you shared i was able to print its status, please find it below:
```JSON
{
    "execution": {
        "jobId": "AFR_OTA-ota-13",
        "status": "IN_PROGRESS",
        "statusDetails": {
            "detailsMap": {
                "self_test": "ready",
                "updatedBy": "0x90002"
            }
        },
        "thingArn": "arn:aws:iot:us-east-1:XXXXXXXXXXX:thing/5c87b08ae4446c5d65fddbfd5cb6093215e7e18439c1c1f2336bafb7b80d06ee",
        "queuedAt": "2020-11-03T22:50:53.741000+02:00",
        "startedAt": "2020-11-03T22:51:46.974000+02:00",
        "lastUpdatedAt": "2020-11-03T22:53:46.331000+02:00",
        "executionNumber": 1,
        "versionNumber": 5
    }
}
```

About the thing name, i have not modified the config for `otaconfigMAX_THINGNAME_LEN` as it says the limit is 64U and i am using a 64 character string for all my devices, while establishing the MQTT however am using the first 23 characters (see [here](https://gist.github.com/AbdeenM/21699c66a321350a60e3efbbc043f254#file-logs-txt-L106)) for the unique client identifier as that is the limit.
Note the thing names am using is actually generated during the fleet provisioning at the very start when my devices are being provisioned, once they are i receive back the thing name which is the certificate id during provisioning and store it in the flash memory within each device for further use. do you think i should change that or shorten it maybe?

Thanks

AbdeenM, 2020-11-18:
Hello,

After using freertos 202007 and making some slight adjustments to the code, issues resolved.
Thanks, closing now

closed 2020-11-18 by AbdeenM