aws/amazon-freertos #2659: Is it possible to connect local mosquitto broker ??

vubon, 2020-10-29:
**Local mosquitto broker connection & AWS broker connection**  
1. Is it possible to connect local broker? As like mosquitto. 
2. If yes, How can I achieve this? 
3. If possible to connect a local broker, then I want to connect both local & aws. Is it possible? 

Please, If anyone knows this or test in previous time. Please give me some suggestions. 
Thank you!

ravibhagavandas, 2020-10-29:
Hi,
Yes its possible to use the mqtt library to connect to a local broker such as Mosquitto. To get started, you can use one of our demos like [mqtt_demo_mutual_auth.c](https://github.com/aws/amazon-freertos/blob/master/demos/coreMQTT/mqtt_demo_mutual_auth.c).
Here are the instructions to setup the mosquitto broker:  [windows](https://github.com/FreeRTOS/FreeRTOS/blob/master/FreeRTOS-Plus/Demo/coreMQTT_Windows_Simulator/MQTT_Mutual_Auth/mqtt_broker_setup.txt), [linux](https://github.com/aws/aws-iot-device-sdk-embedded-C#installing-mosquitto-to-run-mqtt-demos-locally).

You need to configure the credentials in appropriate configs within the demo. The demo can also be used to connect to AWS IoT broker.

lundinc2, 2020-11-12:
Hi @vubon,

Is the above sufficient for your question? If so I will close this issue.

Thanks,

Carl

vubon, 2020-11-12:
Hi @lundinc2 
I am sorry. I need to close the topic. Yes, I have done myself. The parallel connection established both brokers. Thank you for your suggestion.  @ravibhagavandas

closed 2020-11-12 by vubon