aws/amazon-freertos #2661: how to create or add custom source and header file in AWS freeRTOS

SGOSAI602, 2020-10-29:
Describe the bug
i have customised json file, now how add my (json) custom file in AWS freeRTOS?
how to create or add custom source and header file in AWS freeRTOS.

System information
Hardware - ESP32-WROOM-32 D
Operating System Linux-Ubuntu
Version of FreeRTOS (run git describe --tags to find it)
Project AWS FreeRTOS shadow demo

Thank you!

ravibhagavandas, 2020-10-30:
Hi @SGOSAI602,

Could you elaborate little further on what your problem is ? Also could link to shadow demo you are using?

SGOSAI602, 2020-10-31:
Hi ravibhagavandas,

I have a json file optimized and this file is platform independent. We can use any platform, how to use it in free RTOS, how to use this file and send data to the cloud.
Now how to add my (Json) custom file to AWS FreeRTOS?

Thank you!

ravibhagavandas, 2020-11-01:
Hi @SGOSAI602 

Its not clear from the post what you are trying to do with custom JSON document, but from the first post it seems you are trying to include the JSON within a shadow document.

If so, you can take a look at the shadow demo [here](https://github.com/aws/amazon-freertos/tree/master/demos/device_shadow_for_aws_iot_embedded_sdk). You can make an update to the json document from any client by including the changes in `"state" : {  "desired" : { .... }  }` section of the shadow. The delta can be received on the device and device can publish back the applied changes in `"state" : {  "reported" : { .... }  }` section. This will complete the shadow update process for that change.
More details can be found in the document [here](https://docs.aws.amazon.com/iot/latest/developerguide/device-shadow-data-flow.html).

SGOSAI602, 2020-11-02:
Hi ravibhagavandas,

We are trying to have the same code for all platforms, so I have common codes for data convert into JSON format, now how to use it in AWS freertos code.
i hope now you are clear what we saying.

Thank you!

ravibhagavandas, 2020-11-03:
Thanks for the clarification. 
The JSON parser used in the shadow demos is a light weight parser which uses only two APIS: `JSON_Search` and `JSON_Validate` (more details in README  [here](https://github.com/FreeRTOS/coreJSON)] . You can replace this with your custom code for JSON parsing if needed. To convert data into JSON format, demo uses `snprintf` to insert data into a predefined [template](https://github.com/aws/amazon-freertos/blob/master/demos/device_shadow_for_aws_iot_embedded_sdk/shadow_demo_main.c#L89). This could be replaced with your custom code as well.

SGOSAI602, 2020-11-03:
hi ravibhagavandas,
thank you for understand,
And now my question is how to add my custom file to AWS freeRTOS, which I have to use API from define in this file.
 Thank you!

ravibhagavandas, 2020-11-03:
ESP32 project uses CMake as the build system. So you can add your custom source file under `afr_module_sources()` section within the shadow demo [CMakeFile](https://github.com/aws/amazon-freertos/blob/master/demos/coreMQTT/CMakeLists.txt#L9). Header file can be placed in a directory and the directory path can be added to the above CMakeFile as follows:

```
afr_module_include_dirs(
    ${AFR_CURRENT_MODULE}
    INTERFACE
        "<header file directory path>"
)
```

SGOSAI602, 2020-11-04:
hi ravibhagavandas,
as per you suggested add my custom source file under afr_module_sources() section within the shadow demo [CMakeFile.](https://github.com/aws/amazon-freertos/blob/master/demos/coreMQTT/CMakeLists.txt#L9) but in my freertos demo directory not show this [CMakeFile](https://github.com/aws/amazon-freertos/blob/master/demos/coreMQTT/CMakeLists.txt#L9) file.

[image]

leegeth, 2020-11-05:
@SGOSAI602,

You are unable to find the [CMakeLists.txt](https://github.com/aws/amazon-freertos/blob/master/demos/coreMQTT/CMakeLists.txt#L9), since it is pointing to the master branch. As you have correctly suspected, this is a new file added to the master branch recently. You will be able to make the edits proposed by @ravibhagavandas in the _CMakeLists.txt_ file shown in your screen shot. You can find the file from 202007.00 release [here](https://github.com/aws/amazon-freertos/blob/202007.00/demos/CMakeLists.txt).
Hope this answers your question. Please let us know if you have any more questions.

SGOSAI602, 2020-11-23:
thank you ravibhagavandas and leegeth, for support and valuable reply.

dan4thewin, 2020-11-24:
Hi, @SGOSAI602 .  I'm afraid I might be misunderstanding your question, especially what you mean by "merge".  Let me point to portions of the demo, and see if that helps.  The shadow demo uses a custom JSON document, defining it as a macro with a format string for use with snprintf.

This is where it is defined:
https://github.com/aws/amazon-freertos/blob/145dfbe7c9a9de16ba3c86d996e1b8edf3e41dff/demos/device_shadow_for_aws/shadow_demo_main.c#L89-L97

This is where it is used:
https://github.com/aws/amazon-freertos/blob/145dfbe7c9a9de16ba3c86d996e1b8edf3e41dff/demos/device_shadow_for_aws/shadow_demo_main.c#L911-L921

This link gives further details on the JSON format understood by the shadow service:
https://docs.aws.amazon.com/iot/latest/developerguide/device-shadow-document.html

I hope this shows how you can use JSON with the shadow service.  If I may be of further assistance, please try to be as specific as possible, perhaps showing any code you might already have written.  Thank you.

SGOSAI602, 2020-11-25:
hi dan4thewin,

saw Json file but not understand the how to replace AWS JSon file with my custom file and then send data on AWS server.
below attached my custom file and aws_iot_demo_shadow code .

[aws_iot_demo_shadow.zip](https://github.com/aws/amazon-freertos/files/5598652/aws_iot_demo_shadow.zip)

[cJSON.zip](https://github.com/aws/amazon-freertos/files/5598630/cJSON.zip)
I hope you can understand what I am trying to do.

Thank you!

dan4thewin, 2020-11-25:
Hi, @SGOSAI602 .   I hope I understand better now.  You have written a JSON library.  You need only use your functions to create your desired JSON document in a buffer.  This buffer is used as the body of an MQTT publish call (see `pciUpdateDocument` in the call to `PublishToTopic()` above).  The topic for that MQTT publish comes from the shadow library, also visible in the `PublishToTopic()` call above.  The shadow library does not call any JSON-related functions, nor does it call MQTT functions.  These steps happen in the application.

yanjos-dev, 2021-03-11:
I am closing this issue due to inactivity. Please reopen this issue if you have any further questions.

closed 2021-03-11 by yanjos-dev