aws/amazon-freertos #2678: Question: How to delete the wifi network saved by Wi-Fi provisioning service?

lylezhu2012, 2020-11-02:
**Describe the question**
 How to delete the wifi network saved by Wi-Fi provisioning service (remove the saved wifi AP configuration)?

**System information**
- Version: 202007.00 release

**Question**
I have configured a wifi network by following the wifi provisioning service steps, and then demo starts work. But I do not find how to delete the wifi network I saved.
After reboot the board, I tried to delete the saved wifi network by swiping the saved wifi item to the left to delete wifi information by repeating the wifi provisioning steps on my cellphone application.
But the error raised,
`16 17628 [XXXXX] [ERROR][BLE_WIFI_PROV][lu] Failed to process the delete network request. Already a request is being processed`

So, my question is how can I delete the saved wifi network configuration?  Whether the saved wifi netwoork can only be cleared by restoring factory settings?

And I noted that there is a change 893f09ee40f0f6ec46d0763dcb4e4c577b1405ba, the BLE connection will be removed once the wifi AP connected. So, in the wifi provisioning sequence, after the wifi AP is set, the target board will disconnect from the cellphone. As a result, the saved wifi network information cannot be displayed on the cellphone application. However, before this change, in the wifi provisioning phase, the saved network information will be displayed in the cellphone application. Is this behavior normal?

Thank you!

ravibhagavandas, 2020-11-07:
@lylezhu2012 

Demo sample provided, optimizes power usage for the device, so it disables BLE network once the WIFI is connected and application proceeds on WiFi. You can choose to have BLE  always enabled by removing lines [135-141](https://github.com/aws/amazon-freertos/blob/202007.00/demos/demo_runner/iot_demo_freertos.c#L135) and lines [165-171](https://github.com/aws/amazon-freertos/blob/202007.00/demos/demo_runner/iot_demo_freertos.c#L165) from `iot_demo_freertos.c`

The saved network can be deleted by sliding the connected network towards left. The error above seems to occur because a list network request or an save network request is ongoing, as the WiFi driver can only process one request. You can retry from the app to send the request again.
Also note that the demos completes , and it shuts down all the networks as part of cleanup. You may want to wait in the demo if you need to try all the features from app.

lylezhu2012, 2020-11-09:
Hi @ravibhagavandas,

> @lylezhu2012
>
> Demo sample provided, optimizes power usage for the device, so it disables BLE network once the WIFI is connected and application proceeds on WiFi. You can choose to have BLE always [...]

Could you add a configuration for the changes? So, we can configure the feature in application layer.

> The saved network can be deleted by sliding the connected network towards left. The error above seems to occur because a list network request or an save network request is ongoing, as the WiFi [...]

ravibhagavandas, 2020-11-12:
@lylezhu2012 
Thanks for your suggestion. This seems to be more an application defined behavior which can vary across each applications. We will look into how best we can incorporate this into our demo sample.

ravibhagavandas, 2020-11-23:
@lylezhu2012 
I am closing this issue. Please feel free to reopen this if you have any further questions.

closed 2020-11-23 by ravibhagavandas