aws/amazon-freertos #2679: CoreMQTT, MQTT_Init callback context

AndreaPetrelli, 2020-11-02:
**Is your feature request related to a problem? Please describe.**  
With device_shadow_for_aws apis, it would be useful to add a pointer to a callback context.

**Describe the solution you would like.**  
New parameter called pCallbackContext in MQTT_Init

MQTTStatus_t MQTT_Init( MQTTContext_t * pContext,
                        const TransportInterface_t * pTransportInterface,
                        MQTTGetCurrentTimeFunc_t getTimeFunction,
                        MQTTEventCallback_t userCallback,
                        const MQTTFixedBuffer_t * pNetworkBuffer,
                        void * pCallbackContext )

that will be assigned to a new field in MQTTContext_t structure

Thank you!

aggarw13, 2020-11-03:
Hi @AndreaPetrelli, thank you for posting your request for addition of a callback context in the `MQTTContext_t` structure. Please be aware of the option of creating an issue related to the coreMQTT library directly on the [coreMQTT](https://github.com/FreeRTOS/coreMQTT/issues) repository.

To make sure we understand the purpose of your request, would you want the callback context to be provided in invocation of the `MQTTEventCallback_t` type `userCallback` that you register with the library through the `MQTT_Init` API?

We will add your feature request to our internal queue, and consider it for adding it in a future release of the **coreMQTT** 
library.

leegeth, 2020-11-11:
Hi @AndreaPetrelli,

We would also like to know that if there was any specific usecase you were trying to address and found this as a limitation. Please let us know.
Also please let us know if you have any follow up questions.

Thanks

AndreaPetrelli, 2020-11-11:
Hi,
i'm wrapping the library in a c++ class.
I would like to pass the instance of the class in pCallbackContext so i can recover in the MQTTEventCallback_t callback through the MQTTContext_t struct.

Thanks

leegeth, 2020-11-11:
@AndreaPetrelli,

Thanks for the response. So as to understand a bit more about the high level usecase, are you using multiple instances of the class in your application?

Thanks

AndreaPetrelli, 2020-11-29:
Not currently

Thanks

leegeth, 2021-01-20:
@AndreaPetrelli ,

We are closing this issue as we are internally tracking your proposal as a feature request. Please open a new issue if you have any follow up questions.

closed 2021-01-20 by leegeth