aws/amazon-freertos #2680: Heap corruption error

Padmavathi-Balachandran, 2020-11-02:
Hi. I get the Heap Corruption error when i try to use the esp_restart() function. I tried using the heap_caps_malloc() instead of malloc() but not fixed. Any suggestions would be of great help. Thankyou! 

**CORRUPT HEAP: multi_heap.c:432 detected at 0x3ffeac98
abort() was called at PC 0x4009610f on core 0
0x4009610f: multi_heap_assert at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_malloc_impl at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/heap/multi_heap.c:432**

ELF file SHA256: 56db6a67540bf16de7e569074aee1daea94c33baf75a448708dc7573c6d624c3

Backtrace: 0x4008dfa4:0x3ffddb90 0x4008e1d1:0x3ffddbb0 0x4009610f:0x3ffddbd0 0x40095af7:0x3ffddbf0 0x4008540d:0x3ffddc10 0x4008543e:0x3ffddc30 0x40085751:0x3ffddc50 0x40085989:0x3ffddc90 0x400949de:0x3ffddcb0 0x4012593d:0x3ffddcf0 0x40125a2e:0x3ffddd20 0x401217cb:0x3ffddd80 0x4012f414:0x3ffddde0 0x401675e9:0x3ffde040 0x40167626:0x3ffde070 0x4017086e:0x3ffde0b0 0x401711d6:0x3ffde0e0 0x40171d07:0x3ffde100 0x40164de8:0x3ffde140 0x4017429f:0x3ffde170 0x40169ab9:0x3ffde1a0 0x40169acb:0x3ffde1c0 0x4016e1ee:0x3ffde1e0 0x40124367:0x3ffde200
0x4008dfa4: invoke_abort at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x4008e1d1: abort at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x4009610f: multi_heap_assert at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_malloc_impl at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/heap/multi_heap.c:432

0x40095af7: multi_heap_malloc at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/heap/multi_heap_poisoning.c:370

0x4008540d: heap_caps_malloc at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/heap/heap_caps.c:232

0x4008543e: heap_caps_malloc_default at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/heap/heap_caps.c:232

0x40085751: trace_malloc at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/heap/heap_trace.c:188

0x40085989: __wrap_heap_caps_malloc_default at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/heap/heap_trace.c:433

0x400949de: xTaskCreate at /home/firewires/Desktop/6port_ble/ota_test_starboard/freertos_kernel/tasks.c:5068

0x4012593d: wifi_connect_task at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/boards/esp32/aws_demos/application_code/aws_ble_gatt_server_demo.c:1065

0x40125a2e: vWritePassword at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/boards/esp32/aws_demos/application_code/aws_ble_gatt_server_demo.c:859 (discriminator 4)

0x401217cb: _requestWriteCb at /home/firewires/Desktop/6port_ble/ota_test_starboard/libraries/c_sdk/standard/ble/src/iot_ble_gatt.c:529

0x4012f414: prvGATTCharAccessCb at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/boards/esp32/ports/ble/nimble/iot_ble_hal_gatt_server.c:402

0x401675e9: ble_gatts_val_access at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/src/ble_gatts.c:2144

0x40167626: ble_gatts_chr_val_access at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/src/ble_gatts.c:2144

0x4017086e: ble_att_svr_write at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/src/ble_att_svr.c:2697

0x401711d6: ble_att_svr_write_handle at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/src/ble_att_svr.c:2697

0x40171d07: ble_att_svr_rx_write at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/src/ble_att_svr.c:2697

0x40164de8: ble_att_rx at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/src/ble_att.c:518

0x4017429f: ble_hs_hci_evt_acl_process at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:870

0x40169ab9: ble_hs_process_rx_data_queue at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:109

0x40169acb: ble_hs_event_rx_data at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:109

0x4016e1ee: ble_npl_event_run at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:121
 (inlined by) nimble_port_run at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/esp-idf/components/nimble/nimble/porting/nimble/src/nimble_port.c:82

0x40124367: ble_host_task at /home/firewires/Desktop/6port_ble/ota_test_starboard/vendors/espressif/boards/esp32/ports/ble/nimble/iot_ble_hal_common_gap.c:538

Rebooting...

ravibhagavandas, 2020-11-05:
Hi,

Are there some steps to reproduce this behavior ? Usually heap corruptions are caused by buffer overflow writes at some other points of code, however its worthwhile to check if you are running out of heap. APIs `xPortGetFreeHeapSize()` can be used to find the remaining free heap size in bytes and `xPortGetMinimumEverFreeHeapSize()` can be used to find the lowest water mark of heap (useful in finding if it ever ran out of heap).

Also there are some heap corruption debugging tips from Espressif documentation [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#heap-corruption).

lundinc2, 2020-12-15:
Hi @Padmavathi-Balachandran,

Is this issue still happening?

closed 2021-02-25 by lundinc2