aws/amazon-freertos #2717: [MQTT] Wifi disconnects after a certain amount of time and cant restablish mqtt connection after wifi reconnectivity

MrMelon1232, 2020-11-06:
**Describe the bug**
Wifi would disconnect with the following reasons:
   I (1919778) wifi:ap_probe_send over, resett wifi status to disassoc
   I (1919778) wifi:state: run -> init (c800)
   I (1919788) wifi:pm stop, total sleep time: 1741958278 us / 1917948896 us
This would trigger an event that calls a callback function which essentially cleans up the old mqtt connection and reconnects to wifi and mqtt afterwards.
The callback indicates that wifi was disconnected for reason 23 which is "eWiFiReasonBeaconTimeout"
The problem is once wifi is connected, attempting to reconnect to mqtt fails and gives the following error:

6723 193970 [Wifi connectivi] ERROR: Handshake failed with error code <No-High-Level-Code> : BIGNUM - Memory allocation failed 
6724 193971 [Wifi connectivi] TLS_Connect fail (0x10, aau8njf5d2xbz-ats.iot.us-east-1.amazonaws.com)
6725 193972 [Wifi connectivi] [ERROR][NET][1939720] Failed to establish new connection. Socket status: -1.
6726 193973 [Wifi connectivi] [ERROR][MQTT][1939730] Failed to establish new MQTT connection, error NETWORK ERROR.
6727 193974 [Wifi connectivi] [ERROR][MQTT_APP][1939740] Error in establishing a mqtt connection...

**System information**
- ESP32-DevKit-C
- Visual Studio Code
- Linux
- Version of FreeRTOS : Commit be0e99978c8a74a24a659c2fe1dc78a6e80e49e3

**Expected behavior**
When wifi disconnects, restablish mqtt connection and resume app

Thank you!

leegeth, 2020-11-06:
@MrMelon1232 ,

Thanks for bringing this issue to our attention.  A few more details about this issue may help us to have clear understanding of the problem.

1. Which application are you using to test this?
2. Have you modified the code to handle the Wifi disconnect and reconnect to MQTT broker? If so, do you mind sharing it with us?
3. "An important note is that manually disconnecting the wifi and reconnecting to both wifi and mqtt works." , Would you be able to explain more about how you are trying this manual step and how different it is from the Wifi disconnects you were seeing?

Looking forward to hear from you soon.

MrMelon1232, 2020-11-08:
Hi @leegeth,

Thanks for the response! To answer your questions,
1. I am using an application I made myself that connects to wifi and mqtt and performs its tasks constantly (in a loop)
2. I have registered events to handle disconnectivity  (which would essentially free up the old mqtt connection and reconnect to both wifi and mqtt) using this:

WIFIReturnCode_t eventRegister;
//Set wifi callback
eventRegister = WIFI_RegisterEvent(eWiFiEventIPReady, _wifiCallBackFunction);
if( eventRegister == eWiFiSuccess )
{
    IotLogInfo("Registered wifi event ready");
    eventRegister = WIFI_RegisterEvent(eWiFiEventDisconnected, _wifiCallBackFunction);
    if(eventRegister == eWiFiSuccess)
        IotLogInfo("Registered wifi event disconnected");
}

and the function "_wifiCallBackFunction" is a callback function like so: 

static void _wifiCallBackFunction (WIFIEvent_t * xEvent){

    switch(xEvent->xEventType){
    
        case eWiFiEventDisconnected:
            IotLogInfo( "Disconnected from WiFi access point, reason code: %d.", xEvent->xInfo.xDisconnected.xReason );
            mqttApp.FSONET_APP_CURRENT_STATE = FSONET_APP_STATE_DISCONNECT;
            mqttApp.fsonet_app_wifi_connected = false;
            mqttApp.fsonet_app_wifi_error = false;
            IotLogInfo("Initiating reconnecting sequence");
            break;

        case eWiFiEventIPReady:
            mqttApp.FSONET_APP_CURRENT_STATE = FSONET_APP_STATE_INIT_WIFI;
            mqttApp.fsonet_app_wifi_connected = false;
            mqttApp.fsonet_app_wifi_error = false;
            break;
        
         case eWiFiEventConnectionFailed:
            break;
        
        case eWiFiEventReady:
            break;

        case eWiFiEventScanDone:
            break;
        
        case eWiFiEventConnected:
            break;

        case eWiFiEventAPStateChanged:
            break;

        case eWiFiEventAPStationConnected:
            break;
        
        case eWiFiEventAPStationDisconnected:
            break;

        case eWiFiEventWPSSuccess:
            break;
        
        case eWiFiEventWPSFailed:
            break;

        case eWiFiEventWPSTimeout:
            break;

        case eWiFiEventMax:
            break;
    }      
}   
This function sets some flags which causes the app to free the old mqtt connection if it has been established, and go to a reconnect state.
The function being called for connecting wifi is : "xWifiStatus = WIFI_ConnectAP( pxNetworkParams );" where pxNetworkParams are network parameters set beforehand.
The function being called for establishing an mqtt connection is: "connectStatus = IotMqtt_Connect( &networkInfo,
                                     &connectInfo,
                                     5000,
                                     pMqttConnection);"

3. Adding a disconnect function after performing the task once and using a hotspot and closing the connection are the methods used for testing wifi diconnect events. Once disconnected, the app indicates that a disconnect event has been triggered which would trigger the callback function and attempt to reconnect after setting some flags. Theses tests have been successful time and time again with wifi being able to reconnect and mqtt as well. In the case of this issue, connecting to mqtt would output the error above in the post.

MrMelon1232, 2020-11-09:
On an additional note for more details:
1. This issue started happening ever since the new implementations and changes of iot_wifi.c and iot_wifi.h changes to some functions, callbacks, and structures ; commit 24cac0f5906952e0cd360e56be8e28709d73574d
2. These are the entire logs of this issue:
I (1919499) wifi:ap_probe_send over, resett wifi status to disassoc
I (1919499) wifi:state: run -> init (c800)
I (1919499) wifi:pm stop, total sleep time: 1722279307 us / 1917503654 us

I (1919509) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (1919509) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 200
6659 191921 [eventTask] [INFO ][MQTT_APP][1919210] Disconnected from WiFi access point, reason code: 23.
6660 191921 [eventTask] [INFO ][MQTT_APP][1919210] Initiating reconnecting sequence
6675 192483 [Wifi connectivi] [INFO ][MQTT_APP][1924830] An mqtt connection has been established previously, clearing connection now!
6676 192484 [Wifi connectivi] [INFO ][MQTT][1924840] (MQTT connection 0x3ffb0548) Disconnecting connection.
6677 192486 [Wifi connectivi] [WARN ][NET][1924860] Failed to close connection.
6678 192487 [Wifi connectivi] [INFO ][MQTT][1924870] (MQTT connection 0x3ffb0548) Network connection closed.
6679 192489 [Wifi connectivi] [INFO ][MQTT][1924890] (MQTT connection 0x3ffb0548) Network connection destroyed.
6690 192990 [Wifi connectivi] [INFO ][MQTT_APP][1929900] Attempting to reconnect to wifi...
6691 192991 [Wifi connectivi] [INFO ][WIFI_CONNECTION][1929910] Remaining retries: 10
I (1930329) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (1930329) wifi:state: init -> auth (b0)
I (1930339) wifi:state: auth -> assoc (0)
I (1930349) wifi:state: assoc -> run (10)
I (1930359) wifi:connected {wifi information}
I (1930359) wifi:security type: {wifi information}
I (1930359) wifi:pm start, type: 1

I (1930369) WIFI: SYSTEM_EVENT_STA_CONNECTED
I (1930439) wifi:AP's beacon interval = 102400 us, DTIM period = 1

I (1931859) event: sta ip: {ip}, mask: {mask}, gw: {gw}
I (1931859) WIFI: SYSTEM_EVENT_STA_GOT_IP
6709 193655 [Wifi connectivi] [INFO ][MQTT_APP][1936550] Wifi is connected!
E (1936979) PKCS11: failed nvs get file size 4354 0
6711 193732 [Wifi connectivi] ERROR: Handshake failed with error code ECP - Memory allocation failed : <No-Low-Level-Code> 
6712 193733 [Wifi connectivi] TLS_Connect fail (0x4d80, {ENDPOINT})
6713 193734 [Wifi connectivi] [ERROR][NET][1937340] Failed to establish new connection. Socket status: -1.
6714 193735 [Sensor data] [INFO ][SENSOR_APP][1937350] Starting data sampling...
6715 193735 [Wifi connectivi] [ERROR][MQTT][1937350] Failed to establish new MQTT connection, error NETWORK ERROR.
6716 193736 [Wifi connectivi] [ERROR][MQTT_APP][1937360] Error in establishing a mqtt connection...

The following log after another attempt at mqtt connection is: 
6732 194335 [Wifi connectivi] ERROR: Handshake failed with error code RSA - The public key operation failed : BIGNUM - Memory allocation failed 
6733 194335 [Wifi connectivi] TLS_Connect fail (0x4290,  {ENDPOINT})
6734 194337 [Wifi connectivi] [ERROR][NET][1943370] Failed to establish new connection. Socket status: -1.
6735 194339 [Wifi connectivi] [ERROR][MQTT][1943390] Failed to establish new MQTT connection, error NETWORK ERROR.
6736 194340 [Wifi connectivi] [ERROR][MQTT_APP][1943400] Error in establishing a mqtt connection...

Then this log at the third attempt:
6784 195947 [Wifi connectivi] ERROR: Handshake failed with error code X509 - Certificate verification failed, e.g. CRL, CA or signature check failed : <No-Low-Level-Code> 
6785 195948 [Wifi connectivi] TLS_Connect fail (0x2700,  {ENDPOINT})
6786 195949 [Wifi connectivi] [ERROR][NET][1959490] Failed to establish new connection. Socket status: -1.
6787 195950 [Wifi connectivi] [ERROR][MQTT][1959500] Failed to establish new MQTT connection, error NETWORK ERROR.

Then this log at the fourth attempt and the rest of the attempts:
6861 198116 [Wifi connectivi] TLS_Connect fail (0x7f00, {ENDPOINT})
6862 198117 [Wifi connectivi] [ERROR][NET][1981170] Failed to establish new connection. Socket status: -1.
6863 198119 [Wifi connectivi] [ERROR][MQTT][1981190] Failed to establish new MQTT connection, error NETWORK ERROR.

leegeth, 2020-11-11:
@MrMelon1232,

Thanks for all the details and steps in reproducing this issue. From what I could see from the logs, the new MQTT connection fails because of a memory issue.  APIs xPortGetFreeHeapSize() can be used to find the remaining free heap size in bytes and xPortGetMinimumEverFreeHeapSize() can be used to find the lowest water mark of heap (useful in finding if it ever ran out of heap).  The documentation for using these APIs can be found in [FreeRTOS memory management documentation](https://www.freertos.org/a00111.html). Can you please give it a try to find the heap size when this issue happens?

On a related note, you will be able to enable an application hook for malloc failure as mentioned in the documentation [here](https://www.freertos.org/a00110.html#configUSE_MALLOC_FAILED_HOOK).  The default hook can be found in [iot_demo_freertos.c](https://github.com/aws/amazon-freertos/blob/master/demos/demo_runner/iot_demo_freertos.c#L410). While debugging, this application hook will help you to view the call stack for the malloc failure.

Please share your findings and we will be able to help you further.

Alternatively if you are ok in sharing your code, we can also try to reproduce it locally and help with debugging.

Thanks

closed 2020-11-17 by MrMelon1232

leegeth, 2020-11-18:
@MrMelon1232,

I see that you have closed this issue. Do you mind sharing the reason why you have closed this issue? Could you debug and root cause the issue? It will be helpful for us to know the reason so that we can better assist similar questions and issues reported in the future.

MrMelon1232, 2020-11-20:
Hi @leegeth ,

Well since the new implementations of core mqtt and other libraries, I wasn't getting this issue with the new implementations. If the issue arises again I will reopen the issue. Thanks for the follow up! It is much appreciated.

leegeth, 2020-11-20:
@MrMelon1232,

Glad to hear that your problem is resolved with new libraries. Thanks for letting us know.