aws/amazon-freertos #2746: AFR & ESP32: MQTT PUBLISH 0 could not be sent. Error NO RESPONSE

dipen-1337lab, 2020-11-13:
Namaste Forum Member,

The FreeRTOS version in use for development is V202002.00.

**The MQTT Demo runs as desired.** However, after the required modifications in my application code, I am getting the below error for the MQTT Publish operation.

    MQTT PUBLISH 0 could not be sent. Error NO RESPONSE.

The `RunMqttDemo()` in the `iot_demo_mqtt.c` source file has been modified as below.

    int RunMqttDemo()
    {

        ...
        ...
        _establishMqttConnection();
        _modifySubscriptions();
        
        while(1)
        {
            if( true == newReadingsAvailable )
            {
                _publishMessage(OutletOne);
                _publishMessage(OutletTwo);
            }
        }
        ...
        ...
    }

For debugging purpose, the below #define has been set to 1.

`#define PUBLISH_RETRY_LIMIT                      ( 1 )`

With this, the JSON response I get on the AWS IoT console when periodically publishing readings at 15 second interval and, the serial console log are as below.

**The Serial Console Log is shared below.**

    2 14 [iot_thread] [INFO ][DEMO][140] ---------STARTING DEMO---------

    I (153) BTDM_INIT: BT controller compile version [9891bc6]
    3 14 [iot_thread] [INFO ][INIT][140] SDK successfully initialized.
    I (163) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
    W (173) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
    I (313) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 2
    GAP procedure initiated: stop advertising.
    GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=480 adv_itvl_max=960
    I (613) wifi: wifi driver task: 3ffd91cc, prio:23, stack:3584, core=0
    I (613) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
    I (613) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
    I (643) wifi: wifi firmware version: c94f3e6
    I (643) wifi: config NVS flash: enabled
    I (643) wifi: config nano formating: disabled
    I (643) wifi: Init dynamic tx buffer num: 32
    I (643) wifi: Init data frame dynamic rx buffer num: 32
    I (653) wifi: Init management frame dynamic rx buffer num: 32
    I (653) wifi: Init management short buffer num: 32
    I (663) wifi: Init static rx buffer size: 1600
    I (663) wifi: Init static rx buffer num: 10
    I (663) wifi: Init dynamic rx buffer num: 32
    I (673) wifi: mode : sta (4c:11:ae:df:af:e8)
    I (683) WIFI: SYSTEM_EVENT_STA_START
    I (803) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
    I (1783) wifi: state: init -> auth (b0)
    I (1793) wifi: state: auth -> assoc (0)
    I (1793) wifi: state: assoc -> run (10)
    I (1903) wifi: connected with tkkfnB2, channel 1, BW20, bssid = 14:ae:85:e6:45:86
    I (1903) wifi: pm start, type: 1

    I (1903) WIFI: SYSTEM_EVENT_STA_CONNECTED
[... 172 lines cut ...]
          "timestamp": 1605296456143,
          "topic": "dualOutlet/topic/kWh"
        }
      ],
      "qos": 0,
      "topicFilter": "dualOutlet/#"
    }

The timestamp appended, which is part of the Publish payload message has the below values, corresponding to the 15 seconds interval -

        "timestamp": "1605296433",
        "timestamp": "1605296448",

Every 15 seconds, two MQTT messages are published. Further with the `PUBLISH_RETRY_LIMIT` set to 1, both these readings are sent again (retry) after which I receive the below message on the serial console.

`MQTT PUBLISH 0 could not be sent. Error NO RESPONSE.`

**Analysing the serial console logs, I realize that the `MQTT PUBLISH 0 successfully sent.` message is missing, which corresponds to the `_operationCompleteCallback()` function not getting called.** 

Requesting assistance to resolve this issue.

**PS**. Running the firmware for few minutes with PUBLISH_RETRY_LIMIT = 10 along with other application threads results in firmware restart, which as per my analyses is due to the Heap memory deallocation not happening because of the above MQTT Publish error.

Thanks | Regards,
Dipen

lundinc2, 2020-11-13:
Hi @dipen-1337lab,

Thank you for this report. I will begin working on reproducing this issue.

Thanks,

Carl

lundinc2, 2020-11-16:
Hi @dipen-1337lab,

Can you turn on the MQTT debug logging and provide another log? It seems that you are not getting the PUBACK packet, and since you are attempting QoS1 on the device, this is likely the issue. It may help to try to increase the timeout [here](https://github.com/aws/amazon-freertos/blob/4e8219e0923d459900924ff760ac492224b131fb/demos/mqtt/iot_demo_mqtt.c#L157). You can update the MQTT logging [here](https://github.com/aws/amazon-freertos/blob/987d9697ad4dc924bd71333c57b2f68bbe81b172/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h#L51).

dipen-1337lab, 2020-11-17:
Namaste @lundinc2,

[Here](https://github.com/aws/amazon-freertos/blob/987d9697ad4dc924bd71333c57b2f68bbe81b172/demos/include/iot_demo_logging.h#L42), I made the below modification.

```
//#ifdef IOT_LOG_LEVEL_DEMO
//    #define LIBRARY_LOG_LEVEL        IOT_LOG_LEVEL_DEMO
#ifdef IOT_LOG_LEVEL_MQTT
    #define LIBRARY_LOG_LEVEL        IOT_LOG_LEVEL_MQTT
#else
    #ifdef IOT_LOG_LEVEL_GLOBAL
        #define LIBRARY_LOG_LEVEL    IOT_LOG_LEVEL_GLOBAL
    #else
        #define LIBRARY_LOG_LEVEL    IOT_LOG_NONE
    #endif
#endif

``` 

However, I could not see any additional information on the serial console. What am I missing for viewing the MQTT Debug logs?

Also, browsing through the JSON logs on the AWS Console, I encounter the below QoS level. Why is this so, considering that I am setting `publishInfo.qos = IOT_MQTT_QOS_1;` inside the `_publishMessage()` function?

```
{
      "format": "json",
      "payload": {
        "timestamp": "1605296433",
        "deviceId": "esp32DevKitCBoard01",
        "rfidUid": "a5a5a5a5",
        "outletId": "0",
        "Vlt": "0000000.00"
      },
      "qos": 0,
      "timestamp": 1605296436682,
      "topic": "dualOutlet/topic/kWh"
},

```

lundinc2, 2020-11-17:
Hi @dipen-1337lab that change is incorrect, please set the macro I linked in the previous comment to `IOT_LOG_DEBUG`

dipen-1337lab, 2020-11-19:
Namaste @lundinc2,

Thank you for your assistance in order to obtain the Debug logs. Below is the corresponding log observed on the console.

```
2 14 [iot_thread] [INFO ][DEMO][140] ---------STARTING DEMO---------

I (155) BTDM_INIT: BT controller compile version [9891bc6]
3 14 [iot_thread] [INFO ][INIT][140] SDK successfully initialized.
I (165) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
W (175) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (315) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 2
GAP procedure initiated: stop advertising.
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=480 adv_itvl_max=960
I (615) wifi: wifi driver task: 3ffd91cc, prio:23, stack:3584, core=0
I (615) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (615) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (645) wifi: wifi firmware version: c94f3e6
I (645) wifi: config NVS flash: enabled
I (645) wifi: config nano formating: disabled
I (645) wifi: Init dynamic tx buffer num: 32
I (645) wifi: Init data frame dynamic rx buffer num: 32
I (645) wifi: Init management frame dynamic rx buffer num: 32
I (655) wifi: Init management short buffer num: 32
I (665) wifi: Init static rx buffer size: 1600
I (665) wifi: Init static rx buffer num: 10
I (665) wifi: Init dynamic rx buffer num: 32
I (675) wifi: mode : sta (4c:11:ae:df:af:9c)
I (685) WIFI: SYSTEM_EVENT_STA_START
I (805) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (1785) wifi: state: init -> auth (b0)
I (1795) wifi: state: auth -> assoc (0)
I (1795) wifi: state: assoc -> run (10)
I (1905) wifi: connected with tkkfnBYMlkv6TeKM2, channel 1, BW20, bssid = 14:ae:85:e6:45:86
I (1905) wifi: pm start, type: 1

I (1905) WIFI: SYSTEM_EVENT_STA_CONNECTED
I (2635) event: sta ip: 192.168.29.49, mask: 255.255.255.0, gw: 192.168.29.1
I (2635) WIFI: SYSTEM_EVENT_STA_GOT_IP
I (2635) dual_outlet: INFO : Entering a short (20 seconds) delay for the MQTT CONNECT operation to complete before other threads are started.
4 262 [iot_thread] [INFO ][DEMO][2620] Successfully initialized the demo. Network type for the demo: 1
I (2645) dual_outlet: INFO : Thread pvEsp32GetTimeOverSntp() started.
I (2665) dual_outlet: Time is not set yet. Attempting to get the current time over NTP.
I (2675) dual_outlet: Waiting for system time to be set... (1/10)
I (2645) dual_outlet: INFO : Thread pvMqttTask() started. Delaying task...
I (2875) dual_outlet: Notification of a time synchronization event
I (3685) dual_outlet: Current UNIX EPOCH Time is: 1605664799
I (3685) dual_outlet: Equivalent GMT Date Time is: Wed Nov 18 01:59:59 2020
I (12665) dual_outlet: Current UNIX EPOCH Time is: 1605664808
I (12665) dual_outlet: Equivalent GMT Date Time is: Wed Nov 18 02:00:08 2020
I (17685) dual_outlet: INFO : Thread pvMqttTask() re-started.
5 1767 [iot_thread] [INFO ][MQTT][17670] MQTT library successfully initialized.
6 1767 [iot_thread] [INFO ][DEMO][17670] MQTT demo client identifier is esp32DevKitCBoard01 (length 19).
E (18165) PKCS11: failed nvs get file size 4354 0
I (22645) dual_outlet: INFO : Thread pvEsp32AndEnergyMeterCommunication() started.
I (22645) dual_outlet: INFO : Thread pvCleanupTask() started.
I (22665) dual_outlet: Current UNIX EPOCH Time is: 1605664818
I (22665) dual_outlet: Equivalent GMT Date Time is: Wed Nov 18 02:00:18 2020
I (22745) dual_outlet: Outlet Id = 0
E (22745) dual_outlet: Failed to obtain new readings from the Energy Meter.
I (22845) dual_outlet: Outlet Id = 1
E (22845) dual_outlet: Failed to obtain new readings from the Energy Meter.
E (22845) dual_outlet: INFO : Free Heap size = 62760.
E (22845) dual_outlet: INFO : Minimum Ever Free Heap size = 54612.
7 2293 [iot_thread] [INFO ][MQTT][22930] Establishing new MQTT connection.
8 2293 [iot_thread] [DEBUG][MQTT][22930] MQTT PINGREQ packet:
9 2293 [iot_thread] c0 00
10 2293 [iot_thread] [DEBUG][MQTT][22930] (MQTT connection 0x3ffbd9bc) Creating new operation record.
11 2293 [iot_thread] [DEBUG][MQTT][22930] (MQTT connection 0x3ffbd9bc) Reference count changed from 2 to 3.
12 2293 [iot_thread] [INFO ][MQTT][22930] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
[... 172 lines cut ...]
I'm happy to inform that I have resolved this issue. Below is the root cause analysis. Hope it makes sense.

I am running multiple threads as below.
1. OTA Update
2. MQTT Pub-Sub
3. Update Display
4. Fetch new E-meter readings
5. Detect RFID Card swipe
6. Obtain EPOCH timestamp

Except for the **Update Display** thread, all the threads are running periodically by leveraging the `vTaskDelayUntil()` API. The `Update Display` thread is continuously running as a **while(1) loop without any delay**. Also, all threads are of the same priority, that of AWS IoT SDK's `IOT_THREAD_DEFAULT_PRIORITY`.

**It looks like the continuous execution of the `Update Display` thread is somehow causing issues with the normal execution of the MQTT Pub-Sub thread. Making the `Update Display` thread to run periodically every second, leveraging the `vTaskDelayUntil()` API resolves the issue.** Not quite convinced with this behaviour. Can't a thread execute continuously? 

I have tested the application for 30 minutes with all the threads running simultaneously. The `xPortGetFreeHeapSize()` and the 'xPortGetMinimumEverFreeHeapSize()' values are as below, which seem to remain constant over time. Firmware does not restart, as was happening earlier.

```
I (772763) dual_outlet: INFO : Free Heap size = 19560.
I (772763) dual_outlet: INFO : Minimum Ever Free Heap size = 8040.
```

We can close this issue now. Many thanks again for your assistance on this issue! Appreciated.

Thanks | Regards,
Dipen Shah

lundinc2, 2020-11-19:
Hi @dipen-1337lab,

I am very happy to hear this is working for you. 

I am assuming you are only using one core on the ESP32, and if so only one thread can execute at a time.  I hope that answers`Can't a thread execute continuously?`. 

It is possible to pin a task specifically to the second core of the ESP32 which may be worth investigating for your use case.

Thanks,

Carl

dipen-1337lab, 2020-11-20:
> I am assuming you are only using one core on the ESP32, and if so only one thread can execute at a time. I hope that answers Can't a thread execute continuously?

Yes, I'm using only one core on the ESP32. Maybe that is why I encountered this behaviour. My understanding was that the MQTT callback for PUBACK would be independent and, get called once the PUBACK is received. It would wake up the MQTT thread and process the PUBACK. Nevertheless, one thing's for sure, I still need to explore and understand the MQTT functionality.

> It is possible to pin a task specifically to the second core of the ESP32 which may be worth investigating for your use case.

Thank you for suggesting this. Yes, I'm aware. Will keep this in mind to leverage the second core, if need be in the future. 

Thanks | Regards,
Dipen

closed 2020-11-20 by dipen-1337lab