aws/amazon-freertos #2750: How to setup OTA using the new coreMQTT

AbdeenM, 2020-11-15:
Hello,

Am using ESP32 DevKitC and i have a custom application running.
Currently i migrated from the old `iot_mqtt` to `core_mqtt`, previously i had a main thread task that ran something similar to the shadow demo and i was sharing the mqtt connection to the OTA agent, hence having them both running simultaneously and it was working.
Now that am using the `core_mqtt` i noticed that even the OTA demo uses the old iot_mqtt connection and not the new core_mqtt, now my question is how can i run both my custom shadow demo version and ota demo version simultaneously.

The OTA agent seems to run independently really, only thing i cant figure out is what to set for `xOTAConnectionCtx.pvControlClient` as previously it was the shared mqtt connection but now with the core_mqtt passing the `MQTTContext_t` does not work.

Thanks

MrMelon1232, 2020-11-16:
I had asked a similar question here at the issue #2747. Hopefully this answers your question.

lundinc2, 2020-11-16:
Hi @AbdeenM,

As @MrMelon1232 has mentioned, we are working on OTA in a separate repository. When we integrate it to Amazon FreeRTOS we will include an example that uses coreMQTT.

Thanks,

Carl

AbdeenM, 2020-11-16:
Hi @MrMelon1232 and @lundinc2 !

Thanks for the quick comments.
I should have mentioned that i did in fact come across issue #2747 but i opened this issue wondering if there is a work around for this at the moment?
if there is no way around it would you recommend us to downgrade to freertos 202007.00 until OTA is compatible with coreMQTT?

lundinc2, 2020-11-17:
Hi @AbdeenM,

The old OTA will not work with coreMQTT. Until the new version is released, you can either downgrade as you have said, or use the old MQTT for OTA.

Thanks,

Carl

AbdeenM, 2020-11-17:
Hi @lundinc2 ,

Okay thank you for the closure, closing now.

closed 2020-11-17 by AbdeenM