aws/amazon-freertos #2753: Running device_shadow demo establishes mqtt connection and tls and disconnects after a few seconds later

MrMelon1232, 2020-11-16:
**Describe the bug**
Running the shadow demo fails. Logs indicate that app fails to connect to MQTT broker despite outputting a successful response when trying to establish mqtt connection.

**System information**
-ESP32
-Visual Studio Code
- Linux
- Commit #2742 
- Project device_shadow demo

**Expected behavior**
Subscribe to shadow topics and update shadows if delta has been updated.

**Screenshots or console output**
2 4 [iot_thread] [INFO ][DEMO][40] ---------STARTING DEMO---------

I (82) BTDM_INIT: BT controller compile version [0fff744]
3 4 [iot_thread] [INFO ][INIT][40] SDK successfully initialized.
I (92) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
W (102) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (312) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 2
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=187 adv_itvl_max=375
I (632) wifi:wifi driver task: 3ffd9908, prio:23, stack:3584, core=0
I (632) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (632) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (662) wifi:wifi firmware version: 1afdae0
I (662) wifi:config NVS flash: enabled
I (662) wifi:config nano formating: disabled
I (662) wifi:Init dynamic tx buffer num: 32
I (662) wifi:Init data frame dynamic rx buffer num: 32
I (662) wifi:Init management frame dynamic rx buffer num: 32
I (672) wifi:Init management short buffer num: 32
I (672) wifi:Init static rx buffer size: 1600
I (682) wifi:Init static rx buffer num: 10
I (682) wifi:Init dynamic rx buffer num: 32
I (692) wifi:mode : sta (24:6f:28:36:60:e4)
I (702) WIFI: SYSTEM_EVENT_STA_START
I (3352) wifi:new:<>, old:<>, ap:<>, sta:<>, prof:1
I (4342) wifi:state: init -> auth (b0)
I (4382) wifi:state: auth -> assoc (0)
I (4382) wifi:state: assoc -> run (10)
I (4842) wifi:connected with SSID, aid = , channel , , bssid = BSSID
I (4842) wifi:security type:, phy: , rssi: 
I (4842) wifi:pm start, type: 1

I (4842) WIFI: SYSTEM_EVENT_STA_CONNECTED
I (5012) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (9642) event: sta ip: , mask: , gw: 
I (9642) WIFI: SYSTEM_EVENT_STA_GOT_IP
4 960 [eventTask] [INFO ][DEMO][9600] Connected to WiFi access point, ip address: .
5 960 [iot_thread] [INFO ][DEMO][9600] Successfully initialized the demo. Network type for the demo: 1
6 960 [iot_thread] [INFO] [MQTT] [mqtt_demo_helpers.c:272] 7 960 [iot_thread] Create a TCP connection to ENDPOINT[iot_thread] 
E (14472) PKCS11: failed nvs get file size 4354 0
9 2371 [iot_thread] [INFO] [MQTT] [core_mqtt.c:855] 10 2371 [iot_thread] Packet received. ReceivedBytes=2.11 2371 [iot_thread] 
12 2371 [iot_thread] [INFO] [MQTT] [core_mqtt_serializer.c:970] 13 2371 [iot_thread] CONNACK session present bit not set.14 2371 [iot_thread] 
15 2371 [iot_thread] [INFO] [MQTT] [core_mqtt_serializer.c:912] 16 2373 [iot_thread] Connection accepted.17 2373 [iot_thread] 
18 2373 [iot_thread] [INFO] [MQTT] [core_mqtt.c:1532] 19 2373 [iot_thread] Received MQTT CONNACK successfully from broker.20 2373 [iot_thread] 
21 2373 [iot_thread] [INFO] [MQTT] [core_mqtt.c:1798] 22 2373 [iot_thread] MQTT connection established with the broker.23 2373 [iot_thread] 
24 2373 [iot_thread] [INFO] [MQTT] [mqtt_demo_helpers.c:539] 25 2373 [iot_thread] MQTT connection successfully established with broker.

26 2373 [iot_thread] 
27 2373 [iot_thread] [INFO] [MQTT] [mqtt_demo_helpers.c:567] 28 2373 [iot_thread] A clean MQTT connection is established. Cleaning up all the stored outgoing publishes.

29 2373 [iot_thread] 
30 2373 [iot_thread] [ERROR] [ShadowDemo] [shadow_demo_main.c:592] 31 2373 [iot_thread] Failed to connect to MQTT broker.32 2373 [iot_thread] 
33 2373 [iot_thread] [INFO ][DEMO][23730] memory_metrics::freertos_heap::before::bytes::202308
34 2373 [iot_thread] [INFO ][DEMO][23730] memory_metrics::freertos_heap::after::bytes::78384
35 2373 [iot_thread] [INFO ][DEMO][23730] memory_metrics::demo_task_stack::before::bytes::5848
36 2373 [iot_thread] [INFO ][DEMO][23730] memory_metrics::demo_task_stack::after::bytes::1960
I (24772) wifi:state: run -> init (0)
I (24772) wifi:pm stop, total sleep time: 12145663 us / 19926953 us

I (24772) wifi:new:<11,0>, old:<11,2>, ap:<255,255>, sta:<11,2>, prof:1
I (24782) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 8
E (24782) wifi:wifi deinit internal: wifi not stop
37 2473 [iot_thread] [ERROR][DEMO][24730] Error running demo.
38 2474 [eventTask] [INFO ][DEMO][24740] Disconnected from WiFi access point, reason code: 0.
I (24792) wifi:flush txq
I (24802) wifi:stop sw txq
I (24802) wifi:lmac stop hw txq
I (24812) wifi:Deinit lldesc rx mblock:10
GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
39 2479 [iot_thread] [INFO ][INIT][24790] SDK cleanup done.
40 2479 [iot_thread] [INFO ][DEMO][24790] -------DEMO FINISHED-------

**To reproduce**
Steps to reproduce the behavior:
-Set credentials in demo/include folder
-Run device_shadow demo after configuring aws_demos_config.

Thank you!

lundinc2, 2020-11-16:
Hi @MrMelon1232,

If there is an issue with your credentials / policies you will get disconnected at the end of the TLS handshake. Please review that you have configured the credentials properly. A useful way to test your configuration is to use the `mosquitto` commandline tool to publish messages, using the same credentials to AWS IoT Core. Should you need help with this please let me know.

Also, I recommend you work on a tagged release, as the master branch is under active development.

Thanks,

Carl

povi2946, 2020-11-17:
I have the same problem. Other demos work just fine but shadow comes with errors.

lundinc2, 2020-11-17:
Hi @povi2946,

Are you also using the ESP32? What commit are you working on?

Thanks,

Carl

MrMelon1232, 2020-11-18:
Hi @lundinc2,

Once again, thank you for the quick response. I had the necessary policies to run the previous shadows demo and was also able to successfully run other demos. Therefore, I do not believe that the problem is about credentials.

Thanks

lundinc2, 2020-11-18:
Hi @MrMelon1232 and @povi2946,

Sorry for the delay. I have reproduced and opened a pull request that should resolve it. I will post the pull request in here once I have confirmed that the fix is correct.

Thanks,

Carl

lundinc2, 2020-11-19:
Hello,

This has been fixed by https://github.com/aws/amazon-freertos/pull/2764. Please re-open this issue should you have any further trouble.

Thanks,

Carl

closed 2020-11-19 by lundinc2