aws/amazon-freertos #2762: Design intent for BootImageDescriptor_t and BOOTImageHeader_t ?

apullin, 2020-11-18:
**Is your feature request related to a problem? Please describe.**  
This is a straightforward request for clarification on design intent. Code is unlikely to be affected.

**Describe the solution you would like.**  
In early preview versions of the OTA & joblib feature stack, the `BOOTImageHeader_t` has much more reserved space (512B) and the `BOOTImageDescriptor_t` was structured slightly differently.

It looks as though both structure definitions are different in the current codebase, at least for the PIC32 example bootloader port, with `BOOTImageHeader_t` being much smaller.

Are these considered a final design now, format not to be modified?
Are these definitions considered a "degree of freedom" for end-user ports, or should they conform to the format in the example?

**Additional context**  
None.

dan4thewin, 2020-11-24:
Hi, @apullin .  I'll endeavor to find someone to answer your question.

dan4thewin, 2020-11-24:
Hi, @apullin .  The feedback I've received is that these structures are 2 years old.  We're looking at:
https://github.com/aws/amazon-freertos/blob/145dfbe7c9a9de16ba3c86d996e1b8edf3e41dff/vendors/microchip/boards/curiosity_pic32mzef/bootloader/bootloader/include/aws_boot_loader.h#L117-L144

From reading the description, an OTA image generation script is mentioned.  I would imagine, if you need to change the structs, then there would need to be corresponding changes during OTA image generation.

yanjos-dev, 2021-03-11:
Hello @apullin, I'm going to close this issue due to inactivity. Please reopen this issue if you have any further questions.

closed 2021-03-11 by yanjos-dev