aws/amazon-freertos #2769: FreeRTOS failed/force close while send data to greengrass

monhack, 2020-11-19:
Hallo @everyone,

I've managed to run esp32 (amazon freertos) or send data with the local network using shadow. I have done all the instructions in the developer guide and it works, freertos can send data to greengrass with local network. One day when I was running the program, Freertos had successfully connected with Greengrass and got a connection from the internet. But it doesn't send anything even outright error. I did not make any changes suddenly an error. Please help me

**System information**
- ESP32 WROOM32D
- Visual Studio Code
- Operating System : Linux
- Version of Amazon FreeRTOS : 202007.00 release
- Project : MQTTGreengrass|FreeRTOS

**Screenshots or console output**
```
$ python .vendors/espressif/esp-idf/tools/idf.py monitor -p /dev/ttyUSB0 -B directory
.........
Setting IDF_PATH environment variable: /home/monhack/Documents/Magang/freertos/ESP32-GG-1/vendors/espressif/esp-idf
Checking Python dependencies...
Python requirements from /home/monhack/Documents/Magang/freertos/ESP32-GG-1/vendors/espressif/esp-idf/requirements.txt are satisfied.
Running idf_monitor in directory /home/monhack/Documents/Magang/freertos/ESP32-GG-1
Executing "/usr/bin/python /home/monhack/Documents/Magang/freertos/ESP32-GG-1/vendors/espressif/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 /home/monhack/Documents/Magang/freertos/ESP32-GG-1/esp32-GG-1/aws_demos.elf -m '/usr/bin/python' '/home/monhack/Documents/Magang/freertos/ESP32-GG-1/vendors/espressif/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7228
load:0x40078000,len:12652
ho 0 tail 12 room 4
load:0x40080400,len:6708
entry 0x40080778
I (73) boot: Chip Revision: 1
I (73) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v3.3.2-240-g784768652 2nd stage bootloader
I (41) boot: compile time 11:22:39
I (41) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (51) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 4MB
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (77) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (85) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (92) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (100) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (107) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (115) boot: End of partition table
I (119) boot: ota rollback check done
I (123) boot_comm: chip revision: 1, min. application chip revision: 0
I (131) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x2f8c8 (194760) map
I (208) esp_image: segment 1: paddr=0x0004f8f0 vaddr=0x3ffbdb60 size=0x00720 (  1824) load
I (209) esp_image: segment 2: paddr=0x00050018 vaddr=0x400d0018 size=0xbbdc8 (769480) map
0x400d0018: _flash_cache_start at ??:?

I (485) esp_image: segment 3: paddr=0x0010bde8 vaddr=0x3ffbe280 size=0x03140 ( 12608) load
I (490) esp_image: segment 4: paddr=0x0010ef30 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/monhack/Documents/Magang/freertos/ESP32-GG-1/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/xtensa_vectors.S:1713

I (492) esp_image: segment 5: paddr=0x0010f338 vaddr=0x40080400 size=0x1bc08 (113672) load
I (564) boot: Loaded app from partition at offset 0x20000
I (564) boot: Disabling RNG early entropy source...
I (564) cpu_start: Pro cpu up.
I (568) cpu_start: Application information:
I (573) cpu_start: Project name:     esp-idf
I (578) cpu_start: App version:      202011.00-64-g4001598af
I (584) cpu_start: Compile time:     Nov 16 2020 11:21:13
[... 76 lines cut ...]
21 3469 [iot_thread] [INFO ][MQTT][34690] Network connection closed on error.
22 3470 [iot_thread] [INFO ][MQTT][34700] (MQTT connection 0x3ffb5e38) Network connection destroyed.
23 3470 [iot_thread] [ERROR][DEMO][34700] Could not connect to the Broker. Error TIMEOUT.
24 3470 [iot_thread] Heap low watermark: 101716. Stack high watermark: 2824.
25 3470 [iot_thread] [INFO ][MQTT][34700] MQTT library cleanup done.
26 3470 [iot_thread] [INFO ][DEMO][34700] Cleaned up MQTT library.
27 3470 [iot_thread] [INFO ][DEMO][34700] memory_metrics::freertos_heap::before::bytes::193636
28 3470 [iot_thread] [INFO ][DEMO][34700] memory_metrics::freertos_heap::after::bytes::101716
29 3470 [iot_thread] [INFO ][DEMO][34700] memory_metrics::demo_task_stack::before::bytes::8920
30 3470 [iot_thread] [INFO ][DEMO][34700] memory_metrics::demo_task_stack::after::bytes::2824
I (35723) wifi:state: run -> init (0)
I (35723) wifi:pm stop, total sleep time: 30110241 us / 33892187 us

I (35723) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (35723) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 8
E (35733) wifi:wifi deinit internal: wifi not stop
31 3570 [iot_thread] [ERROR][DEMO][35700] Error running demo.
32 3571 [eventTask] [INFO ][DEMO][35710] Disconnected from WiFi access point, reason code: 0.
I (35753) wifi:flush txq
I (35753) wifi:stop sw txq
I (35753) wifi:lmac stop hw txq
I (35753) wifi:Deinit lldesc rx mblock:10
33 3575 [iot_thread] [INFO ][INIT][35750] SDK cleanup done.
34 3575 [iot_thread] [INFO ][DEMO][35750] -------DEMO FINISHED-------
```

dan4thewin, 2020-11-20:
Hi, @monhack .  Above, I see this in the log, "Failed to establish new MQTT connection, error TIMEOUT."  It might be worth checking that greengrassd is up and listening on the configured MQTT port (default 8883), or perhaps restarting the daemon.  It might also be valuable to take a packet capture of traffic on the port.

monhack, 2020-11-20:
Thanks, @dan4thewin.

I have done the steps that you provide. But it still doesn't work, even a new error appears like this:

```
Setting IDF_PATH environment variable: /home/monhack/Documents/Magang/freertos/ESP32-GG-2/vendors/espressif/esp-idf
Checking Python dependencies...
Python requirements from /home/monhack/Documents/Magang/freertos/ESP32-GG-2/vendors/espressif/esp-idf/requirements.txt are satisfied.
Running idf_monitor in directory /home/monhack/Documents/Magang/freertos/ESP32-GG-2
Executing "/usr/bin/python /home/monhack/Documents/Magang/freertos/ESP32-GG-2/vendors/espressif/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 /home/monhack/Documents/Magang/freertos/ESP32-GG-2/esp32-GG-2/aws_demos.elf -m '/usr/bin/python' '/home/monhack/Documents/Magang/freertos/ESP32-GG-2/vendors/espressif/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7228
load:0x40078000,len:12652
ho 0 tail 12 room 4
load:0x40080400,len:6708
entry 0x40080778
I (72) boot: Chip Revision: 1
I (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot: ESP-IDF v3.3.2-239-g29889b575 2nd stage bootloader
I (40) boot: compile time 09:41:04
I (40) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 40MHz
I (50) boot: SPI Mode       : DIO
I (54) boot: SPI Flash Size : 4MB
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (76) boot:  1 otadata          OTA data         01 00 00016000 00002000
I (84) boot:  2 phy_init         RF data          01 01 00018000 00001000
I (91) boot:  3 ota_0            OTA app          00 10 00020000 00177000
I (99) boot:  4 ota_1            OTA app          00 11 001a0000 00177000
I (106) boot:  5 storage          WiFi data        01 02 00317000 00010000
I (114) boot: End of partition table
I (118) boot: ota rollback check done
I (122) boot_comm: chip revision: 1, min. application chip revision: 0
I (129) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x2b6b0 (177840) map
I (201) esp_image: segment 1: paddr=0x0004b6d8 vaddr=0x3ffbdb60 size=0x03814 ( 14356) load
I (207) esp_image: segment 2: paddr=0x0004eef4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/monhack/Documents/Magang/freertos/ESP32-GG-2/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1713

I (208) esp_image: segment 3: paddr=0x0004f2fc vaddr=0x40080400 size=0x00d14 (  3348) load
I (218) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xb5678 (743032) map
0x400d0018: _flash_cache_start at ??:?

I (486) esp_image: segment 5: paddr=0x00105698 vaddr=0x40081114 size=0x1a470 (107632) load
0x40081114: heap_caps_malloc at /home/monhack/Documents/Magang/freertos/ESP32-GG-2/vendors/espressif/esp-idf/components/heap/heap_caps.c:232

I (547) boot: Loaded app from partition at offset 0x20000
I (547) boot: Disabling RNG early entropy source...
I (548) cpu_start: Pro cpu up.
I (551) cpu_start: Application information:
I (556) cpu_start: Project name:     esp-idf
I (561) cpu_start: App version:      202007.00-200-g178f6f348
I (567) cpu_start: Compile time:     Nov 18 2020 09:40:18
I (573) cpu_start: ELF file SHA256:  fed37674afb66404...
I (579) cpu_start: ESP-IDF:          v3.3.2-239-g29889b575
I (586) cpu_start: Single core mode
I (590) heap_init: Initializing. RAM available for dynamic allocation:
I (597) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (603) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (609) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (615) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
[... 73 lines cut ...]
27 3799 [iot_thread] Socket closed.
28 3799 [iot_thread] Stack high watermark for discovery helper task: 4328.
29 3799 [iot_thread] GGD - Can't connect to greengrass Core
30 3799 [iot_thread] [ERROR][DEMO][37990] Auto-connect: Failed to retrieve Greengrass address and certificate.
31 3799 [iot_thread] [INFO ][MQTT][37990] MQTT library cleanup done.
32 3799 [iot_thread] [INFO ][DEMO][37990] Cleaned up MQTT library.
33 3799 [iot_thread] [INFO ][DEMO][37990] memory_metrics::freertos_heap::before::bytes::193948
34 3799 [iot_thread] [INFO ][DEMO][37990] memory_metrics::freertos_heap::after::bytes::97340
35 3799 [iot_thread] [INFO ][DEMO][37990] memory_metrics::demo_task_stack::before::bytes::8920
36 3799 [iot_thread] [INFO ][DEMO][37990] memory_metrics::demo_task_stack::after::bytes::4328
I (39330) wifi:state: run -> init (0)
I (39330) wifi:pm stop, total sleep time: 33147138 us / 37128722 us

I (39330) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (39340) WIFI: SYSTEM_EVENT_STA_DISCONNECTED: 8
E (39340) wifi:wifi deinit internal: wifi not stop
37 3899 [iot_thread] [ERROR][DEMO][38990] Error running demo.
I (39350) wifi:flush txq
I (39350) wifi:stop sw txq
I (39350) wifi:lmac stop hw txq
I (39360) wifi:Deinit lldesc rx mblock:10
38 3903 [iot_thread] [INFO ][INIT][39030] SDK cleanup done.
39 3903 [iot_thread] [INFO ][DEMO][39030] -------DEMO FINISHED-------
```
does the power supply affect the performance of freertos (esp32)?

dan4thewin, 2020-11-24:
Hi, @monhack .  The power supply would not be my first guess as to the issue.  A packet capture on the greengrass might shed light.  Also, for the sake of comparison, you could try the MQTT mutual auth demo using the same client certificate and key.  This would allow us to rule out any problems with the certificate and permissions, should it work.

monhack, 2021-03-09:
Thanks @dan4thewin.

Finally solved, i want to closed this issue

Best regard
Fikri Al Wahid

closed 2021-03-09 by monhack