aws/amazon-freertos #2774: Handshake failed with error code X509 Certificate verification failed coreMQTT happened at version 202011

eelaix, 2020-11-20:
**Describe the bug**
when update from version 2020.07 to 2020.11, 
Handshake failed with error code X509 Certificate verification failed happend
coreMQTT

All Code Works OK BEFORE VERSION 202011 from git source

**System information**
- stm32f4
- ac6
- macos
- newest git source and download from amazonaws.com
- when upgrade the newst coreMQTT
- run demo application: CONFIG_CORE_MQTT_CONNECTION_SHARING_DEMO_ENABLED

**Expected behavior**
34 7014 [iot_thread] [INFO] [MQTTDemo] [mqtt_demo_connection_sharing.c:958] 
35 7021 [iot_thread] Creating a TLS connection to aaaaaaaamyqnami.ats.iot.xxxxxxxxxxxxxxxxxxxx:8883.
36 7030 [iot_thread] 
37 8200 [iot_thread] ERROR: Handshake failed with error code X509 - Certificate verification failed, e.g. CRL, CA or signature check failed : <No-Low-Level-Code> 
38 8214 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:323] 
39 8222 [iot_thread] Failed to establish new connection. secureSocketStatus=-1002.
40 8229 [iot_thread] 
41 8231 [iot_thread] [WARN] [MQTTDemo] [mqtt_demo_connection_sharing.c:966] 
42 8238 [iot_thread] Connection to the broker failed. Retrying connection with backoff and jitter.
43 8246 [iot_thread] 
44 8300 [EMAC] Network buffers: 6 lowest 4
45 9248 [iot_thread] [INFO] [MQTTDemo] [mqtt_demo_connection_sharing.c:958] 

**Screenshots or console output**

**To reproduce**

**Code to reproduce the bug**
**Additional context**

dan4thewin, 2020-11-20:
Hi, @eelaix .  Could you compare your mqtt_demo_mutual_auth_config.h in the working version to the non-working version?  I'm wondering about the root CA define.  If those files are the same, perhaps comparing the remaining configs would help.

eelaix, 2020-11-20:
> Hi, @eelaix . Could you compare your mqtt_demo_mutual_auth_config.h in the working version to the non-working version? I'm wondering about the root CA define. If those files are the same, perhaps comparing the remaining configs would help.

I think This file a configure setting for one of demo application, It MAY NOT affect coreMQTT.
I have remove this file form my project early, so This file is not exists in my project.
I don't think this file a the reason.

eelaix, 2020-11-20:
MORE logs showed HERE:
========================================================
63 19137 [iot_thread] [INFO ][DEMO][19137] Successfully initialized the demo. Network type for the demo: 4
64 19147 [iot_thread] [INFO] [MQTTDemo] [mqtt_demo_connection_sharing.c:958] 65 19154 [iot_thread] Creating a TLS connection to a1mnz5pmyqnami.ats.iot.cn-north-1.amazonaws.com.cn:8883.66 19163 [iot_thread] 
67 19165 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:4755] 68 19171 [iot_thread] Successfully generated 4 random bytes.69 19177 [iot_thread] 
70 19179 [IP-task] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:4755] 71 19184 [IP-task] Successfully generated 4 random bytes.72 19189 [IP-task] 
73 19302 [IP-task] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:4755] 74 19307 [IP-task] Successfully generated 4 random bytes.75 19312 [IP-task] 
76 19314 [IP-task] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:4755] 77 19320 [IP-task] Successfully generated 8 random bytes.78 19325 [IP-task] 
79 19423 [iot_thread] [WARN] [PKCS11] [core_pkcs11_mbedtls.c:1338] 80 19428 [iot_thread] Failed to initialize PKCS #11. PKCS #11 was already initialized.81 19436 [iot_thread] 
82 19438 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:1560] 83 19444 [iot_thread] Successfully Returned a PKCS #11 slot with ID 1 with a count of 1.84 19452 [iot_thread] 
85 19454 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:1798] 86 19460 [iot_thread] Assigned a 0x2 Type Session.87 19464 [iot_thread] 
88 19466 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:1809] 89 19472 [iot_thread] Assigned Mechanisms to no operation in progress.90 19479 [iot_thread] 
91 19481 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:1825] 92 19487 [iot_thread] Current session count at 193 19491 [iot_thread] 
94 19493 [iot_thread] [WARN] [PKCS11] [core_pkcs11_mbedtls.c:1913] 95 19499 [iot_thread] C_Login is not implemented.96 19503 [iot_thread] 
97 19506 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:4755] 98 19512 [iot_thread] Successfully generated 32 random bytes.99 19517 [iot_thread] 
100 19521 [iot_thread] [WARN] [PKCS11] [core_pkcs11_mbedtls.c:1913] 101 19527 [iot_thread] C_Login is not implemented.102 19531 [iot_thread] 
103 19533 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:3006] 104 19539 [iot_thread] Search parameters other than label are ignored.105 19546 [iot_thread] 
106 19548 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:3107] 107 19554 [iot_thread] Could not find the object handle in the list. Trying to search PKCS #11 PAL for object.108 19564 [iot_thread] 
109 19566 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:3134] 110 19572 [iot_thread] Found object in PAL. Adding object handle to list.111 19578 [iot_thread] 
112 19580 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:913] 113 19586 [iot_thread] Found object in list by handle.114 19591 [iot_thread] 
115 19594 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:2671] 116 19600 [iot_thread] Object was a private key.117 19604 [iot_thread] 
118 19606 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:3006] 119 19612 [iot_thread] Search parameters other than label are ignored.120 19618 [iot_thread] 
121 19621 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:3107] 122 19627 [iot_thread] Could not find the object handle in the list. Trying to search PKCS #11 PAL for object.123 19636 [iot_thread] 
124 19638 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:3134] 125 19645 [iot_thread] Found object in PAL. Adding object handle to list.126 19651 [iot_thread] 
127 19653 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:913] 128 19659 [iot_thread] Found object in list by handle.129 19664 [iot_thread] 
130 19666 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:2687] 131 19672 [iot_thread] Object was a certificate.132 19677 [iot_thread] 
133 19679 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:913] 134 19685 [iot_thread] Found object in list by handle.135 19690 [iot_thread] 
136 19692 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:2687] 137 19698 [iot_thread] Object was a certificate.138 19702 [iot_thread] 
139 19705 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:3006] 140 19711 [iot_thread] Search parameters other than label are ignored.141 19717 [iot_thread] 
142 19719 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:3107] 143 19725 [iot_thread] Could not find the object handle in the list. Trying to search PKCS #11 PAL for object.144 19735 [iot_thread] 
145 20819 [iot_thread] [INFO] [PKCS11] [core_pkcs11_mbedtls.c:1880] 146 20825 [iot_thread] Successfully closed PKCS #11 session.147 20830 [iot_thread] 
148 20832 [iot_thread] ERROR: Handshake failed with error code X509 - Certificate verification failed, e.g. CRL, CA or signature check failed : <No-Low-Level-Code> 
149 20847 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:323] 150 20855 [iot_thread] Failed to establish new connection. secureSocketStatus=-1002.151 20862 [iot_thread] 
152 20865 [iot_thread] [WARN] [MQTTDemo] [mqtt_demo_connection_sharing.c:966] 153 20871 [iot_thread] Connection to the broker failed. Retrying connection with backoff and jitter.154 20880 [iot_thread] 
155 20882 [iot_thread] [INFO] [MQTTDemo] [mqtt_demo_connection_sharing.c:958] 156 20889 [iot_thread] Creating a TLS connection to a1mnz5pmyqnami.ats.iot.cn-north-1.amazonaws.com.cn:8883.157 20899 [iot_thread] 
[... cut ...]
420 30804 [iot_thread] [INFO] [MQTTDemo] [mqtt_demo_connection_sharing.c:958] 421 30810 [iot_thread] Creating a TLS connection to a1mnz5pmyqnami.ats.iot.cn-north-1.amazonaws.com.cn:8883.422 30820 [iot_thread] 
423 30822 [iot_thread] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:4755] 424 30828 [iot_thread] Successfully generated 4 random bytes.425 30834 [iot_thread] 
426 30836 [IP-task] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:4755] 427 30842 [IP-task] Successfully generated 4 random bytes.428 30847 [IP-task] 
429 30850 [IP-task] [DEBUG] [PKCS11] [core_pkcs11_mbedtls.c:4755] 430 30856 [IP-task] Successfully generated 4 random bytes.431 30861 [IP-task]

dan4thewin, 2020-11-20:
Hi, I'm sorry I referenced the wrong file.  Can you compare your `aws_clientcredential_keys.h` file in the working version to the non-working version?  Also, do you have logs available from your working version?

eelaix, 2020-11-21:
> Hi, I'm sorry I referenced the wrong file. Can you compare your `aws_clientcredential_keys.h` file in the working version to the non-working version? Also, do you have logs available from your working version?

This file never changed in my project and i will check all files from remove git server, my "amazon-freertos" directory will keep sync to git server every day.

I don't think this file is the problem, because this file never changed in my project.
my devcert and devprivatekey is provided by "vendor/my/boards/myboard/ports/core_pkcs11_pal.c"
and devcert/privatekey(s) are saved in I2C memory.

I am wonder what changed in source coreMQTT or corePKCS11 at version of 2020.11

eelaix, 2020-11-21:
When i execute secure_socket_tests with TLS (FULL_TCP_TESTS) i got these Logs,  anybody can help me on this?
===============================================================================
---------STARTING TESTS---------
TEST(Full_TCP, AFQP_SOCKETS_CloseInvalidParams) PASS
TEST(Full_TCP, AFQP_SOCKETS_CloseWithoutReceiving)/dir/amazon-freertos/libraries/abstractions/secure_sockets/test/iot_test_tcp.c:1199::FAIL: Expected 0 Was -116. Failed to connect
TEST(Full_TCP, AFQP_SOCKETS_ShutdownInvalidParams) PASS
TEST(Full_TCP, AFQP_SOCKETS_ShutdownWithoutReceiving)/dir/amazon-freertos/libraries/abstractions/secure_sockets/test/iot_test_tcp.c:1275::FAIL: Expected 0 Was -116. Failed to connect
TEST(Full_TCP, AFQP_SOCKETS_Recv_On_Unconnected_Socket)/dir/amazon-freertos/libraries/abstractions/secure_sockets/test/iot_test_tcp.c:1325::FAIL: Expected 0 Was -116. Failed to connect
TEST(Full_TCP, AFQP_SOCKETS_Connect_InvalidParams) PASS
TEST(Full_TCP, AFQP_SOCKETS_Connect_InvalidAddressLength)/dir/amazon-freertos/libraries/abstractions/secure_sockets/test/iot_test_tcp.c:2309::FAIL: Expected 0 Was -116. Socket connect failed.
TEST(Full_TCP, AFQP_SOCKETS_Socket_TCP) PASS
TEST(Full_TCP, AFQP_SOCKETS_SetSockOpt_RCVTIMEO)/dir/amazon-freertos/libraries/abstractions/secure_sockets/test/iot_test_tcp.c:1420::FAIL: Expected 0 Was -116. Failed to connect
TEST(Full_TCP, AFQP_SOCKETS_NonBlocking_Test)/dir/amazon-freertos/libraries/abstractions/secure_sockets/test/iot_test_tcp.c:1501::FAIL: Expected 0 Was -116. Failed to connect
TEST(Full_TCP, AFQP_SOCKETS_SetSockOpt_InvalidParams)/dir/amazon-freertos/libraries/abstractions/secure_sockets/test/iot_test_tcp.c:1589::FAIL: Expected 0 Was -116. Failed to connect
TEST(Full_TCP, AFQP_SOCKETS_Shutdown)/dir/amazon-freertos/libraries/abstractions/secure_sockets/test/iot_test_tcp.c:1683::FAIL: Expected 0 Was -116. Failed to connect
TEST(Full_TCP, AFQP_SOCKETS_Close)/dir/amazon-freertos/libraries/abstractions/secure_sockets/test/iot_test_tcp.c:1789::FAIL: Expected 0 Was -116. Failed to connect
TEST(Full_TCP, AFQP_SOCKETS_Recv_ByteByByte)/dir/amazon-freertos/libraries/abstractions/secure_sockets/test/iot_test_tcp.c:1855::FAIL: Expected 0 Was -116. Failed to connect
TEST(Full_TCP, AFQP_SOCKETS_SendRecv_VaryLength)/dir/amazon-freertos/libraries/abstractions/secure_sockets/test/iot_test_tcp.c:1934::FAIL: Expected 0 Was -116. Failed to connect
TEST(Full_TCP, AFQP_SOCKETS_Socket_ConcurrentCount) PASS
TEST(Full_TCP, AFQP_SOCKETS_Socket_InvalidInputParams) PASS
TEST(Full_TCP, AFQP_SOCKETS_Send_Invalid)/dir/amazon-freertos/libraries/abstractions/secure_sockets/test/iot_test_tcp.c:2366::FAIL: Expected 0 Was -116. Failed to connect
TEST(Full_TCP, AFQP_SOCKETS_Recv_Invalid)/dir/amazon-freertos/libraries/abstractions/secure_sockets/test/iot_test_tcp.c:2433::FAIL: Expected 0 Was -116. Failed to connect
TEST(Full_TCP, AFQP_SOCKETS_htons_HappyCase) PASS
TEST(Full_TCP, AFQP_SOCKETS_inet_addr_quick_HappyCase) PASS
TEST(Full_TCP, AFQP_SECURE_SOCKETS_CloseInvalidParams) PASS
13 328750 [TestRunner] ERROR: Failed to parse custom server certificates X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected : <No-Low-Level-Code> 
14 329012 [TestRunner] ERROR: Failed to parse custom server certificates X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected : <No-Low-Level-Code> 
15 329429 [TestRunner] ERROR: Failed to parse custom server certificates X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected : <No-Low-Level-Code> 
16 330165 [TestRunner] ERROR: Failed to parse custom server certificates X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected : <No-Low-Level-Code> 
17 331485 [TestRunner] ERROR: Failed to parse custom server certificates X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected : <No-Low-Level-Code> 
18 334011 [TestRunner] ERROR: Failed to parse custom server certificates X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected : <No-Low-Level-Code> 
19 338926 [TestRunner] ERROR: Failed to parse custom server certificates X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected : <No-Low-Level-Code> 
TEST(Full_TCP, AFQP_SECURE_SOCKETS_CloseWithoutReceiving)/dir/amazon-freertos/libraries/abstractions/secure_sockets/test/iot_test_tcp.c:1199::FAIL: Expected 0 Was -1002. Failed to connect
TEST(Full_TCP, AFQP_SECURE_SOCKETS_ShutdownInvalidParams) PASS
20 339376 [TestRunner] ERROR: Failed to parse custom server certificates X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected : <No-Low-Level-Code> 
21 339635 [TestRunner] ERROR: Failed to parse custom server certificates X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected : <No-Low-Level-Code> 
22 340045 [TestRunner] [ERROR] [PKCS11] [core_pkcs11_mbedtls.c:1817] 23 340051 [TestRunner] Could not open a session. All sessions have been taken. Consider increasing value of pkcs11configMAX_SESSIONS.24 340063 [TestRunner] 
25 340764 [TestRunner] [ERROR] [PKCS11] [core_pkcs11_mbedtls.c:1817] 26 340771 [TestRunner] Could not open a session. All sessions have been taken. Consider increasing value of pkcs11configMAX_SESSIONS.27 340782 [TestRunner] 
28 342091 [TestRunner] [ERROR] [PKCS11] [core_pkcs11_mbedtls.c:1817] 29 342097 [TestRunner] Could not open a session. All sessions have been taken. Consider increasing value of pkcs11configMAX_SESSIONS.30 342109 [TestRunner] 
[... cut ...]
306 489564 [TestRunner] [ERROR] [PKCS11] [core_pkcs11_mbedtls.c:1817] 307 489571 [TestRunner] Could not open a session. All sessions have been taken. Consider increasing value of pkcs11configMAX_SESSIONS.308 489582 [TestRunner] 
309 492085 [TestRunner] [ERROR] [PKCS11] [core_pkcs11_mbedtls.c:1817] 310 492091 [TestRunner] Could not open a session. All sessions have been taken. Consider increasing value of pkcs11configMAX_SESSIONS.311 492103 [TestRunner] 
312 497004 [TestRunner] [ERROR] [PKCS11] [core_pkcs11_mbedtls.c:1817] 313 497011 [TestRunner] Could not open a session. All sessions have been taken. Consider increasing value of pkcs11configMAX_SESSIONS.314 497022

eelaix, 2020-11-21:
WoW!!!   MQTTv4 all tests PASSed, BUT FULL_TCP_TLS fail.    the following logs are from MQTTv4 tests.
WHAT happened??????????????
======================================================================
---------STARTING TESTS---------
TEST(MQTT_Unit_Validate, ValidateConnectInfo) PASS
TEST(MQTT_Unit_Validate, ValidatePublish) PASS
TEST(MQTT_Unit_Validate, ValidateOperation) PASS
TEST(MQTT_Unit_Validate, ValidateSubscriptionList) PASS
TEST(MQTT_Unit_Subscription, ListInsertRemove) PASS
TEST(MQTT_Unit_Subscription, ListFindByTopicFilter) PASS
TEST(MQTT_Unit_Subscription, ListFindByPacket) PASS
TEST(MQTT_Unit_Subscription, SubscriptionRemoveByPacket) PASS
TEST(MQTT_Unit_Subscription, SubscriptionRemoveByTopicFilter) PASS
TEST(MQTT_Unit_Subscription, SubscriptionAddDuplicate) PASS
TEST(MQTT_Unit_Subscription, SubscriptionAddMallocFail) PASS
TEST(MQTT_Unit_Subscription, ProcessPublish) PASS
TEST(MQTT_Unit_Subscription, ProcessPublishMultiple) PASS
TEST(MQTT_Unit_Subscription, SubscriptionReferences) PASS
TEST(MQTT_Unit_Subscription, TopicFilterMatchTrue) PASS
TEST(MQTT_Unit_Subscription, TopicFilterMatchFalse) PASS
TEST(MQTT_Unit_Receive, DecodeRemainingLength) PASS
TEST(MQTT_Unit_Receive, InvalidPacket) PASS
TEST(MQTT_Unit_Receive, ConnackValid) PASS
TEST(MQTT_Unit_Receive, ConnackInvalid) PASS
TEST(MQTT_Unit_Receive, PublishValid) PASS
TEST(MQTT_Unit_Receive, PublishInvalid) PASS
TEST(MQTT_Unit_Receive, PubackValid) PASS
TEST(MQTT_Unit_Receive, PubackInvalid) PASS
TEST(MQTT_Unit_Receive, SubackValid) PASS
TEST(MQTT_Unit_Receive, SubackInvalid) PASS
TEST(MQTT_Unit_Receive, UnsubackValid) PASS
TEST(MQTT_Unit_Receive, UnsubackInvalid) PASS
TEST(MQTT_Unit_Receive, Pingresp) PASS
TEST(MQTT_Unit_API, OperationCreateDestroy) PASS
TEST(MQTT_Unit_API, OperationWaitTimeout) PASS
TEST(MQTT_Unit_API, ConnectParameters) PASS
TEST(MQTT_Unit_API, ConnectMallocFail) PASS
TEST(MQTT_Unit_API, ConnectRestoreSessionMallocFail) PASS
TEST(MQTT_Unit_API, DisconnectMallocFail) PASS
TEST(MQTT_Unit_API, PublishQoS0Parameters) PASS
TEST(MQTT_Unit_API, PublishQoS0MallocFail) PASS
TEST(MQTT_Unit_API, PublishQoS1) PASS
TEST(MQTT_Unit_API, SubscribeUnsubscribeParameters) PASS
TEST(MQTT_Unit_API, SubscribeMallocFail) PASS
TEST(MQTT_Unit_API, UnsubscribeMallocFail) PASS
14 27759 [TestRunner] 
15 29361 [iot_thread] KeepAlivePeriodic 1 of 10 DONE at 29361 ms.
16 30461 [iot_thread] KeepAlivePeriodic 2 of 10 DONE at 30461 ms (+ 1100 ).
17 31561 [iot_thread] KeepAlivePeriodic 3 of 10 DONE at 31561 ms (+ 1100 ).
18 32661 [iot_thread] KeepAlivePeriodic 4 of 10 DONE at 32661 ms (+ 1100 ).
19 33761 [iot_thread] KeepAlivePeriodic 5 of 10 DONE at 33761 ms (+ 1100 ).
20 34861 [iot_thread] KeepAlivePeriodic 6 of 10 DONE at 34861 ms (+ 1100 ).
21 35961 [iot_thread] KeepAlivePeriodic 7 of 10 DONE at 35961 ms (+ 1100 ).
22 37061 [iot_thread] KeepAlivePeriodic 8 of 10 DONE at 37061 ms (+ 1100 ).
23 38161 [iot_thread] KeepAlivePeriodic 9 of 10 DONE at 38161 ms (+ 1100 ).
24 39261 [iot_thread] KeepAlivePeriodic 10 of 10 DONE at 39261 ms (+ 1100 ).
TEST(MQTT_Unit_API, KeepAlivePeriodic) PASS
TEST(MQTT_Unit_API, KeepAliveJobCleanup) PASS
TEST(MQTT_Unit_API, WaitAfterDisconnect) PASS
TEST(MQTT_Unit_Metrics, Generate_MD5Hash) PASS
TEST(MQTT_Unit_Metrics, Get_DeviceIdentifier) PASS
TEST(MQTT_Unit_Metrics, Get_DeviceMetrics) PASS
TEST(MQTT_System, SubscribePublishWaitQoS0) PASS
TEST(MQTT_System, SubscribePublishWaitQoS1) PASS
TEST(MQTT_System, SubscribePublishAsync) PASS
TEST(MQTT_System, LastWillAndTestament) PASS
TEST(MQTT_System, RestorePreviousSession) PASS
TEST(MQTT_System, SubscribeCompleteReentrancy) PASS
TEST(MQTT_System, IncomingPublishReentrancy) PASS

-----------------------
54 Tests 0 Failures 0 Ignored 
OK
-------ALL TESTS FINISHED-------

eelaix, 2020-11-21:
Following logs are SHADOWv4 working perfect!  BUT FULL_TCP_TLS fail, what is wrong
========================================================================
---------STARTING TESTS---------
TEST(Shadow_Unit_Parser, StatusValid) PASS
TEST(Shadow_Unit_Parser, StatusInvalid) PASS
TEST(Shadow_Unit_Parser, JsonValid) PASS
TEST(Shadow_Unit_Parser, JsonInvalid) PASS
TEST(Shadow_Unit_Parser, ErrorDocument) PASS
TEST(Shadow_Unit_Parser, ErrorDocumentInvalid) PASS
TEST(Shadow_Unit_Parser, ThingName) PASS
TEST(Shadow_Unit_API, Init) PASS
TEST(Shadow_Unit_API, OperationInvalidParameters) PASS
TEST(Shadow_Unit_API, DocumentInvalidParameters) PASS
TEST(Shadow_Unit_API, WaitInvalidParameters) PASS
TEST(Shadow_Unit_API, DeleteMallocFail) PASS
TEST(Shadow_Unit_API, GetMallocFail) PASS
TEST(Shadow_Unit_API, UpdateMallocFail) PASS
TEST(Shadow_System, UpdateGetDeleteAsyncQoS0) PASS
TEST(Shadow_System, UpdateGetDeleteAsyncQoS1) PASS
13 32980 [EMAC] Network buffers: 5 lowest 4
TEST(Shadow_System, UpdateGetDeleteBlockingQoS0) PASS
TEST(Shadow_System, UpdateGetDeleteBlockingQoS1) PASS
TEST(Shadow_System, DeltaCallback) PASS
TEST(Shadow_System, UpdatedCallback) PASS
14 51632 [TestRunner] Heap Before: 126024, Heap After: 123192, Diff: 2832 
TEST(Full_MemoryLeak, CheckHeap)/Users/saico/FreeRTOS_full/amazon-freertos/libraries/c_sdk/standard/common/test/iot_memory_leak.c:71::FAIL: Expected 0 Was 2832. Free heap before and after tests was not the same.

-----------------------
21 Tests 1 Failures 0 Ignored 
FAIL

lundinc2, 2020-11-23:
Hi @eelaix,

Have you followed the steps outlined in the porting guide here https://docs.aws.amazon.com/freertos/latest/portingguide/afr-echo-server.html for the Full_TCP test group?

Thanks,

Carl

eelaix, 2020-11-23:
> Hi @eelaix,
> 
> Have you followed the steps outlined in the porting guide here https://docs.aws.amazon.com/freertos/latest/portingguide/afr-echo-server.html for the Full_TCP test group?
> 
> Thanks,
> 
> Carl

yes! i Have followed that step porting guide .

you can see The Demo "CONFIG_CORE_MQTT_CONNECTION_SHARING_DEMO_ENABLED" Shows error first, so it is not the porting guide step error.

==================================================
i have do more and more testing, The following test working good

testrunnerFULL_MQTTv4_ENABLED
testrunnerFULL_SHADOWv4_ENABLED
testrunnerFULL_DEVICE_SHADOW_ENABLED
testrunnerFULL_TCP_ENABLED   (without TLS)

The following part failed:

#define testrunnerFULL_MQTT_AGENT_ENABLED          0
#define testrunnerFULL_TCP_ENABLED                 1
#define testrunnerFULL_SHADOW_ENABLED              0
#define testrunnerFULL_PKCS11_ENABLED              0
CONFIG_CORE_MQTT_CONNECTION_SHARING_DEMO_ENABLED

the following never started:

#define testrunnerFULL_CRYPTO_ENABLED              0
#define testrunnerFULL_TLS_ENABLED                 0

lundinc2, 2020-11-24:
@eelaix,

Are you receiving any output on the echo server? You can get verbose output with `verbose` set to `"true"` in the echo server config.json. Is the echo server on the same network as your device?

Thanks,

Carl

eelaix, 2020-11-24:
> @eelaix,
> 
> Are you receiving any output on the echo server? You can get verbose output with `verbose` set to `"true"` in the echo server config.json. Is the echo server on the same network as your device?
> 
> Thanks,
> 
> Carl

yes i can recieve message in echo server console such as :

2020/11/24 09:37:32 Opening secure TCP server listening to port 8443

2020/11/24 09:44:57 Read 1 bytes.
2020/11/24 09:44:57 Message:
 
2020/11/24 09:44:57 Succesfully echoed back 1 bytes.
2020/11/24 09:45:04 Error remote error: tls: unknown certificate authority while reading data. Expected an EOF to signal end of connection
2020/11/24 09:45:46 Read 1200 bytes.
2020/11/24 09:45:46 Message:
 0 TxRx with of length 1200KLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~0123456789:;<=>
2020/11/24 09:45:46 Succesfully echoed back 1200 bytes.
2020/11/24 09:45:53 Error read tcp 192.168.1.105:8443->192.168.1.100:6300: read: connection reset by peer while reading data. Expected an EOF to signal end of connection
2020/11/24 09:46:31 Read 2 bytes.
2020/11/24 09:46:31 Message:
 0
2020/11/24 09:46:31 Succesfully echoed back 2 bytes.
2020/11/24 09:46:31 Read 7 bytes.
2020/11/24 09:46:31 Message:
 1 TxRx
2020/11/24 09:46:31 Succesfully echoed back 7 bytes.
2020/11/24 09:46:31 Read 8 bytes.
2020/11/24 09:46:31 Message:
 2 TxRx 
2020/11/24 09:46:31 Succesfully echoed back 8 bytes.
2020/11/24 09:46:31 Read 9 bytes.
2020/11/24 09:46:31 Message:
 3 TxRx w
2020/11/24 09:46:31 Succesfully echoed back 9 bytes.
2020/11/24 09:46:31 Read 20 bytes.
2020/11/24 09:46:31 Message:
 4 TxRx with of leng
2020/11/24 09:46:31 Succesfully echoed back 20 bytes.
2020/11/24 09:46:37 Read 1 bytes.
2020/11/24 09:46:37 Message:
 
2020/11/24 09:46:37 Succesfully echoed back 1 bytes.
2020/11/24 09:46:37 Read 1 bytes.
2020/11/24 09:46:37 Message:
 
2020/11/24 09:46:37 Succesfully echoed back 1 bytes.
2020/11/24 09:46:37 Read 1 bytes.
2020/11/24 09:46:37 Message:
 
2020/11/24 09:46:37 Succesfully echoed back 1 bytes.
2020/11/24 09:46:37 Read 1 bytes.
2020/11/24 09:46:37 Message:
 
2020/11/24 09:46:37 Succesfully echoed back 1 bytes.
2020/11/24 09:46:37 Read 1 bytes.
2020/11/24 09:46:37 Message:
 
2020/11/24 09:46:37 Succesfully echoed back 1 bytes.
2020/11/24 09:46:37 Read 1 bytes.
2020/11/24 09:46:37 Message:
 
2020/11/24 09:46:37 Succesfully echoed back 1 bytes.
2020/11/24 09:46:37 Read 1 bytes.
2020/11/24 09:46:37 Message:
 
2020/11/24 09:46:37 Succesfully echoed back 1 bytes.
[... 61 lines cut ...]
2020/11/24 09:46:50 Read 7 bytes.
2020/11/24 09:46:50 Message:
 22 TxR
2020/11/24 09:46:50 Succesfully echoed back 7 bytes.
2020/11/24 09:46:50 Read 7 bytes.
2020/11/24 09:46:50 Message:
 23 TxR
2020/11/24 09:46:50 Succesfully echoed back 7 bytes.
2020/11/24 09:46:50 Read 7 bytes.
2020/11/24 09:46:50 Message:
 24 TxR
2020/11/24 09:46:50 Succesfully echoed back 7 bytes.
2020/11/24 09:46:50 Read 7 bytes.
2020/11/24 09:46:50 Message:
 25 TxR
2020/11/24 09:46:50 Succesfully echoed back 7 bytes.
2020/11/24 09:46:50 Read 7 bytes.
2020/11/24 09:46:50 Message:
 26 TxR
2020/11/24 09:46:50 Succesfully echoed back 7 bytes.
2020/11/24 09:46:50 Read 7 bytes.
2020/11/24 09:46:50 Message:
 27 TxR
2020/11/24 09:46:50 Succesfully echoed back 7 bytes.
............................

eelaix, 2020-11-25:
@lundinc2

is is a bug?   how to fix it?

lundinc2, 2020-11-26:
Hi @eelaix,

I tested this with the Windows simulator and did not have any issue. On Friday I will check with an STML4. Are you running two echo servers at once? The expectation is you have one for TLS and one for non-TLS connections, I hope that helps. 

Is my understanding correct, that now the only issue is with the Full_TCP suite, and the other test groups are working?

Thanks,

Carl

eelaix, 2020-11-26:
> Hi @eelaix,
>
> I tested this with the Windows simulator and did not have any issue. On Friday I will check with an STML4. Are you running two echo servers at once? The expectation is you have one for [...]

Hi @lundinc2 

I execute every test complete compiled and restart the echo server.   every test runs more than one time.
the FULL_TCP_TEST without TLS work perfect.

for above logs shows, i even found the DEMO application and FULL_TCP_TEST (WITH TLS) give me the same issue: "Handshake Failed with error code X509 Certificate verification failed".

Today, i Downgrade to git tag source Version 202007, NOW i had success run demo application and my own application perfect, i will continue to run tests application in few hours.

lundinc2, 2020-11-30:
Hi @eelaix,

I am sorry for the delay. I was able to pass the TLS tests on the STML4. I believe the issue you are facing is the server CA certificate not being correctly configured in https://github.com/aws/amazon-freertos/blob/b2217ded7a33f1b041fc5e630fe19ca2dbcd4068/tests/include/aws_test_tcp.h#L62. It may be worth revisiting the instructions here in case of a mistake https://github.com/aws/amazon-freertos/tree/master/tools/echo_server.

I verified my connection using the below OpenSSL command. You can use this to make sure your credentials are correct, before trying to establish the connection on your device.

`openssl s_client -connect <ip address>:<port> -cert client.pem -key client.key -CAfile server.pem`.

I am really sorry that this has been a pain point for you, I hope this helps unblock you, please let me know if you are unable to connect after following these steps.

eelaix, 2020-12-02:
Hi  @lundinc2 ,
I Checked everything as your words, i am sure my setting/codes is always correct, i have read the document more times, execute the command more times, the same error appears.

lundinc2, 2020-12-02:
Hi @eelaix,

Can you post a pastebin with the output if you have mbed TLS debug enabled? Judging by the logs it seems that you are running mbed TLS, it can be enabled by uncommenting this line https://github.com/aws/amazon-freertos/blob/d6736703410b6bf0a71278f764f580717d52cee5/libraries/3rdparty/mbedtls_config/aws_mbedtls_config.h#L2221.

I do not see the STM32F4 in our device catalog, did you mean to say you are running on the STM32 L4? I will need to order an STM 32F4, but I don't believe the code we are hosting in this repository will run on it.

Thanks,

Carl

eelaix, 2020-12-03:
> Hi @eelaix,
>
> Can you post a pastebin with the output if you have mbed TLS debug enabled? Judging by the logs it seems that you are running mbed TLS, it can be enabled by uncommenting this line [...]

YES！ I am running amazon-freertos VER 202007 on my custom Board STM32F407VGTx+ I2CMemory + RMIIETH +LTE(UART6,X-CUBE-CELLULAR5.2) + SPIsensor(templete,ATT7053...).  Everythings worked perfect in 202007.
I have saved the DevCertificate and PrivateKey in I2C Memory then read and use from "*_pkcs11_pal.c"

I will TEST VER 202011 by your advice later.

timdesi, 2021-01-18:
Hi, looks like i have same issue.

I am testing mqtt_demo_mutual_auth.c demo.

```
51 549 [iot_thread] ERROR: Handshake failed with error code X509 - Certificate verification failed, e.g. CRL, CA or signature check failed : <No-Low-Level-Code> 
52 549 [iot_thread] TLS_Connect fail (0x2700, NULL)
```

I am testing with latest version from git, ESP32 hardware and Windows Simulator, both have same error. 

Also test with version 202011, 202012 and firmware downloaded from AWS IoT console all are with same results.

Try to enable mbedTLS debug option with MBEDTLS_DEBUG_C, unfortunately could not see debug logs.

lundinc2, 2021-03-01:
Hi @timdesi,

Are you using the "ats" endpoint or the regular endpoint? Please see https://aws.amazon.com/blogs/iot/aws-iot-core-ats-endpoints/ for some background. Currently the default demo will use Amazon CA1, which is used on the ats endpoint.

Here is an example of what an ATS endpoint would look like: `{account}-ats.iot.us-west-2.amazonaws.com`.

timdesi, 2021-03-02:
Hi @lundinc2,

I am testing with ATS endpoint (********-ats.iot.eu-central-1.amazonaws.com). 

I have double check certificate and policy attached.  

```
269 [eventTask] [INFO ][DEMO][2690] Connected to WiFi access point, ip address: 192.168.1.224.
5 270 [iot_thread] [INFO ][DEMO][2700] Successfully initialized the demo. Network type for the demo: 1
6 270 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:745] 7 270 [iot_thread] Creating a TLS connection to **********-ats.iot.eu-central-1.amazonaws.com:8883.8 270 [iot_thread] 
GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
E (3164) PKCS11: failed nvs get file size 4354 0
9 301 [iot_thread] ERROR: Handshake failed with error code X509 - Certificate verification failed, e.g. CRL, CA or signature check failed : <No-Low-Level-Code> 
10 301 [iot_thread] TLS_Connect fail (0x2700, NULL)
11 301 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:323] 12 301 [iot_thread] Failed to establish new connection. secureSocketStatus=-1.13 301 [iot_thread] 
14 303 [iot_thread] [WARN] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:753] 15 303 [iot_thread] Connection to the broker failed. Attempting connection retry after backoff delay.16 303 [iot_thread] 
17 347 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:689] 18 347 [iot_thread] Retry attempt 2 out of maximum retry attempts 5.19 347 [iot_thread] 
20 347 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:745] 21 347 [iot_thread] Creating a TLS connection to  269 [eventTask] [INFO ][DEMO][2690] Connected to WiFi access point, ip address: 172.22.173.224.
5 270 [iot_thread] [INFO ][DEMO][2700] Successfully initialized the demo. Network type for the demo: 1
6 270 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:745] 7 270 [iot_thread] Creating a TLS connection to ************-ats.iot.eu-central-1.amazonaws.com:8883.8 270 [iot_thread] 
GAP procedure initiated: stop advertising.
GAP procedure initiated: stop advertising.
E (3164) PKCS11: failed nvs get file size 4354 0
9 301 [iot_thread] ERROR: Handshake failed with error code X509 - Certificate verification failed, e.g. CRL, CA or signature check failed : <No-Low-Level-Code> 
10 301 [iot_thread] TLS_Connect fail (0x2700, NULL)
11 301 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:323] 12 301 [iot_thread] Failed to establish new connection. secureSocketStatus=-1.13 301 [iot_thread] 
14 303 [iot_thread] [WARN] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:753] 15 303 [iot_thread] Connection to the broker failed. Attempting connection retry after backoff delay.16 303 [iot_thread] 
17 347 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:689] 18 347 [iot_thread] Retry attempt 2 out of maximum retry attempts 5.19 347 [iot_thread] 
20 347 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:745] 21 347 [iot_thread] Creating a TLS connection to ***********-ats.iot.eu-central-1.amazonaws.com:8883.22 347 [iot_thread] 
E (3884) PKCS11: failed nvs get file size 4354 0
23 373 [iot_thread] ERROR: Handshake failed with error code X509 - Certificate verification failed, e.g. CRL, CA or signature check failed : <No-Low-Level-Code> 
24 373 [iot_thread] TLS_Connect fail (0x2700, NULL)
25 373 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:323] 26 374 [iot_thread] Failed to establish new connection. secureSocketStatus=-1.27 374 [iot_thread] 
28 375 [iot_thread] [WARN] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:753] 29 375 [iot_thread] Connection to the broker failed. Attempting connection retry after backoff delay.30 375 [iot_thread] 
31 388 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:689] 32 388 [iot_thread] Retry attempt 3 out of maximum retry attempts 5.33 388 [iot_thread] 
34 388 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:745] 35 388 [iot_thread] Creating a TLS connection to ************-ats.iot.eu-central-1.amazonaws.com:8883.36 388 [iot_thread] 
E (4284) PKCS11: failed nvs get file size 4354 0
37 414 [iot_thread] ERROR: Handshake failed with error code X509 - Certificate verification failed, e.g. CRL, CA or signature check failed : <No-Low-Level-Code> 
38 414 [iot_thread] TLS_Connect fail (0x2700, NULL)
39 414 [iot_thread] [ERROR] [Transport_Secure_Sockets] [transport_secure_sockets.c:323] 40 414 [iot_thread] Failed to establish new connection. secureSocketStatus=-1.41 414 [iot_thread] 
42 415 [iot_thread] [WARN] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:753] 43 415 [iot_thread] Connection to the broker failed. Attempting connection retry after backoff delay.44 415 [iot_thread] 
45 443 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:689] 46 443 [iot_thread] Retry attempt 4 out of maximum retry attempts 5.47 443 [iot_thread] 
48 443 [iot_thread] [INFO] [MQTT_MutualAuth_Demo] [mqtt_demo_mutual_auth.c:745] 49 443 [iot_thread] Creating a TLS connection to a3pzitw5orfizl-ats.iot.eu-central-1.amazonaws.com:8883.50 443 [iot_thread] 
 269 [eventTask] [INFO ][DEMO][2690] Connected to WiFi access point, ip address: 172.22.173.224.
5 270 [iot_thread] [INFO ][DEMO][2700] Successfully initialized the demo. Network type for the demo: 1
[... cut ...]
[INFO] [MQTT] [core_mqtt.c:1045] De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
[INFO] [MQTT] [core_mqtt.c:1058] State record updated. New state=MQTTPubAckSend.
[INFO] [DEMO] [mqtt_demo_mutual_auth.c:812] Incoming QOS : 1.
[INFO] [DEMO] [mqtt_demo_mutual_auth.c:827] Incoming Publish Topic Name: aws-freertos-test1/example/topic matches subscribed topic.
Incoming Publish message Packet Id is 1.
Incoming Publish Message : Hello World!.

[INFO] [DEMO] [mqtt_demo_mutual_auth.c:1442] Delay before continuing to next iteration.

```

closed 2022-01-07 by eelaix