aws/amazon-freertos #2780: Classic (BR/EDR) BT support query

MarkWangChinese, 2020-11-23:
Hi,

I want to ask whether the Classic BT APIs are implemented in FreeRTOS or ACS?
I see the head files (amazon-freertos/libraries/abstractions/ble_hal/include/bt_hal_avsrc_profile.h and bt_hal_manager_adapter_classic.h, but I don't see any implementation of these APIs that are defined in these two head files.

Thank you!

ravibhagavandas, 2020-11-28:
Hi,

FreeRTOS does not implement Classic Bluetooth APIs, we support only Bluetooth Low Energy GATT Peripheral role APIs, as they are more optimized for power consumption.

For details on ACS implementation of the classic Bluetooth APIs you can visit the page [here](https://developer.amazon.com/acs-devices).

yourslab, 2020-12-05:
Resolving this issue unless there are further questions. Please feel free to open another one again.

closed 2020-12-05 by yourslab