aws/amazon-freertos #2783: How to run 'make menuconfig' on esp-idf within an amazon-freertos project?

octopusbunch, 2020-11-24:
I am using the 202011.00 release and an ESP32-Wrover devkit. From the `<amazon-freertos>\vendors\espressif\esp-idf` directory in amazon-freertos I run the command`tools/idf.py menuconfig` but I get the result 

`Executing "ninja menuconfig"...
ninja: error: unknown target 'menuconfig'
ninja failed with exit code 1`

I've also tried: 

```
amazon-freertos\vendors\espressif\boards\esp32\aws_demos
λ make menuconfig
Makefile:6: *** Make build system is not supported, please use CMake build system. For more details please refer to getting started guide.  Stop.

```

mahavirj, 2020-11-25:
@octopusbunch 

Following instructions should work for getting to configuration prompt:

```
$ cd <amazon-freertos>
$ unset IDF_PATH
$ cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B build
$ ./vendors/espressif/esp-idf/tools/idf.py menuconfig
```

octopusbunch, 2020-12-01:
Thanks! These instructions worked except I had to specify the cmake file to use or my system would default to build for MS Visual Studio:
`cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -DCMAKE_TOOLCHAIN_FILE=./tools/cmake/toolchains/xtensa-esp32.cmake -GNinja -S . -B build`

yourslab, 2020-12-05:
Seems like the issue is resolved, so I will be closing this. Feel free to open another if you encounter any other issues.

closed 2020-12-05 by yourslab

bastianhjaeger, 2021-04-17:
Hi, 

I stumbled across this and thought I leave a note for whom might be interested. 
I did some CMakeLists.txt optimizations and got that same error 

```
... ninja: error: unknown target 'menuconfig'
```

After some tesing I found out, that the CMakeLists.txt needs to look like this:

```
cmake_minimum_required(VERSION 3.5)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

add_definitions(-Wno-unused-variable)
set(PROJECT_VER "whatever")
set(EXTRA_COMPONENT_DIRS 
    ./src
    ${IDF_PATH}/../esp-idf-lib/components)
...

project(sensor_cube_v2)
```

Where the critical part was/is, that the line `project(sensor_cube_v2)` not only needs to be after the include of `...prokect.cmake` but also in the complete end of the file to make it work again. 

Hope this helps someone.