aws/amazon-freertos #2785: Amazon FreeRTOS Cross Compile for ESP32 [Build-Errors]

victorK8, 2020-11-24:
- ESP32-WROOM-32 Platform
- Ubuntu 20.04 
- Last version of FreeRTOS  (I download it from aws iot core service)

I create build system via "cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DCMAKE_BUILD_TYPE=Debug -GNinja -S . -B [build-folder-path]" and when I compile it by using "ninja" or "cmake --build [build-path]". Appear the next errors:

Errors 
cc1: some warnings being treated as errors
[9/963] Building C object CMakeFiles/afr_kernel.dir/freertos_kernel/tasks.c.obj
ninja: build stopped: subcommand failed.

Moreover, I try to quit -Werror=all flags, but it shows another errors. 

Some help? Thanks!

shubhamkulkarni97, 2020-11-25:
@victorK8 

From the provided error log, I'm unable to get exact error. Can you share entire build log?
As a part of basic setup, `IDF_PATH` environment variable must be unset and toolchain version 5.2.0 must be used (To check toolchain version, run `xtensa-esp32-elf-gcc --version`).

victorK8, 2020-11-25:
Thank @shubhamkulkarni97 for helping me.

I check my esp32 compiler version via  `xtensa-esp32-elf-gcc --version`. It shows these logs:

                xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
                Copyright (C) 2018 Free Software Foundation, Inc.
                This is free software; see the source for copying conditions.  There is NO
                warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Logs show that compiler's version is 8.4.0. Moreover, I share my last build log.

[  1%] Built target afr_kernel
[  2%] Built target afr_wifi
[ 11%] Built target afr_3rdparty_mbedtls
[ 12%] Built target afr_crypto
[ 13%] Built target afr_pkcs11_utils
[ 13%] Built target afr_pkcs11_implementation
[ 13%] Built target afr_logging
[ 14%] Built target afr_secure_sockets
[ 14%] Built target afr_common
[ 14%] Built target afr_platform
[ 15%] Built target afr_pkcs11
[ 16%] Built target afr_tls
[ 16%] Built target afr_freertos_plus_cli
[ 16%] Built target afr_transport_interface_secure_sockets
[ 16%] Built target afr_retry_utils
[ 16%] Built target afr_core_http
[ 16%] Built target afr_http_demo_helpers
[ 16%] Built target afr_core_http_demo_dependencies
[ 16%] Built target afr_https
[ 17%] Built target partition_table
[ 17%] Built target idf_component_ulp
[ 17%] Built target afr_core_mqtt
[ 18%] Built target afr_mqtt_demo_helpers
[ 18%] Built target afr_3rdparty_tinycbor
[ 19%] Built target afr_serializer
[ 20%] Built target afr_ble_hal
[ 22%] Built target afr_ble
[ 23%] Built target afr_core_mqtt_demo_dependencies
[ 25%] Built target idf_component_ulp_sections_info
[ 26%] Built target idf_component_soc
[ 26%] Built target idf_component_soc_sections_info
[ 26%] Built target idf_component_log
[ 26%] Built target idf_component_log_sections_info
[ 27%] Built target idf_component_heap
[ 27%] Built target idf_component_heap_sections_info
[ 29%] Built target idf_component_freertos
[ 29%] Built target idf_component_freertos_sections_info
[ 29%] Built target idf_component_vfs
[ 29%] Built target idf_component_vfs_sections_info
[ 30%] Built target idf_component_newlib
[ 30%] Built target idf_component_newlib_sections_info
[ 30%] Built target idf_component_esp_ringbuf
[ 30%] Built target idf_component_esp_ringbuf_sections_info
[ 32%] Built target idf_component_driver
[ 33%] Built target idf_component_driver_sections_info
[ 33%] Built target idf_component_esp_event
[ 33%] Built target idf_component_esp_event_sections_info
[ 34%] Built target idf_component_ethernet
[ 34%] Built target idf_component_ethernet_sections_info
[ 35%] Built target idf_component_micro-ecc
[ 35%] Built target idf_component_micro-ecc_sections_info
[ 35%] Built target idf_component_efuse
[ 35%] Built target idf_component_efuse_sections_info
[ 36%] Built target idf_component_bootloader_support
[ 36%] Built target idf_component_bootloader_support_sections_info
[ 36%] Built target idf_component_app_update
[ 37%] Built target idf_component_app_update_sections_info
[ 38%] Built target idf_component_spi_flash
[ 38%] Built target idf_component_spi_flash_sections_info
[ 38%] Building CXX object esp-idf/nvs_flash/CMakeFiles/idf_component_nvs_flash.dir/src/nvs_api.cpp.obj
In file included from /home/maluma/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/cstdlib:75,
                 from /home/maluma/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/ext/string_conversions.h:41,
                 from /home/maluma/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/basic_string.h:6400,
                 from /home/maluma/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/string:52,
                 from /home/maluma/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/stdexcept:39,
                 from /home/maluma/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/array:39,
                 from /home/maluma/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/tuple:39,
                 from /home/maluma/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/unique_ptr.h:37,
                 from /home/maluma/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/memory:80,
                 from /home/maluma/esp32-avs/amazon-freertos/vendors/espressif/esp-idf/components/nvs_flash/src/nvs.hpp:18,
                 from /home/maluma/esp32-avs/amazon-freertos/vendors/espressif/esp-idf/components/nvs_flash/src/nvs_api.cpp:14:
/home/maluma/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/home/maluma/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
                                                    ^~~~~~~~~~~~~
make[2]: *** [esp-idf/nvs_flash/CMakeFiles/idf_component_nvs_flash.dir/build.make:82: esp-idf/nvs_flash/CMakeFiles/idf_component_nvs_flash.dir/src/nvs_api.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:5262: esp-idf/nvs_flash/CMakeFiles/idf_component_nvs_flash.dir/all] Error 2
make: *** [Makefile:103: all] Error 2

shubhamkulkarni97, 2020-11-25:
@victorK8 

Logs show that incorrect toolchain version is used.
Please run following commands to install correct toolchain:

* `./vendors/espressif/esp-idf/install.sh` (Installs required tools)
* `. ./vendors/espressif/esp-idf/export.sh` (Adds required tools to PATH)
* `unset IDF_PATH`

After executing these commands, you should be able to build application with cmake command mentioned in the description.

Thanks,
Shubham

victorK8, 2020-11-25:
@shubhamkulkarni97 

Thank so much!! It works well now.

Victor

closed 2020-11-25 by victorK8