aws/amazon-freertos #2786: how to my custom Json file merge with AWS cloud

SGOSAI602, 2020-11-24:
Describe the bug
how to my custom Json file merge with AWS cloud and send data on AWS server?

System information
Hardware - ESP32-WROOM-32 D
Operating System Linux-Ubuntu
Version of FreeRTOS (run git describe --tags to find it)
Project AWS FreeRTOS shadow demo

Thank you!

aggarg, 2020-11-28:
Are you asking if you can send a custom JSON to the AWS Device Shadow service? If so, yes, you can report whatever values you want to report in your shadow document as long as the document is valid. You can read more about Shadow Document structure here: https://docs.aws.amazon.com/iot/latest/developerguide/device-shadow-document.html

Thanks.

lundinc2, 2021-03-01:
Hi @SGOSAI602,

I am closing this issue as it has gone stale. Please re-open this if you have further questions!

closed 2021-03-01 by lundinc2