aws/amazon-freertos #2804: Agent OTA : IOT MQTT SERVER REFUSED

elieDaan, 2020-11-30:
**Describe the bug**
Hello, 

We don't know why but since this week end we have got a random issue when launching our check if OTA is available when our system is booting. 
**System information**
ESP32 with AFR2019 

**Expected behavior**
Connect to MQTT and check if there is a job available or not 

**Screenshots or console output**
```c
5 1586 [iot_thread] [INFO ][MQTT][15860] Establishing new MQTT connection.
6 1586 [iot_thread] [INFO ][MQTT][15860] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
7 1588 [iot_thread] [INFO ][MQTT][15880] (MQTT connection 0x3ffd7c1c, CONNECT operation 0x3ffd7d38) Waiting for operation completion.
8 1609 [iot_thread] [INFO ][MQTT][16090] (MQTT connection 0x3ffd7c1c, CONNECT operation 0x3ffd7d38) Wait complete with result SUCCESS.
9 1609 [iot_thread] [INFO ][MQTT][16090] New MQTT connection 0x3ffc0c0c established.
10 1610 [iot_thread] Connected to broker.
11 1611 [iot_thread] [OTA_AgentInit_internal] OTA Task is Ready.
12 1611 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [Ready] Event [Start] New state [RequestingJob] 
13 1611 [OTA Agent Task] [INFO ][MQTT][16110] (MQTT connection 0x3ffd7c1c) SUBSCRIBE operation scheduled.
14 1611 [OTA Agent Task] [INFO ][MQTT][16110] (MQTT connection 0x3ffd7c1c, SUBSCRIBE operation 0x3ffd7d38) Waiting for operation completion.
15 1616 [OTA Agent Task] [INFO ][MQTT][16160] (MQTT connection 0x3ffd7c1c, SUBSCRIBE operation 0x3ffd7d38) Wait complete with result SERVER REFUSED.
16 1616 [OTA Agent Task] [prvSubscribeToJobNotificationTopics] Failed: $aws/things/AC:67:B2:41:AC:90/jobs/$next/get/accepted
17 1616 [OTA Agent Task] [prvOTAAgentTask] Handler failed. Current State [RequestingJob] Event  [RequestJobDocument] Error Code [134217728] 
18 1619 [NetRecv] [ERROR][MQTT][16190] (MQTT connection 0x3ffd7c1c) Packet parser status SERVER REFUSED.
19 2616 [OTA Agent Task] [INFO ][MQTT][26160] (MQTT connection 0x3ffd7c1c) SUBSCRIBE operation scheduled.
20 2616 [OTA Agent Task] [INFO ][MQTT][26160] (MQTT connection 0x3ffd7c1c, SUBSCRIBE operation 0x3ffd7d38) Waiting for operation completion.
21 2620 [OTA Agent Task] [INFO ][MQTT][26200] (MQTT connection 0x3ffd7c1c, SUBSCRIBE operation 0x3ffd7d38) Wait complete with result SERVER REFUSED.
22 2620 [OTA Agent Task] [prvSubscribeToJobNotificationTopics] Failed: $aws/things/AC:67:B2:41:AC:90/jobs/$next/get/accepted
23 2620 [OTA Agent Task] [prvOTAAgentTask] Handler failed. Current State [RequestingJob] Event  [RequestTimer] Error Code [134217728] 
```
**To reproduce**

**Code to reproduce the bug**

https://gist.github.com/elieDaan/17f906c76bc991806a830828535fe3a3

**Additional context**

Any help is welcomed because it's a bug in production :( 

Thank you!

aggarg, 2020-11-30:
It seems that the connection is successful but the subscribe operation fails. Is your policy configured correctly? Can you share your policy document? 

Thanks.

pvyawaha, 2020-11-30:
Hello,

Thank you for sharing details about the issue and the logs. It seems device is connecting to the endpoint but further publish/subscribe is failing. In the AWS IoT console please make sure that there is a policy attached to the certificate as all publish/subscribe operations are denied by default. Check the policy attached to the certificate for correct authorizations.

elieDaan, 2020-12-01:
Thanks you for your response. I don't know why but some clients after let the device off 24 hours succeed to connect. But the bug is still here. 
Our devices have this policy : 
```
{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": "iot:Connect",
      "Resource": "arn:aws:iot:eu-west-1:26XXXX:*"
    },
    {
      "Effect": "Allow",
      "Action": "iot:Publish",
      "Resource": "arn:aws:iot:eu-west-1:26XXXX:*"
    },
    {
      "Effect": "Allow",
      "Action": "iot:Subscribe",
      "Resource": "arn:aws:iot:eu-west-1:26XXXX:*"
    },
    {
      "Effect": "Allow",
      "Action": "iot:Receive",
      "Resource": "arn:aws:iot:eu-west-1:26XXXX:*"
    }
  ]
}`

elieDaan, 2020-12-01:
I've tried to remove in the policy attached to this object the autorization to subscribe to iot topic but it's not the same log. You get a TIMEOUT ERROR and not a SERVER REFUSED so I don't think it's a policy issue here. 
```c
5 1269 [iot_thread] [INFO ][MQTT][12690] Establishing new MQTT connection.
6 1269 [iot_thread] [INFO ][MQTT][12690] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
7 1271 [iot_thread] [INFO ][MQTT][12710] (MQTT connection 0x3ffd7e74, CONNECT operation 0x3ffd7c4c) Waiting for operation completion.
8 1285 [iot_thread] [INFO ][MQTT][12850] (MQTT connection 0x3ffd7e74, CONNECT operation 0x3ffd7c4c) Wait complete with result SUCCESS.
9 1285 [iot_thread] [INFO ][MQTT][12850] New MQTT connection 0x3ffc0c0c established.
10 1285 [iot_thread] Connected to broker.
11 1286 [iot_thread] [OTA_AgentInit_internal] OTA Task is Ready.
12 1286 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [Ready] Event [Start] New state [RequestingJob] 
13 1286 [OTA Agent Task] [INFO ][MQTT][12860] (MQTT connection 0x3ffd7e74) SUBSCRIBE operation scheduled.
14 1288 [OTA Agent Task] [INFO ][MQTT][12880] (MQTT connection 0x3ffd7e74, SUBSCRIBE operation 0x3ffd7c4c) Waiting for operation completion.
15 4288 [OTA Agent Task] [WARN ][TASKPOOL][42880] Attempt to cancel a job that is already executing, or canceled.
16 4288 [OTA Agent Task] [INFO ][MQTT][42880] (MQTT connection 0x3ffd7e74, SUBSCRIBE operation 0x3ffd7c4c) Wait complete with result TIMEOUT.
```

aggarg, 2020-12-02:
Do you still see this issue? Is it possible that you saw subscribe failures due to the recent outage: https://twitter.com/AWSSupport/status/1331687607172993024

Thanks.

elieDaan, 2020-12-03:
Hello @aggarg ,

Thanks you very much !  It seems that we haven't succeed to get the SERVER REFUSED error yesterday so yes it could be linked. It's creazy how all is related to Kinesis.

We need know to increase the reliabilty of our system in case AWS IOT is down. Have you got an idea to get out properly of OTA Internal ? We think about use the Abort callback but in fact it's also called if everything is fine and it's the end of the task. For us, it's not the same action if the connection was okay or not..

closed 2020-12-07 by elieDaan