aws/amazon-freertos #2805: common_io not available

ChristyBrouwer, 2020-11-30:
**Describe the bug**
When using amazon-freertos as a library the common_io library is not available to use in custom components

**System information**
- ESP32
- VSCode
- Operating System: Windows 10
- Project: Custom Application

**Expected behavior**
Expecting to be able to include iot_i2c.h without cmake issues. When using target_link_libraries() I should not get the following error:

**Screenshots or console output**
CMake Error at .../project/amazon-freertos/vendors/espressif/esp-idf/tools/cmake/component_utils.cmake:20 (message): Required component common_io is not found in any of the provided  COMPONENT_DIRS

**Additional context**
I found a similar issue from a while back, reacted to this as well but wasn't sure if a separate issue would be better.
Comment on similar issue: https://github.com/aws/amazon-freertos/issues/2508#issuecomment-735814979_

aggarg, 2020-11-30:
Would you please share your CMake code for adding `common_io`? Are your declaring the dependency like this: https://github.com/aws/amazon-freertos/blob/28ab489d0a83be1bb59ebfb63121aed704244f3f/demos/cli/CMakeLists.txt#L13

Thanks.

ChristyBrouwer, 2020-12-01:
My CMake for adding `common_io` is as follows.

In ./project/CMakeLists.txt
`target_link_libraries(base PRIVATE AFR::common_io)`

In ./project/components/common_types/include/common_types.h I try to include iot_ic2 using the standard
`#include "iot_i2c.h"`

In ./project/components/common_types/CMakeLists.txt I link to common_io as follows
`target_link_libraries(${COMPONENT_TARGET} PRIVATE AFR::common_io)`

tgsong, 2020-12-02:
@ChristyBrouwer How are  you using amazon-freertos as a library? Are you putting it in a subfolder in your own project? If that's the case, it might be helpful to take a look at this example repo, https://github.com/tgsong/amazon-freertos-examples/

I have added `AFR::common_io` as a dependency here, https://github.com/tgsong/amazon-freertos-examples/blob/99ca388a22b9419c04f1f48f90d89b30d38b7677/CMakeLists.txt#L33

And in the `main.c` (this file is copied from the [esp32 demo main.c](https://github.com/aws/amazon-freertos/blob/fd27d7bbd0d370004817c217b7fb95242f136b0d/vendors/vendor/boards/board/aws_demos/application_code/main.c)) file it's including it, https://github.com/tgsong/amazon-freertos-examples/blob/99ca388a22b9419c04f1f48f90d89b30d38b7677/src/main.c#L65

ChristyBrouwer, 2020-12-03:
Hey @tgsong , 
I have amazon-freertos as a subfolder. Namely at ./project/amazon-freertos. All the project's components are at ./project/components/componentname. 

Thank you for the example codes, I took a look at them but the project I'm working on is almost exactly the same in terms of setup yet it's not working. So sadly it didn't fix it, nor did I find something I'm missing in my own CMakes

It's important to note I'm only having this problem with common_io. I've already included many of the other AFR libraries in my components without issues. `Common_io` is the only one giving problems.

tgsong, 2020-12-03:
1. which version of amazon-freertos are you running? My example is updated to 202011.00.
1. can you share the full call stack of that cmake error?
1. what is the first line in `amazon-freertos/libraries/abstractions/common_io/CMakeLists.txt`? Looks like we made a fix 2 months ago, before this target doesn't compile into a real library, which may cause some issues. See https://github.com/aws/amazon-freertos/pull/2509

ChristyBrouwer, 2020-12-04:
I'm running the latest version of the `master`
I've fixed the issue by changing the way I implemented `common_io` in my `common_types/CMakeLists.txt`

In `./project/components/common_types/CMakeLists.txt` I now link to common_io as follows
`target_link_libraries(${COMPONENT_TARGET} PUBLIC AFR::common_io)`

Used to be
`target_link_libraries(${COMPONENT_TARGET} PRIVATE AFR::common_io)`

Not entirely sure why this fix worked, but as long as it does I'm happy

closed 2020-12-04 by ChristyBrouwer