aws/amazon-freertos #2808: trouble linking with curiosity_pic32mzef & xc32 on windows

dudmuck, 2020-11-30:
The first library it tries to link is tinycbor, it fails linking.

**System information**
curiosity_pic32mzef building using xc32 v2.50
mingw-make and cmake-3.18.0
windows 10
amazon-freertos from git: 202011.00-109-gb2217ded7

this is my cmake command line:
cmake -G "MinGW Makefiles" `
    -DVENDOR=microchip `
    -DBOARD=curiosity_pic32mzef `
    -DCOMPILER=xc32 `
    -DMCHP_HEXMATE_PATH="c:/Program Files (x86)/Microchip/MPLABX/v5.40/mplab_platform/bin" `
    -DAFR_TOOLCHAIN_PATH="C:/Program Files/Microchip/xc32/v2.50/bin" `
    ..

this is the last part of building VERBOSE=1

[  2%] Building C object libraries/3rdparty/CMakeFiles/afr_3rdparty_tinycbor.dir/tinycbor/src/cborparser_dup_string.c.obj
cd /d C:\src\amazon-freertos\build\libraries\3rdparty && C:\PROGRA~1\MICROC~1\xc32\v2.50\bin\xc32-gcc.exe  -IC:\src\amazon-freertos\libraries\3rdparty\tinycbor\src -w -g -x c -ffunction-sections -O1 -mprocessor=32MZ2048EFM100 -mnewlib-libc -std=gnu99 -fgnu89-inline -o CMakeFiles\afr_3rdparty_tinycbor.dir\tinycbor\src\cborparser_dup_string.c.obj -c C:\src\amazon-freertos\libraries\3rdparty\tinycbor\src\cborparser_dup_string.c
[  2%] Linking C static library libafr_3rdparty_tinycbor.a
cd /d C:\src\amazon-freertos\build\libraries\3rdparty && "C:\Program Files\CMake\bin\cmake.exe" -P CMakeFiles\afr_3rdparty_tinycbor.dir\cmake_clean_target.cmake
cd /d C:\src\amazon-freertos\build\libraries\3rdparty && "C:\Program Files\CMake\bin\cmake.exe" -E cmake_link_script CMakeFiles\afr_3rdparty_tinycbor.dir\link.txt --verbose=1
C:\PROGRA~1\MICROC~1\xc32\v2.50\bin\xc32-ar.exe qc libafr_3rdparty_tinycbor.a CMakeFiles/afr_3rdparty_tinycbor.dir/tinycbor/src/cborpretty.c.obj CMakeFiles/afr_3rdparty_tinycbor.dir/tinycbor/src/cborpretty_stdio.c.obj CMakeFiles/afr_3rdparty_tinycbor.dir/tinycbor/src/cborencoder.c.obj CMakeFiles/afr_3rdparty_tinycbor.dir/tinycbor/src/cborencoder_close_container_checked.c.obj CMakeFiles/afr_3rdparty_tinycbor.dir/tinycbor/src/cborerrorstrings.c.obj CMakeFiles/afr_3rdparty_tinycbor.dir/tinycbor/src/cborparser.c.obj CMakeFiles/afr_3rdparty_tinycbor.dir/tinycbor/src/cborparser_dup_string.c.obj
C:\PROGRA~1\MICROC~1\xc32\v2.50\bin\XC32-R~1.EXE libafr_3rdparty_tinycbor.a
libraries\3rdparty\CMakeFiles\afr_3rdparty_tinycbor.dir\build.make:191: recipe for target 'libraries/3rdparty/libafr_3rdparty_tinycbor.a' failed
gmake[2]: *** [libraries/3rdparty/libafr_3rdparty_tinycbor.a] Error 255
gmake[2]: *** Deleting file 'libraries/3rdparty/libafr_3rdparty_tinycbor.a'
gmake[2]: Leaving directory 'C:/src/amazon-freertos/build'
CMakeFiles\Makefile2:2553: recipe for target 'libraries/3rdparty/CMakeFiles/afr_3rdparty_tinycbor.dir/all' failed
gmake[1]: *** [libraries/3rdparty/CMakeFiles/afr_3rdparty_tinycbor.dir/all] Error 2
gmake[1]: Leaving directory 'C:/src/amazon-freertos/build'
Makefile:101: recipe for target 'all' failed
gmake: *** [all] Error 2

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 

**To reproduce**
Steps to reproduce the behavior:

1. ...
2. ...

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
Add any other context about the problem here.

Thank you!

aggarg, 2020-12-02:
Please try using "Ninja" generator:

```
cmake -G "Ninja" 
    -DVENDOR=microchip 
    -DBOARD=curiosity_pic32mzef 
    -DCOMPILER=xc32 
    -DMCHP_HEXMATE_PATH="c:/Program Files (x86)/Microchip/MPLABX/v5.40/mplab_platform/bin" 
    -DAFR_TOOLCHAIN_PATH="C:/Program Files/Microchip/xc32/v2.50/bin" 
```

Thanks.

muneebahmed10, 2020-12-22:
Hi @dudmuck,

Did the [above](https://github.com/aws/amazon-freertos/issues/2808#issuecomment-736903284) response resolve your issue?

Thanks

muneebahmed10, 2020-12-23:
I will close this issue. Please feel free to open a new one if you have any further questions.

Thank you

closed 2020-12-23 by muneebahmed10