aws/amazon-freertos #2822: Build folder does not generate .elf file when creating own project ESP32

MrMelon1232, 2020-12-03:
Hi,

Building your own project is quite different than building the demos project (example: using different commands for building, build folder, etc). Referring to this guide :``` https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project```, specifically section ```Using FreeRTOS in your own CMake project for ESP32```, I have no trouble building a project and running code. However, the executable file being created is not an .elf file while building demos does generate an elf file, notably: aws_demos.elf. I was wondering what were the necessary parts of a CMakeLists.txt required to generate this .elf file. Another note is that using JTAG debugging for demos works while using it for your own project does not. 

**System information**
- Hardware: ESP32
- Operating System : Linux
- Version of FreeRTOS: Commit d6604453bcb9bc9aba230f8af70b8c94da4e9f77
- Project: Custom project

**Project hierarchy**
```
my_project:
   -build
   -freertos-configs
   -src:
          -main.c
   -CMakeLists.txt
```

**CMakeLists.txt content**
```
cmake_minimum_required(VERSION 3.13)

project(my_project)

add_executable(my_app src/main.c)

# Tell IDF build to link against this target.
set(IDF_PROJECT_EXECUTABLE my_app)

# As of now there's no offical way to redefine config files outside of FreeRTOS source tree.
# This is a temporary approach to inject an include path so that this takes precedence over the
# config file directory inside FreeRTOS.
include_directories(BEFORE freertos-configs)

# Add freertos as an subdirectory. AFR_BOARD tells which board to target.
set(AFR_BOARD espressif.esp32_devkitc CACHE INTERNAL "")
add_subdirectory(freertos)

# Link against the mqtt demo so that we can use it. Dependencies of this demo are transitively
# linked.
target_link_libraries(my_app PRIVATE AFR::demo_core_mqtt)
target_link_libraries(my_app PRIVATE AFR::demo_device_shadow)
target_link_libraries(my_app PRIVATE AFR::demo_ota)
target_link_libraries(my_app PRIVATE AFR::shadow)
#target_link_libraries(my_app PRIVATE AFR::common_io)
include_directories(my_app PRIVATE fsonet-configs)
#include_directories(my_app PRIVATE  "amazon-freertos/demos/network_manager")
include_directories(my_app PRIVATE "amazon-freertos/demos/dev_mode_key_provisioning/include")
```

**Commands being used to build**
```
cmake -S . -B build -DCMAKE_TOOLCHAIN_FILE=freertos/tools/cmake/toolchains/xtensa-esp32.cmake -DCMAKE_BUILD_TYPE=debug
cmake --build build/ --clean-first --parallel 8 && ./freertos/vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p /dev/ttyUSB0 -B build/
```
**Commands and output for JTAG**
```
OpenOCD output:
Warn : No symbols for FreeRTOS!
Info : Halt timed out, wake up GDB.
Error: timed out while waiting for target halted / 3 - 2
Error executing event gdb-attach on target esp32.cpu0:

Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected

Debug consol output:
Reading symbols from build/my_app...done.
gdbinit:1: Error in sourced command file:
```

Thank you!

yourslab, 2020-12-04:
If you take a look at `vendors/espressif/boards/esp32/CMakeLists.txt`, you will be able to see what is required to build FreeRTOS for esp32 with `cmake`. In that same directory, there are several `.cmake` that are included depending on the type of esp32 board.

MrMelon1232, 2020-12-04:
Thanks for the quick response! I'll look into it.

MrMelon1232, 2020-12-04:
Hi @yourslab,

I was able to find the commands to add .elf suffix to the executable along with linking esp-idf libraries but still no luck with attempting to debug the application with JTAG. Would you happen to know what are the specific build/CMakeLists.txt requirements for this debugger ?

yourslab, 2020-12-05:
Could this be related to #2820? JTAG debugging is still not working properly even when using `gdbstub.c` and `tasks.c` provided from `esp-idf`? Is it still the same output as what you posted above?

MrMelon1232, 2020-12-07:
It is a JTAG issue but it is not related however, I was able to use JTAG debugging with the aws demos project but not with my own personal project. This issue #2820 was for trying to enable a feature in make menuconfig and it failing for both the aws demos project and my own project.

shubhamkulkarni97, 2020-12-16:
Hi @MrMelon1232,

I checked building an external project and was able to use JTAG debugging with OpenOCD+GDB setup.

Looking at the provided CMakeLists.txt, I agree with you that executable file doesn't have .elf extension but that will not cause any issues with JTAG debugging.

From the CMakeLists.txt you have provided, my_app will be the executable file generated.
You can use following command to start JTAG debugging:
`xtensa-esp32-elf-gdb build/my_app`

MrMelon1232, 2020-12-16:
HI @shubhamkulkarni97,

I compared the repository you linked with mine and couldn't see any differences that would impact OpenOCD+GDB setup. I have however used the command you suggested notably ``` xtensa-esp32-elf-gdb build/my_app``` and JTAG debugging looks like its working although it says the program is not running. Would you have any insight on this? Also, I realized running the command ```xtensa-esp32-elf-gdb -x gdbinit build/my_app``` fails with the same error code mentioned earlier in the post. My gdbinit file contains the following lines:
```
target remote :3333
set remote hardware-watchpoint-limit 2
mon reset halt
flushregs
thb app_main
c
```
Could there be a reason why this would fail ?

shubhamkulkarni97, 2020-12-22:
Hi @MrMelon1232, Is your issue resolved?

MrMelon1232, 2020-12-22:
Hi @shubhamkulkarni97,
I can confirm that this issue is closed as of now. Thank you for your help and quick responses.
I will close this issue now.

closed 2020-12-22 by MrMelon1232

Yoimer, 2021-03-21:
> Hi @MrMelon1232,
> 
> I checked building an external project and was able to use JTAG debugging with OpenOCD+GDB setup.
> 
> Looking at the provided CMakeLists.txt, I agree with you that executable file doesn't have .elf extension but that will not cause any issues with JTAG debugging.
> 
> From the CMakeLists.txt you have provided, my_app will be the executable file generated.
> You can use following command to start JTAG debugging:
> `xtensa-esp32-elf-gdb build/my_app`
> 
> I used [amazon-freertos-examples](https://github.com/tgsong/amazon-freertos-examples) repository which demonstrates custom project with Amazon FreeRTOS.

@shubhamkulkarni97 @yourslab  Hello. Did you use any UI when working with JTAG debugging? I would like to make breakpoints, step in out debugging for the AWS FreeRTOS ESP32 demos. Is that possible already?. Thanks in advance.