aws/amazon-freertos #2837: IotClock_TimerXXX api may cause troubles in timer task link list

pakls, 2020-12-07:
If IotClock_TimerCreate is invoked, followed by IotClock_TimerArm, then IotClock_TimerDestroy.
We will have this timer in timer task linked-list and stopped.

If we call IotClock_TimerCreate again with the same static buffer, we may have the timer linked-list mangled.

lundinc2, 2020-12-14:
Hi @pakls,

Thank you for this report, I am investigating this issue.

lundinc2, 2020-12-22:
Hi @pakls,

Can you provide some further details, is this an issue that you ran into? The timer is being removed from the timer linked list as the IotClock_TimerDestroy is processed

lundinc2, 2021-02-25:
I am closing this as a stale issue. Please re-open with further details on the issue.

Thanks,
Carl

closed 2021-02-25 by lundinc2