aws/amazon-freertos #2838: CONFIG_MQTT_BLE_TRANSPORT_DEMO_ENABLED Demo on nRF52840

bitpet, 2020-12-07:
Hi everyone

I was using the olter demo (`CONFIG_MQTT_DEMO_ENABLED`) out of the box without problems on the Nordic nRF52840. When activating the new demo (`CONFIG_MQTT_BLE_TRANSPORT_DEMO_ENABLED`), my board looses the connection to the broker after a while.
~~~ 

0 0 [iot_thread] [INFO ][DEMO][0] ---------STARTING DEMO---------

1 0 [iot_thread] [INFO ][INIT][0] SDK successfully initialized.
2 27 [iot_thread] [INFO ][DEMO][270] No networks connected for the demo. Waiting for a network connection.
3 47 [BLE] [ERROR][BLE_HAL][470] PM_UNKNOWN, 0.
4 47 [BLE] [INFO ][DEMO][470] BLE Connected to remote device, connId = 0

5 47 [iot_thread] [INFO ][DEMO][470] Successfully initialized the demo. Network type for the demo: 2
6 64 [BLE] [ERROR][BLE_HAL][640] UNKNOWN BLE EVENT 35.
7 144 [BLE] [ERROR][BLE_HAL][1440] UNKNOWN BLE EVENT 34.
8 239 [BLE] [ERROR][BLE_HAL][2390] UNKNOWN BLE EVENT 35.
9 244 [iot_thread] [INFO] [MQTTBLEDemo] [C:\ai\nordic\amazon_freertos_tasks_test\demos\ble\mqtt_ble\mqtt_demo_ble_transport.c:268] 
10 244 [iot_thread] The channel was initialized successfully
11 244 [iot_thread]
12 244 [iot_thread] [INFO] [MQTTBLEDemo] [C:\ai\nordic\amazon_freertos_tasks_test\demos\ble\mqtt_ble\mqtt_demo_ble_transport.c:367] 
13 244 [iot_thread] Trying to send a connect packet to the broker
14 244 [iot_thread]
15 244 [iot_thread] [INFO] [MQTTBLEDemo] [C:\ai\nordic\amazon_freertos_tasks_test\demos\ble\mqtt_ble\mqtt_demo_ble_transport.c:384] 
16 244 [iot_thread] Generated client identifier is iotdemo244
17 244 [iot_thread]
18 244 [iot_thread] [INFO] [MQTTBLEDemo] [C:\ai\nordic\amazon_freertos_tasks_test\demos\ble\mqtt_ble\mqtt_demo_ble_transport.c:408] 
19 244 [iot_thread] Successfully sent a connect packet to the broker
20 244 [iot_thread]
21 244 [iot_thread] [INFO] [MQTTBLEDemo] [C:\ai\nordic\amazon_freertos_tasks_test\demos\ble\mqtt_ble\mqtt_demo_ble_transport.c:409] 
22 244 [iot_thread] Waiting for a connection acknowledgement from the broker
23 244 [iot_thread]
24 446 [iot_thread] [INFO] [MQTT] [C:\ai\nordic\amazon_freertos_tasks_test\libraries\coreMQTT\source\core_mqtt_serializer.c:970] 
25 446 [iot_thread] CONNACK session present bit not set.
26 446 [iot_thread]
27 446 [iot_thread] [INFO] [MQTT] [C:\ai\nordic\amazon_freertos_tasks_test\libraries\coreMQTT\source\core_mqtt_serializer.c:912] 
28 446 [iot_thread] Connection accepted.
29 446 [iot_thread]
30 446 [iot_thread] [INFO] [MQTTBLEDemo] [C:\ai\nordic\amazon_freertos_tasks_test\demos\ble\mqtt_ble\mqtt_demo_ble_transport.c:420] 
31 446 [iot_thread] Successfully received a CONNACK packet
32 446 [iot_thread]
33 446 [iot_thread] [INFO] [MQTTBLEDemo] [C:\ai\nordic\amazon_freertos_tasks_test\demos\ble\mqtt_ble\mqtt_demo_ble_transport.c:428] 
34 446 [iot_thread] Successfully connected with the broker35 446 [iot_thread]
36 446 [iot_thread] [INFO] [MQTTBLEDemo] [C:\ai\nordic\amazon_freertos_tasks_test\demos\ble\mqtt_ble\mqtt_demo_ble_transport.c:443] 
37 446 [iot_thread] Trying to send a subscribe packet to the broker
38 446 [iot_thread]
45 509 [iot_thread] [INFO] [MQTTBLEDemo] [C:\ai\nordic\amazon_freertos_tasks_test\demos\ble\mqtt_ble\mqtt_demo_ble_transport.c:636] 
46 509 [iot_thread] Successfully subscribed to an MQTT topic (received SUBACK)
47 509 [iot_thread]
48 509 [iot_thread] [INFO] [MQTTBLEDemo] [C:\ai\nordic\amazon_freertos_tasks_test\demos\ble\mqtt_ble\mqtt_demo_ble_transport.c:703] 
49 509 [iot_thread] Trying to receive an incoming packet
50 509 [iot_thread]
~~~

Did anyone experience this issue or sees the solution? Your help is much appreciated.

lundinc2, 2020-12-10:
Hi @bitpet,

We will work on reproducing this.

Thanks,

Carl

dachalco, 2020-12-14:
Hi @bitpet 

I will try to reproduce the issue locally for debugging. Questions:
- Is that the end of the log, where it seems to hang waiting for an incoming packet?
- Which commit ID, or release, are you using?
- What are the contents of your and `iot_mqtt_agent_config.h`?

bitpet, 2020-12-16:
Hi @dachalco 

Thanks your for addressing this issue. 

Is that the end of the log, where it seems to hang waiting for an incoming packet?
- _Yes, that is the end of the log!_

Which commit ID, or release, are you using?
- _Commit ID: fd27d7bbd, Basically I pulled the release-branch before running into this issue_

What are the contents of your and iot_mqtt_agent_config.h?
~~~
#ifndef _AWS_MQTT_AGENT_CONFIG_H_
#define _AWS_MQTT_AGENT_CONFIG_H_

#include "FreeRTOS.h"

#define mqttconfigENABLE_METRICS    ( 1 )

#define mqttconfigMETRIC_SDK         "SDK=AmazonFreeRTOS"               /**< The SDK used by this device. */
#define mqttconfigMETRIC_VERSION     "Version="tskKERNEL_VERSION_NUMBER /**< The version number of this SDK. */
#define mqttconfigMETRIC_PLATFORM    "Platform=EspressifESP32"          /**< The platform that this SDK is running on. */

#define mqttconfigKEEP_ALIVE_INTERVAL_SECONDS         ( 1200 )

#define mqttconfigKEEP_ALIVE_ACTUAL_INTERVAL_TICKS    ( pdMS_TO_TICKS(300000) )

#define mqttconfigKEEP_ALIVE_TIMEOUT_TICKS            ( 1000 )

#define mqttconfigMQTT_TASK_MAX_BLOCK_TICKS           ( 100 )

#define mqttconfigMQTT_TASK_STACK_DEPTH    ( configMINIMAL_STACK_SIZE * 8 )
#define mqttconfigMQTT_TASK_PRIORITY       ( tskIDLE_PRIORITY + 5 )

#define mqttconfigMAX_BROKERS            ( 2 )

#define mqttconfigMAX_PARALLEL_OPS       ( 5 )

#define mqttconfigTCP_SEND_TIMEOUT_MS    ( 2000 )

#define mqttconfigRX_BUFFER_SIZE         ( 1024 + 128 )
~~~

Your help is much appreciated,
Péter

dachalco, 2020-12-16:
Hi @bitpet,

Thanks for the bringing this to our attention. In tandem with the info you provided, I was able to locally reproduce your issue and created/merged a fix in the above PR. Could you please pull in these updates and see if your issue persists?

dachalco, 2020-12-28:
Hi @bitpet 

Closing as there is a fix merged for this demo. Please merge the the changes in the PR above and reopen if this particular issue persists. Happy holidays!

closed 2020-12-28 by dachalco