aws/amazon-freertos #2868: cmake command line build error with -DVENDOR=cypress -DBOARD=CY8CKIT_064S0S2_4343W

taotaodong527, 2020-12-11:
**Describe the bug**
cmake command line build error with -DVENDOR=cypress -DBOARD=CY8CKIT_064S0S2_4343W
 C : ERROR : [Errno 2] No such file or directory: 'amazonfreertos/vendors/cypress/MTB/psoc6/psoc64tfm/security/keys/**TFM_S_KEY_PRIV.pem**'. Check the log for details
Error: Failed processing!
**System information**
- VENDOR=cypress -DBOARD=CY8CKIT_064S0S2_4343W
- NONE
- Linux
- 202011.00-161-g650cad85f

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots or console output**
cmake -DVENDOR=cypress -DBOARD=CY8CKIT_064S0S2_4343W -DCOMPILER=arm-gcc -S . -B cypress_build
====================Configuration for FreeRTOS====================
  Version:                 202011.00
  Git version:             202011.00-161-g650cad85f

Target microcontroller:
  vendor:                  Cypress
  board:                   CY8CKIT_064S0S2_4343W
  description:             PSoC 64 Wi-Fi BT Pioneer Kit
  family:                  PSoC 6
  data ram size:           1024KB
  program memory size:     2MB

Host platform:
  OS:                      Linux-5.4.0-56-generic
  Toolchain:               arm-gcc
  Toolchain path:          /usr
  CMake generator:         Unix Makefiles

FreeRTOS modules:
  Modules to build:        backoff_algorithm, ble, ble_hal, ble_wifi_provisioning,
                           common, core_http, core_http_demo_dependencies, core_json,
                           core_mqtt, core_mqtt_demo_dependencies, crypto, defender,
                           dev_mode_key_provisioning, device_defender, device_defender_
                           demo_dependencies, device_shadow, device_shadow_demo_
                           dependencies, freertos_plus_cli, greengrass, http_demo_
                           helpers, https, jobs, jobs_demo_dependencies, kernel, logging,
                           mqtt, mqtt_demo_helpers, pkcs11, pkcs11_helpers,
                           pkcs11_implementation, pkcs11_utils, platform, secure_sockets,
                           serializer, shadow, tls, transport_interface_secure_sockets,
                           wifi
  Enabled by user:         ble, ble_hal, ble_wifi_provisioning, core_http_demo_
                           dependencies, core_json, core_mqtt_demo_dependencies,
                           defender, device_defender, device_defender_demo_dependencies,
                           device_shadow, device_shadow_demo_dependencies, freertos_plus_
                           cli, greengrass, https, jobs, jobs_demo_dependencies, logging,
                           pkcs11, pkcs11_helpers, pkcs11_implementation, pkcs11_utils,
                           platform, secure_sockets, shadow, wifi
  Enabled by dependency:   backoff_algorithm, common, core_http, core_mqtt, crypto,
                           demo_base, dev_mode_key_provisioning, freertos, http_demo_
                           helpers, kernel, mqtt, mqtt_demo_helpers, pkcs11_psa,
                           secure_sockets_lwip, serializer, tls, transport_interface_
                           secure_sockets, utils
  3rdparty dependencies:   jsmn, lwip, lwip_osal, mbedtls, pkcs11, tinycbor
  Available demos:         demo_core_http, demo_core_mqtt, demo_device_defender,
                           demo_device_shadow, demo_gatt_server, demo_greengrass_
                           connectivity, demo_jobs, demo_mqtt_ble, demo_numeric_
                           comparison, demo_shadow_ble, demo_tcp, demo_wifi_provisioning
  Available tests:
=========================================================================

-- Configuring done
-- Generating done
-- Build files have been written to: /home/dongtaotao/freeRTOS_with_git/amazon-freertos/cypress_build

cmake --build cypress_build/
[ 98%] Linking C executable aws_demos.elf
2020-12-11 13:52:26,060 : **C : **ERROR : [Errno 2] No such file or directory: '/home/dongtaotao/freeRTOS_with_git/amazon-freertos/vendors/cypress/MTB/psoc6/psoc64tfm/security/keys/TFM_S_KEY_PRIV.pem'. Check the log for details****
Error: Failed processing!
CMakeFiles/aws_demos.dir/build.make:4041: recipe for target 'aws_demos.elf' failed
make[2]: *** [aws_demos.elf] Error 1
make[2]: *** Deleting file 'aws_demos.elf'
CMakeFiles/Makefile2:1759: recipe for target 'CMakeFiles/aws_demos.dir/all' failed
make[1]: *** [CMakeFiles/aws_demos.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2

**To reproduce**
Steps to reproduce the behavior:

1. cmake -DVENDOR=cypress -DBOARD=CY8CKIT_064S0S2_4343W -DCOMPILER=arm-gcc -S . -B cypress_build
2. cmake --build cypress_build/

Thank you!

mingyue86010, 2020-12-11:
Hi @taotaodong527 ,

Thank you for reaching out. This is actually not a bug. We removed the default key pairs from our repo due to the securty concern (exposing the private key to public). We think it's not a good guidence to customer. Since people may use the default keys into the product witout aware of the risk. So we enforce users that using Cypress CY8CKIT_064S0S2_4343W to re-generate a key pair and provision the board following the Getting Started Guide and Cypress board provisioning guideline.

There are two ways to make your board workable:
1. You can check the <freertos>/vendors/cypress/MTB/psoc6/psoc64tfm/security folder's histroy on the repo and find out the default keys pair. Copy thoes keys into the corresponding location at your local repo. But please ensure ONLY USE IT FOR DEBUGGING purpose.

2. The seconds way is what we most suggest. Generate a new key pairs and re-provisioning the board. Here is how to do that:
a. Go to the CY8CKIT_064S0S2_4343W Getting Started Guide:
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_cypress_psoc64.html

b. In section Setting up your hardware --> Provision your kit:
Find the PDF document "Provisioning Guide for the Cypress CY8CKIT-064S0S2-4343W Kit"

c. In the document:
Follow the steps in section 1.2 "Provisioning the kit", and DO NOT skip the step 4 "Create a new signing key pair(s)", though it marked as "optional".

d. After you done the keys will be generated into "<freertos>/vendors/cypress/MTB/psoc6/psoc64tfm/security/keys/" folder.

Please let me know if you have any further questions.

Thank you!

Regards,
Ming

taotaodong527, 2020-12-14:
Hi @mingyue86010 ：
Thanks for your quick feedback. It's reasonable not exposing the private key to public :), got your point.
Following your clear guidance, I could sign the image the build process goes well.

closed 2020-12-14 by taotaodong527