aws/amazon-freertos #2877: Is it possible to enable Secure Boot and Flash Encryption for ESP32 using AWS FreeRTOS project?

octopusbunch, 2020-12-12:
In espressif's esp-idf documentation for [Flash Encryption](https://docs.espressif.com/projects/esp-idf/en/v3.3.2/security/flash-encryption.html#reflashing-via-pregenerated-flash-encryption-key) there is mention of a "Security Features" menu option that should come up when running `.\vendors\espressif\esp-idf\tools\idf.py menuconfig` but I don't see it.

And in the documentation for [Secure Boot](https://docs.espressif.com/projects/esp-idf/en/v3.3.2/security/secure-boot.html#how-to-enable-secure-boot) it mentions that there should be a "Secure Boot Configuration" menu after running `.\vendors\espressif\esp-idf\tools\idf.py menuconfig` but I don't see that either.

Is there any other way to configure and enable the Flash Encryption and Secure Boot features for my esp32 based project or will I have to take a specific version of esp-idf with these features already configured and enabled and link it into my aws freertos project?

shubhamkulkarni97, 2020-12-15:
Hi @octopusbunch,

I'm looking into this issue and will get back to you soon.

octopusbunch, 2020-12-15:
Hi @shubhamkulkarni97,

Thank you, I look forward to any feedback or help you can give me.

shubhamkulkarni97, 2020-12-17:
Hi @octopusbunch,

I have created a [PR](https://github.com/aws/amazon-freertos/pull/2897) to enable security features with CMake. Please try the changes and let me know if it works for you.

Thanks,
Shubham

octopusbunch, 2020-12-18:
Hi @shubhamkulkarni97,

Thank you for the fix, I can see the "Security Features" options show up now with the menuconfig. 

But I'm wondering what is the correct version of the esp-idf documentation to reference for the esp-afr-sdk submodule in amazon freertos? I had been using version 3.3.2, but I see there is differences between what that documentation says and the actual menuconfig options. Thank you.

shubhamkulkarni97, 2020-12-21:
@octopusbunch, esp-afr-sdk submodule is based on ESP-IDF v3.3.2.

What differences did you notice in the documentation and menuconfig?

octopusbunch, 2020-12-22:
Thanks @shubhamkulkarni97,

Thank you for the confirmation. In the Secure Boot documentation for [v3.3.2](https://docs.espressif.com/projects/esp-idf/en/v3.3.2/security/secure-boot.html#how-to-enable-secure-boot) the instructions say to enable Secure Boot under Bootloader Config menu after running menuconfig, but those options for Secure Boot seem to be under Security Features now. "Run make menuconfig, navigate to “Secure Boot Configuration” and select the option “One-time Flash”.

Since esp-afr-sdk is based on esp-idf v3.3.2, does that mean that Secure Boot V2 is not supported with esp-afr-sdk? I would like to take advantage of that feature since I am using an ESP32 V3 for my application.

shubhamkulkarni97, 2020-12-22:
@octopusbunch,
> Thank you for the confirmation. In the Secure Boot documentation for [v3.3.2](https://docs.espressif.com/projects/esp-idf/en/v3.3.2/security/secure-boot.html#how-to-enable-secure-boot) the [...]

Documentation states that `Pay particular attention to the “Bootloader Config” options, as you can only flash the bootloader once`. Because in One-time Flash mode you cannot flash bootloader again.

It doesn't state that these options are under Bootloader Config.
> 
> Since esp-afr-sdk is based on esp-idf v3.3.2, does that mean that Secure Boot V2 is not supported with esp-afr-sdk? I would like to take advantage of that feature since I am using an ESP32 V3 for my application.

Yes, Secure Boot V2 is not supported in esp-afr-sdk currently.
However, we already have a PR to update esp-afr-sdk to ESP-IDF v4.2 release. You can try Secure Boot V2 in PR https://github.com/aws/amazon-freertos/pull/2893

Hope this helps!

octopusbunch, 2020-12-22:
Hi @shubhamkulkarni97,
Sorry I made a mistake on my earlier post, let me correct my response. Step 1 under [How to Enable Secure Boot in docs v3.3.2](https://docs.espressif.com/projects/esp-idf/en/v3.3.2/security/secure-boot.html#how-to-enable-secure-boot) reads `Run make menuconfig, navigate to “Secure Boot Configuration” and select the option “One-time Flash”. (To understand the alternative “Reflashable” choice, see Re-Flashable Software Bootloader.)`

But I didn't see a "Secure Boot Configuration" option in the main menu of menuconfig, it looks like the options for Secure Boot are under "Security features" instead. It's not really a big deal though because I was still able to find it.

Anyways , thanks for the PR. I will try using that version instead for Secure Boot V2.

mingyue86010, 2020-12-22:
The issue looks resolved by PR https://github.com/aws/amazon-freertos/pull/2897 which is a fix for ESP-IDF v3.3.

For the Secure Boot V2, it will be available in ESP-IDF v4.2 once PR #2893 get merged.

closed 2020-12-22 by mingyue86010