aws/amazon-freertos #2880: cmake -DVENDOR=nordic -DBOARD=nrf52840-dk -DCOMPILER=arm-gcc -S . -B nordic_build failed

taotaodong527, 2020-12-14:
**Describe the bug**
cmake -DVENDOR=nordic -DBOARD=nrf52840-dk -DCOMPILER=arm-gcc -S . -B nordic_build failed:
~/freeRTOS_with_git/amazon-freertos$ cmake --build nordic_build/ --clean-first
[  0%] Building C object CMakeFiles/afr_kernel.dir/freertos_kernel/event_groups.c.obj
arm-none-eabi-gcc: error: unrecognized command line option '-quiet'
CMakeFiles/afr_kernel.dir/build.make:82: recipe for target 'CMakeFiles/afr_kernel.dir/freertos_kernel/event_groups.c.obj' failed
make[2]: *** [CMakeFiles/afr_kernel.dir/freertos_kernel/event_groups.c.obj] Error 1
CMakeFiles/Makefile2:1292: recipe for target 'CMakeFiles/afr_kernel.dir/all' failed
make[1]: *** [CMakeFiles/afr_kernel.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2

**System information**
- nrf52840-dk
- NONE
- Linux
- 202011.00-161-g650cad85f
- amr-gcc version as below:
**arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620**
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE

**Expected behavior**
The build should be successfully excuted

**Screenshots or console output**
:~/freeRTOS_with_git/amazon-freertos$ **cmake --build nordic_build/ --clean-first**
Scanning dependencies of target afr_kernel
[  0%] Building C object CMakeFiles/afr_kernel.dir/freertos_kernel/event_groups.c.obj
arm-none-eabi-gcc: error: **unrecognized command line option '-quiet'**
CMakeFiles/afr_kernel.dir/build.make:82: recipe for target 'CMakeFiles/afr_kernel.dir/freertos_kernel/event_groups.c.obj' failed
make[2]: *** [CMakeFiles/afr_kernel.dir/freertos_kernel/event_groups.c.obj] Error 1
CMakeFiles/Makefile2:1292: recipe for target 'CMakeFiles/afr_kernel.dir/all' failed
make[1]: *** [CMakeFiles/afr_kernel.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2

**To reproduce**
1. cmake -DVENDOR=nordic -DBOARD=nrf52840-dk -DCOMPILER=arm-gcc -S . -B nordic_build
2. cmake --build nordic_build/ --clean-first

**Additional context**
Briefly, there are 2 issues for me when build nordic with command-line 
1.**unrecognized command line option '-quiet'**
2. After **comment out the --quiet** option vendors/nordic/boards/nrf52840-dk/bootloader.cmake and vendors/nordic/boards/nrf52840-dk/CMakeLists.txt, the build still failed with missing inclue files

Thank you!

lundinc2, 2020-12-14:
Hello @taotaodong527,

Thank you for this report, I am investigating your issue.

lundinc2, 2020-12-14:
Hi @taotaodong527,

Can you try using `-DCOMPILER=arm-segger` instead of `-DCOMPILER=arm-gcc`

taotaodong527, 2020-12-15:
Hi @lundinc2 ：
After install the Setup_EmbeddedStudio_ARM_v532a_linux_x64.tar.gz
With below steps：
1. cmake -DVENDOR=nordic -DBOARD=nrf52840-dk **-DCOMPILER=arm-segger** -DAFR_TOOLCHAIN_PATH=/usr/share/segger_embedded_studio_for_arm_5.32a/gcc/arm-none-eabi/bin -S . -B nordic_build
2. cmake --build nordic_build/
**Build failed with below logs:**
[ 48%] Building C object CMakeFiles/bootloader.dir/vendors/nordic/nRF5_SDK_15.2.0/components/libraries/crypto/backend/cc310_bl/cc310_bl_backend_shared.c.obj
**make[2]: *** No rule to make target '/usr/share/segger_embedded_studio_for_arm_5.32a/gcc/arm-none-eabi/bin/../../../lib/libdebugio_mempoll_v7em_fpv4_sp_d16_hard_t_le_eabi.a', needed by 'bootloader.hex'.  Stop.**
CMakeFiles/Makefile2:1374: recipe for target 'CMakeFiles/bootloader.dir/all' failed
make[1]: *** [CMakeFiles/bootloader.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2

taotaodong527, 2020-12-15:
Hi @lundinc2 :
Following https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_nordic.html this guide.
It works for me to build and fuse the aws_demos.emProject using the **SEGGER Embedded Studio**
Unfortunately, the commad line buid still not works for me (both -DCOMPILER=arm-gcc and -DCOMPILER=arm-segger)

lundinc2, 2020-12-21:
Hi @taotaodong527,

Here is a PR that fixes this issue https://github.com/aws/amazon-freertos/pull/2902

taotaodong527, 2020-12-23:
Hi @lundinc2:
Thanks. I will feedback soon after check with your PR.

taotaodong527, 2020-12-23:
Hi @lundinc2:
PR #2902 works as expected in my side

taotaodong527, 2020-12-24:
Hi @lundinc2:
I have to reopen this issue. Seems PR #2902 only fixes the **-DCOMPILER=arm-segger** compile issue。

But when using -DCOMPILER=arm-gcc，below error still reported：
~/freeRTOS_with_git/amazon-freertos$ cmake -DVENDOR=nordic -DBOARD=nrf52840-dk **-DCOMPILER=arm-gcc** -S . -B nordic_build
-- The C compiler identification is GNU 6.3.1
-- The CXX compiler identification is GNU 6.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Found Git: /usr/bin/git (found version "2.17.1")
-- Submodule update
Skipping submodule '../../../libraries/abstractions/backoff_algorithm/'
Skipping submodule '../../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/abstractions/pkcs11/corePKCS11/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreHTTP/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreJSON/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/coreMQTT/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_defender_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../../libraries/device_shadow_for_aws/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../libraries/freertos_plus/standard/freertos_plus_tcp/'
Skipping submodule '../../../libraries/jobs_for_aws/'
Skipping submodule '../../../libraries/jobs_for_aws/'
Skipping submodule '../../../libraries/jobs_for_aws/'
=========================Resolving dependencies==========================
module disabled: wifi
reason:          wifi::mcu_port is not defined by vendor.
dependency path: ble_wifi_provisioning->wifi->wifi::mcu_port

module disabled: secure_sockets
reason:          secure_sockets::mcu_port is not defined by vendor.
dependency path: https->core_http_demo_dependencies->http_demo_helpers->transport_
                 interface_secure_sockets->secure_sockets->secure_sockets::mcu_port

module disabled: pkcs11_implementation
reason:          pkcs11_implementation::mcu_port is not defined by vendor.
dependency path: https->core_http_demo_dependencies->http_demo_helpers->pkcs11_
                 helpers->pkcs11_implementation->pkcs11_implementation::mcu_port

module disabled: common_io
reason:          common_io::mcu_port is not defined by vendor.
dependency path: freertos_cli_plus_uart->common_io->common_io::mcu_port

====================Configuration for FreeRTOS====================
  Version:                 202012.00
  Git version:             202012.00-188-ga75739e71

Target microcontroller:
  vendor:                  Nordic
  board:                   nRF52840 DK
  description:             Nordic nrf52840-dk Development Kit
  family:                  NRF52
  data ram size:           UNKNOWN
  program memory size:     UNKNOWN

Host platform:
  OS:                      Linux-5.4.0-58-generic
  Toolchain:               arm-gcc
  Toolchain path:          /usr
  CMake generator:         Unix Makefiles

FreeRTOS modules:
  Modules to build:        backoff_algorithm, ble, ble_hal, common, core_json, core_mqtt,
                           core_mqtt_demo_dependencies, crypto, device_defender,
                           device_shadow, device_shadow_demo_dependencies, freertos_plus_
                           cli, jobs, kernel, logging, mqtt, ota, pkcs11_utils, platform,
                           posix, serializer, shadow
  Enabled by user:         ble, ble_hal, core_json, core_mqtt_demo_dependencies,
                           device_defender, device_shadow, device_shadow_demo_
                           dependencies, freertos_plus_cli, jobs, logging, ota,
                           pkcs11_utils, platform, posix, shadow
  Enabled by dependency:   backoff_algorithm, common, core_mqtt, crypto, demo_base,
                           freertos, freertos_plus_posix, kernel, mqtt, ota_mqtt,
                           serializer
  3rdparty dependencies:   jsmn, mbedtls, pkcs11, tinycbor, tinycrypt
  Available demos:         demo_gatt_server, demo_mqtt_ble, demo_numeric_comparison,
                           demo_ota, demo_posix, demo_shadow_ble
  Available tests:
=========================================================================

-- Configuring done
-- Generating done
-- Build files have been written to: /home/dongtaotao/freeRTOS_with_git/amazon-freertos/nordic_build
@-OptiPlex-5070:~/freeRTOS_with_git/amazon-freertos$ **cmake --build nordic_build/ --clean-first**
Scanning dependencies of target afr_kernel
[  0%] Building C object CMakeFiles/afr_kernel.dir/freertos_kernel/event_groups.c.obj
**arm-none-eabi-gcc: error: unrecognized command line option '-quiet'**
CMakeFiles/afr_kernel.dir/build.make:82: recipe for target 'CMakeFiles/afr_kernel.dir/freertos_kernel/event_groups.c.obj' failed
make[2]: *** [CMakeFiles/afr_kernel.dir/freertos_kernel/event_groups.c.obj] Error 1
CMakeFiles/Makefile2:1292: recipe for target 'CMakeFiles/afr_kernel.dir/all' failed
make[1]: *** [CMakeFiles/afr_kernel.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2
dongtaotao@dongtaotao-OptiPlex-5070:~/freeRTOS_with_git/amazon-freertos$

lundinc2, 2021-01-11:
Hi @taotaodong527,

Currently we only host this arm-segger project, which is a modified version of arm-gcc.

closed 2021-02-01 by aggarw13