aws/amazon-freertos #2892: OTA, signing, and BootImageTrailer?

apullin, 2020-12-16:
**Is your feature request related to a problem? Please describe.**  

This is really just a request for clarification and state-of-support of a feature, specifically on the operation of image signing, tools in the repo.

**Describe the solution you would like.**  

Is the BootImageTrailer containing the signature (and length and digest type metadata) supposed to be delivered along with the application binary (prepended with the BootImageDescriptor)?

The utility `ota_iamge_generator.py` does not seem to generate or append a trailer.

A trailer is generated and appended in `factory_image_generator.py`, but that only appends the trailer to the factory image, and is not included in the OTA image.

Doesn't the signature in the trailer have to change each time the contents of the application binary changes?

pvyawaha, 2020-12-17:
Hello,

`ota_iamge_generator.py` is provided to generate firmware files that are uploaded while creating the Over-the-Air update job. The script generates the image descriptor which is prepended to the firmware binary and has information like sequence number, start address, end address, execution address, hardware ID etc. 

Please note the image generated above is not signed and code signing is done after the image is uploaded while creating OTA job. The default Mplab IDE project provided runs this script as a post build step to generate OTA image with updated header.

`factory_image_generator.py ` is provided as an example to sign the initial/factory image on the device. This appends trailer to the image and contains the signature type, length and signature. This helps in enabling crypto signature verification by bootloader even on the factory images. 

More details regarding bootloader - https://docs.aws.amazon.com/freertos/latest/userguide/microchip-bootloader.html

OTA Prereqs - https://docs.aws.amazon.com/freertos/latest/userguide/ota-prereqs.html

Code Signing - https://docs.aws.amazon.com/freertos/latest/userguide/ota-code-sign-cert.html

Please let me know if you have more questions.

apullin, 2020-12-22:
Interesting. While porting and extending the bootloader project (for an STM32L4, but starting from the PIC32 example), I wanted to be able to verify the operation of the bootloader for an OTA binary, but without having to actually download the binary back from AWS.

I ended up refactoring the `ota_image_generator.py` to generate the signature and add the trailer there, and not in the `factory_image_generator.py`.
Although I realized that this will leave me with no way to generate a with-header/without-trailer binary that I would need to send up to AWS to use the in-cloud binary signing facility.

pvyawaha, 2020-12-29:
Do you have any more question regarding this topic?

apullin, 2020-12-29:
nope

closed 2020-12-29 by apullin