aws/amazon-freertos #2894: Does Amazon FreeRTOS supports development with Arduino IDE or PlatformIO?

Erickrk, 2020-12-16:
Hello guys, I've been looking into the basic tutorials on AWS and some YouTube videos too but failed at finding guidance on how to write my own code and then run it on top of FreeRTOS. I'm using an ESP32-DevKitC.

I've done this tutorial [https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html](url) but I haven't found the folder where my ESP's firmware is written, meaning that I don't even know where the app file is.

I'm aware that FreeRTOS is written in C, but it would be very much appreciated for prototyping reasons if I were able to reuse my cpp code and Arduino libraries in these FreeRTOS projects.

For me, this is a very basic feature and should be supported by the operational system, but I will understand if this solution only supports low-level development code. I wonder too if there's at least a way to use ESP's idf + Arduino Libraries with FreeRTOS.
Anyway, I'm just looking for guidance, this information has been kind of hard finding online so maybe someone here can help more. 
Every tip is more than welcome!

rashedtalukder, 2020-12-16:
Hi @Erickrk,

First, I recommend posting questions like these on the [FreeRTOS Community Forums](https://forums.freertos.org) since it's not a bug or feature request. For your needs specifically, you will find that the Arduino framework you are using on your ESP-32 by Espressif actually has the ESP-IDF (FreeRTOS w/ dual Xtensa core SMP support) underneath with wrappers. The loop() function actually is a FreeRTOS task (with low priority) and you can mix FreeRTOS API usage with your CPP libraries. You can use the [AWS IoT C++](https://docs.aws.amazon.com/iot/latest/developerguide/iot-sdks.html) SDK for connectivity to AWS and that would be a simpler path if you are looking to keep your application in a Arduino-like paradigm.

Erickrk, 2020-12-16:
Hey @rashedtalukder, thank you so much for your answer and information!

I'm going to make the question in the forum and close this issue.

closed 2020-12-16 by Erickrk