aws/amazon-freertos #2901: Lwip + sys compilation errors

ChristyBrouwer, 2020-12-21:
**Describe the bug**
Fatal error in including lwip and sys libraries in AWS FreeRTOS base code, different errors in windows vs mac. In a custom application, both my colleague and I are running into compilation errors within the AWS libraries. Windows has issues finding sys libraries (and in earlier freertos versions lwip libraries as well), Mac has issues finding lwip libraries. Own code runs via BLE and isn't using any wifi configurations at the moment. 

**System information**
Windows -> ESP32
Mac -> ESP32

Both running FreeRTOS version 202012.00-186-g3d41cf92c

Custom Application running via BLE, no relevant code snippets for issue

**Expected behavior**
Expected behavior is running the code and possibly getting errors in custom application, not in the base FreeRTOS code.

**Screenshots or console output**
Windows:
In file included from ../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/httpd_main.c:24:0:
../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/esp_httpd_priv.h:20:24: fatal error: sys/socket.h: No such file or directory
compilation terminated.

Mac:
In file included from ../vendors/espressif/esp-idf/components/wifi_provisioning/src/wifi_config.c:23:0:
../vendors/espressif/esp-idf/components/wifi_provisioning/include/wifi_provisioning/wifi_config.h:18:23: fatal error: lwip/inet.h: No such file or directory
compilation terminated.

**To reproduce**
Steps to reproduce the behavior:

1. Have latest FreeRTOS commit with up to date submodules
2. Compile

Thank you in advance for helping me figure this out!

dachalco, 2020-12-21:
Hi @ChristyBrouwer 

Can you please elaborate on how you build the code and arrive at this error?
I'm unable to reproduce the issue. Instead, my esp32 `aws_demos` builds succeed

It looks like the LWIP includes are not being forwarded to those IDF component builds. Both `sys/socket.h` and `lwip/inet.h` should be sourced from applications LWIP stack.

ChristyBrouwer, 2020-12-22:
Hey @dachalco 

Please note it's a custom application, I am not using the `aws_demos`. As for building, I build the application using CLI cmake. I've used the following guide for setting up this project including instructions on how to build and flash it: [AWS getting started with FreeRTOS](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html )

Hope this gives some more insight

dachalco, 2020-12-29:
Hi @ChristyBrouwer 

To clarify, is your application structure/build based on guide section "_Using FreeRTOS in your own CMake project for ESP32_"?
i.e. Does it build out-of-tree like [this reference repo](https://github.com/tgsong/amazon-freertos-examples)? If so, are you able to post your `CMakeLists.txt`?

Based on what the app uses, a wifi dependency may be imposed. For example, `NetworkManager`'s wifi provisioning uses ble _and_ wifi and may require both to build.

ChristyBrouwer, 2020-12-30:
Hey @dachalco 

Yes it is based on the mentioned guide section, which `CMakeLists.txt` would you like me to post? Each component has in the project has its own, as required. I've pasted the general `CMakeLists.txt` in the following [gist](https://gist.github.com/ChristyBrouwer/e002506630daca8c19ade4bf77e2a822). 

As for the `NetworkManager` component mentioned in the gist, there's no references to wifi code in there, just to be sure I do have `target_link_libraries(${COMPONENT_TARGET} PRIVATE AFR::wifi)` in its `CMakeLists.txt`, I have never had trouble with it nor did I change anything in it.

Hope this helps. Let me know if I need to give any further information. Hope you have good holidays!

dachalco, 2020-12-30:
@ChristyBrouwer 

Are you referring to `NetworkManager` as defined in `components/network_manager` or as defined in `amazon-freertos/demos/network_manager`? Please note the latter had recent changes that add softap wifi provisioning for esp32 -- this uses wifi. 

If these changes are _not_ included, at least AFR would not need to build `esp_http_server` component -- which depends on LWIP. More importantly, FreeRTOS+TCP is selected in root `CMakeLists.txt`, so even if the changes _were included_ the LWIP dependent files/functionality should be excluded from build.

Do you know which build target attempted to build these files? If it's a static library target, it will be much easier for me to isolate the issue and reproduce locally.

Finally, there is a [related ticket](https://github.com/aws/amazon-freertos-ble-android-sdk/issues/38) which has a [proposed fix](https://github.com/aws/amazon-freertos/pull/2895) . Did this build issue occur after _exclusively_ including the fix? The fix itself should not have imposed any new dependencies, but instead alters how subscriptions are forwarded from MCU to phone. I don't suspect it's responsible for this build issue.

And thank you! Happy Holidays to you too!

ChristyBrouwer, 2021-01-04:
@dachalco 
Happy new year!

I am referring to the one defined in `components/network_manager`. 

As for the build target. I've pasted the whole build error below, let me know if this is enough or if you need more information. At the moment it gives the `lwip` error, I'm unsure if this means one of the updates has fixed the `sys` error or if it just failed `lwip` first and terminated on that immediately.

```
[1/847] Building C object amazon-freertos/CMakeFiles/afr_wifi.dir/vendors/espressif/boards/esp32/ports/wifi/iot_softap_wifi_provisioning.c.obj
FAILED: amazon-freertos/CMakeFiles/afr_wifi.dir/vendors/espressif/boards/esp32/ports/wifi/iot_softap_wifi_provisioning.c.obj
C:\ESP-IDFTools\tools\bin\xtensa-esp32-elf-gcc.exe -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3.2-241-g1d2d93acd\" -I../aws-configs -I../amazon-freertos/libraries/abstractions/wifi/include -I../amazon-freertos/libraries/c_sdk/standard/common/include -I../amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/include -I../amazon-freertos/freertos_kernel/include -I../amazon-freertos/vendors/espressif/esp-idf/components/app_update/include -I../amazon-freertos/vendors/espressif/esp-idf/components/bootloader_support/include -I../amazon-freertos/vendors/espressif/esp-idf/components/esp32/include -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_event/include -I../amazon-freertos/vendors/espressif/esp-idf/components/efuse/include -I../amazon-freertos/vendors/espressif/esp-idf/components/driver/include -I../amazon-freertos/vendors/espressif/esp-idf/components/heap/include -I../amazon-freertos/vendors/espressif/esp-idf/components/log/include -I../amazon-freertos/vendors/espressif/esp-idf/components/nvs_flash/include -I../amazon-freertos/vendors/espressif/esp-idf/components/pthread/include -I../amazon-freertos/vendors/espressif/esp-idf/components/newlib/include -I../amazon-freertos/vendors/espressif/esp-idf/components/newlib/platform_include -I../amazon-freertos/vendors/espressif/esp-idf/components/soc/esp32/include -I../amazon-freertos/vendors/espressif/esp-idf/components/soc/include -I../amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/include -I../amazon-freertos/vendors/espressif/esp-idf/components/vfs/include -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_ringbuf/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos_tcpip/ethernet/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos_tcpip/smartconfig_ack/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos_tcpip/tcpip_adapter/include -I../amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/portable/Compiler/GCC -I../amazon-freertos/vendors/espressif/boards/esp32/aws_demos/config_files/default_pkcs11_config -I../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/include -I../amazon-freertos/demos/include -I../amazon-freertos/vendors/espressif/boards/esp32/aws_demos/config_files -Iconfig -mlongcalls -mlongcalls -Og -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -MD -MT amazon-freertos/CMakeFiles/afr_wifi.dir/vendors/espressif/boards/esp32/ports/wifi/iot_softap_wifi_provisioning.c.obj -MF amazon-freertos\CMakeFiles\afr_wifi.dir\vendors\espressif\boards\esp32\ports\wifi\iot_softap_wifi_provisioning.c.obj.d -o amazon-freertos/CMakeFiles/afr_wifi.dir/vendors/espressif/boards/esp32/ports/wifi/iot_softap_wifi_provisioning.c.obj -c ../amazon-freertos/vendors/espressif/boards/esp32/ports/wifi/iot_softap_wifi_provisioning.c
../amazon-freertos/vendors/espressif/boards/esp32/ports/wifi/iot_softap_wifi_provisioning.c:17:22: fatal error: lwip/err.h: No such file or directory
compilation terminated.
```

The related ticket with the proposed fix are also mine, the build issue happened before including the fix, although it also still occurs with the fix included.

dachalco, 2021-01-04:
Hi @ChristyBrouwer 

For the `lwip` build error, could you locally modify the [code block at amazon-freertos/vendors/espressif/boards/esp32/CMakeLists.txt:180](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/CMakeLists.txt#L180), to look like the following:
```cmake
if(AFR_ESP_FREERTOS_TCP)
target_sources(
    AFR::wifi::mcu_port
    INTERFACE 
        "${afr_ports_dir}/wifi/iot_wifi.c"
)
else()
target_sources(
    AFR::wifi::mcu_port
    INTERFACE 
        "${afr_ports_dir}/wifi/iot_wifi.c"
        "${afr_ports_dir}/wifi/iot_softap_wifi_provisioning.c"
        # Due to nghttp's dependency on it, http_parser is added to generate metadata
        # so that it is present in the code downloaded from the FreeRTOS console.
        "${AFR_ROOT_DIR}/libraries/coreHTTP/source/dependency/3rdparty/http_parser/http_parser.c"
        "${AFR_ROOT_DIR}/libraries/coreHTTP/source/dependency/3rdparty/http_parser/http_parser.h"
)
endif()
```

Seeing that FreeRTOS+TCP is in use, the above should exclude building these wifi files that rely on LWIP stack. This should circumvent the current build isssue.

If the initial `sys/` build errors return, could you post similar failure output that shows build target which prompted the source build?

ChristyBrouwer, 2021-01-05:
Hey @dachalco 

The fix solved the LWIP issue! 

The initial `sys` build error returned so here is the output as requested. It's a double fatal error via the same file. 

```
[711/841] Building C object amazon-freertos/esp-idf/esp_http_server/CMakeFiles/idf_component_esp_http_server.dir/src/httpd_main.c.obj
FAILED: amazon-freertos/esp-idf/esp_http_server/CMakeFiles/idf_component_esp_http_server.dir/src/httpd_main.c.obj
C:\ESP-IDFTools\tools\bin\xtensa-esp32-elf-gcc.exe -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3.2-241-g1d2d93acd\" -I../aws-configs -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/include -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/port/esp32 -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/util -Iconfig -I../amazon-freertos/libraries/coreHTTP/source/dependency/3rdparty/http_parser -I../amazon-freertos/vendors/espressif/esp-idf/components/esp32/include -I../amazon-freertos/vendors/espressif/esp-idf/components/driver/include -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_ringbuf/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos_tcpip/tcpip_adapter/include -I../amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/include -I../amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/portable/Compiler/GCC -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_event/include -I../amazon-freertos/vendors/espressif/esp-idf/components/log/include -I../amazon-freertos/vendors/espressif/esp-idf/components/efuse/include -I../amazon-freertos/vendors/espressif/esp-idf/components/efuse/esp32/include -I../amazon-freertos/vendors/espressif/esp-idf/components/newlib/platform_include -I../amazon-freertos/vendors/espressif/esp-idf/components/newlib/include -I../amazon-freertos/vendors/espressif/esp-idf/components/vfs/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos/include 
-I../amazon-freertos/freertos_kernel/include -I../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/include -I../amazon-freertos/vendors/espressif/boards/esp32/aws_demos/config_files -I../amazon-freertos/vendors/espressif/esp-idf/components/heap/include -I../amazon-freertos/vendors/espressif/esp-idf/components/soc/esp32/include 
-I../amazon-freertos/vendors/espressif/esp-idf/components/soc/include -mlongcalls -mlongcalls -Og -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -MD -MT amazon-freertos/esp-idf/esp_http_server/CMakeFiles/idf_component_esp_http_server.dir/src/httpd_main.c.obj -MF amazon-freertos\esp-idf\esp_http_server\CMakeFiles\idf_component_esp_http_server.dir\src\httpd_main.c.obj.d -o amazon-freertos/esp-idf/esp_http_server/CMakeFiles/idf_component_esp_http_server.dir/src/httpd_main.c.obj -c ../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/httpd_main.c
In file included from ../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/httpd_main.c:24:0:
../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/esp_httpd_priv.h:20:24: fatal error: sys/socket.h: No such file or directory
compilation terminated.
[712/841] Building C object amazon-freertos/esp-idf/esp_http_server/CMakeFiles/idf_component_esp_http_server.dir/src/httpd_parse.c.obj
FAILED: amazon-freertos/esp-idf/esp_http_server/CMakeFiles/idf_component_esp_http_server.dir/src/httpd_parse.c.obj
C:\ESP-IDFTools\tools\bin\xtensa-esp32-elf-gcc.exe -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3.2-241-g1d2d93acd\" -I../aws-configs -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/include -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/port/esp32 -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/util -Iconfig -I../amazon-freertos/libraries/coreHTTP/source/dependency/3rdparty/http_parser -I../amazon-freertos/vendors/espressif/esp-idf/components/esp32/include -I../amazon-freertos/vendors/espressif/esp-idf/components/driver/include -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_ringbuf/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos_tcpip/tcpip_adapter/include -I../amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/include -I../amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/portable/Compiler/GCC -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_event/include -I../amazon-freertos/vendors/espressif/esp-idf/components/log/include -I../amazon-freertos/vendors/espressif/esp-idf/components/efuse/include -I../amazon-freertos/vendors/espressif/esp-idf/components/efuse/esp32/include -I../amazon-freertos/vendors/espressif/esp-idf/components/newlib/platform_include -I../amazon-freertos/vendors/espressif/esp-idf/components/newlib/include -I../amazon-freertos/vendors/espressif/esp-idf/components/vfs/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos/include 
[... cut ...] o amazon-freertos/esp-idf/esp_http_server/CMakeFiles/idf_component_esp_http_server.dir/src/httpd_parse.c.obj -c ../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/httpd_parse.c
In file included from ../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/httpd_parse.c:23:0:
../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/esp_httpd_priv.h:20:24: fatal error: sys/socket.h: No such file or directory
compilation terminated.
```

dachalco, 2021-01-12:
Unaware of your app's custom `./components` build files, I can't accurately comment on the dependency graph.
However, somewhere, a build dependency is drawn on `esp_http_server` when it's not needed nor should be built. 

Ideally, this dependency is fixed/removed. It may be easier to identify and locate by drawing your app's dependency graph with a tool like [`Graphviz`](https://gitlab.kitware.com/cmake/community/-/wikis/doc/cmake/Graphviz).

Alternatively, nullifying the `esp_http_server` component build might also be sufficient. For example, see this `esp_http_server` definition which overrides its standard component [here](https://github.com/aws/amazon-freertos/tree/master/vendors/espressif/boards/components/esp_http_server). For background details see this [ESP32 Forum On Removing IDF Components](https://esp32.com/viewtopic.php?t=9416)

ChristyBrouwer, 2021-01-15:
Hey @dachalco 

I'll double check, but we're pretty sure our custom `./components` don't have dependencies on `esp_http_server`. 
Apart from this, I'll also try nullifying, I'll get back to you when I have an update, probably early next week.

ChristyBrouwer, 2021-01-20:
Hi @dachalco

Quick update: I've double checked none of the custom `./components` are connecting to `esp_http_server`. I found some places where `target_link_libraries(${COMPONENT_TARGET} PRIVATE AFR::wifi` was in the makefile, but considering the project doesn't use that I removed them. After removing these, the error still continued. However, the output is now as follows

```
[11/73] Building C object amazon-freertos/esp-idf/esp_http_server/CMakeFiles/idf_component_esp_http_server.dir/src/httpd_main.c.obj
FAILED: amazon-freertos/esp-idf/esp_http_server/CMakeFiles/idf_component_esp_http_server.dir/src/httpd_main.c.obj
C:\ESP-IDFTools\tools\bin\xtensa-esp32-elf-gcc.exe -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3.2-241-g1d2d93acd\" -I../aws-configs -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/include -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/port/esp32 -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/util -Iconfig -I../amazon-freertos/libraries/coreHTTP/source/dependency/3rdparty/http_parser -I../amazon-freertos/vendors/espressif/esp-idf/components/esp32/include -I../amazon-freertos/vendors/espressif/esp-idf/components/driver/include -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_ringbuf/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos_tcpip/tcpip_adapter/include -I../amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/include -I../amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/portable/Compiler/GCC -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_event/include -I../amazon-freertos/vendors/espressif/esp-idf/components/log/include -I../amazon-freertos/vendors/espressif/esp-idf/components/efuse/include -I../amazon-freertos/vendors/espressif/esp-idf/components/efuse/esp32/include -I../amazon-freertos/vendors/espressif/esp-idf/components/newlib/platform_include -I../amazon-freertos/vendors/espressif/esp-idf/components/newlib/include -I../amazon-freertos/vendors/espressif/esp-idf/components/vfs/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos/include 
-I../amazon-freertos/freertos_kernel/include -I../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32_IDF3/include -I../amazon-freertos/vendors/espressif/boards/esp32/aws_demos/config_files -I../amazon-freertos/vendors/espressif/esp-idf/components/heap/include -I../amazon-freertos/vendors/espressif/esp-idf/components/soc/esp32/include 
-I../amazon-freertos/vendors/espressif/esp-idf/components/soc/include -mlongcalls -mlongcalls -Og -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -MD -MT amazon-freertos/esp-idf/esp_http_server/CMakeFiles/idf_component_esp_http_server.dir/src/httpd_main.c.obj -MF amazon-freertos\esp-idf\esp_http_server\CMakeFiles\idf_component_esp_http_server.dir\src\httpd_main.c.obj.d -o amazon-freertos/esp-idf/esp_http_server/CMakeFiles/idf_component_esp_http_server.dir/src/httpd_main.c.obj -c ../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/httpd_main.c
In file included from ../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/httpd_main.c:24:0:
../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/esp_httpd_priv.h:20:24: fatal error: sys/socket.h: No such file or directory
compilation terminated.
[12/73] Building C object amazon-freertos/esp-idf/esp_http_server/CMakeFiles/idf_component_esp_http_server.dir/src/httpd_uri.c.obj
FAILED: amazon-freertos/esp-idf/esp_http_server/CMakeFiles/idf_component_esp_http_server.dir/src/httpd_uri.c.obj
C:\ESP-IDFTools\tools\bin\xtensa-esp32-elf-gcc.exe -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3.2-241-g1d2d93acd\" -I../aws-configs -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/include -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/port/esp32 -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/util -Iconfig -I../amazon-freertos/libraries/coreHTTP/source/dependency/3rdparty/http_parser -I../amazon-freertos/vendors/espressif/esp-idf/components/esp32/include -I../amazon-freertos/vendors/espressif/esp-idf/components/driver/include -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_ringbuf/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos_tcpip/tcpip_adapter/include -I../amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/include -I../amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/portable/Compiler/GCC -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_event/include -I../amazon-freertos/vendors/espressif/esp-idf/components/log/include -I../amazon-freertos/vendors/espressif/esp-idf/components/efuse/include -I../amazon-freertos/vendors/espressif/esp-idf/components/efuse/esp32/include -I../amazon-freertos/vendors/espressif/esp-idf/components/newlib/platform_include -I../amazon-freertos/vendors/espressif/esp-idf/components/newlib/include -I../amazon-freertos/vendors/espressif/esp-idf/components/vfs/include -I../amazon-freertos/vendors/espressif/b [... cut ...]
In file included from ../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/httpd_parse.c:23:0:
../amazon-freertos/vendors/espressif/esp-idf/components/esp_http_server/src/esp_httpd_priv.h:20:24: fatal error: sys/socket.h: No such file or directory
compilation terminated.
```

After this I tried nullifying the `esp_http_server` component according to the example and forum, but couldn't get it working. The errors I get are files not being able to find the `esp_http_server` files, so I might be doing something wrong in nullifying it.

dachalco, 2021-01-26:
@ChristyBrouwer 

Just want to let you know I got response and should have a response by tomorrow.

dachalco, 2021-01-27:
Hi @ChristyBrouwer 

Could you please apply the `esp32/CMakeLists.txt` changes from this [pending PR](https://github.com/aws/amazon-freertos/pull/2951/files#diff-529e7fa4e93a9864d9151cd1d198436c6183152ba408c41e2d88d6f0c589e355)?

ChristyBrouwer, 2021-02-01:
Hey @dachalco 

To confirm, I did download the `amazon-freertos` from a git clone.

At this moment in time I am not able to set up dummy code to reproduce the error. Our code at this moment in time does not have any references to the wifi libraries, apart from including `#include "iot_ble_wifi_provisioning.h"` . However, we can't seem to link this back to `esp_http_server` unless it takes a very long route to get there. This would mean the problem might be including said file, any thoughts on that?

As a head's up: at the end of this week, my colleague @YTvW will take over this issue. So you'll be communicating with him then instead of me

paulbartell, 2021-12-10:
There have not been any new post to this issue in quite a while so I will close it. Feel free to re-open if this has not already been solved.

closed 2021-12-10 by paulbartell