aws/amazon-freertos #2906: All save network requests fail after sending a delete network request if there are no networks saved

dgranahan, 2020-12-22:
**Describe the bug**
If no networks are saved and a delete network request is issued to index 0, for example, the delete network request will succeed.

In https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/standard/ble/src/services/wifi_provisioning/iot_ble_wifi_provisioning.c `_deleteNetworkTask` calls `_popNetwork` which calls `WIFI_NetworkDelete`.

The implementation of `WIFI_NetworkDelete` for ESP32 (https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c#L1071-L1127) will always return success even with no networks saved. This is because `xRegistry.usNumNetworks` will be zero so `usIndex < xRegistry.usNumNetworks` will always be false so `eWiFiSuccess` will be returned.

Jumping back to `_popNetwork` if `WIFI_NetworkDelete` returns success it will decrement `wifiProvisioning.numNetworks`. This will cause `numNetworks` which is an unsigned int to underflow and become `INT_MAX`.

With `numNetworks` now at `INT_MAX`, any attempt to add a new network will fail in `_addNetworkTask` because of the max saved networks check.

```
        if( wifiProvisioning.numNetworks < IOT_BLE_WIFI_PROVISIONING_MAX_SAVED_NETWORKS )
        {
            ret = _addNewNetwork( &wifiProvisioning.addNetworkRequest.network,
                                  wifiProvisioning.addNetworkRequest.connect );
        }
        else
        {
            IotLogError( "Failed to add a new network, max networks limit (%d) reached", IOT_BLE_WIFI_PROVISIONING_MAX_SAVED_NETWORKS );
            ret = eWiFiFailure;
        }
```

**System information**
Espressif ESP32

**Expected behavior**
Issuing a delete network request even when no networks are saved should not break the ability to connect to another network.

**To reproduce**
Steps to reproduce the behavior:

1. Clear all networks so no networks are saved.
2. Issue a delete network request with index 0 which will return success.
3. Issue a save network request with connect true which will fail.

muneebahmed10, 2020-12-23:
Hi @dgranahan,

Thank you for bringing this to our attention. `numNetworks` should not be decremented when nothing is deleted, so I've created #2909 to fix this issue, and I'll let you know once it's been merged. Please let us know if you have any further questions.

Thank you

muneebahmed10, 2020-12-24:
Hi @dgranahan,

I'll close this issue as the PR has been merged.

closed 2020-12-24 by muneebahmed10