aws/amazon-freertos #2910: AFR & ESP32: coreHTTP demo generates CMake Error - list GET given empty list

dipen-1337lab, 2020-12-23:
**Describe the bug**
The AFR Version 202012.00 generates the below
**CMake Error at vendors/espressif/esp-idf/tools/cmake/scripts/expand_requirements.cmake:144 (list):
  list GET given empty list**

**System information**

- Hardware board: ESP32-DevKitC
- IDE used: N/A
- Operating System [Windows 10 x64 -- Version: 20H2 -- OS Build: 19042.685]
- Version of FreeRTOS: AFR Version 202012.00 downloaded from console.aws.amazon.com
- Project: coreHTTP demo (#define CONFIG_CORE_HTTP_MUTUAL_AUTH_DEMO_ENABLED)

**Expected behavior**
The below cmake command should execute without errors.
**cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DAFR_ESP_LWIP=1 -GNinja -S . -B build**

**Screenshots or console output**
Below is the console log

```
E:\FreeRTOS Device Software Configurations\coreHTTP_demo-1608663093143\FreeRTOS>cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DAFR_ESP_LWIP=1 -GNinja -S . -B build
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.29.2.windows.3")
-- Checking Python dependencies...
Python requirements from E:/FreeRTOS Device Software Configurations/coreHTTP_demo-1608663093143/FreeRTOS/vendors/espressif/esp-idf\requirements.txt are satisfied.
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
CMake Error at vendors/espressif/esp-idf/tools/cmake/scripts/expand_requirements.cmake:144 (list):
  list GET given empty list
Call Stack (most recent call first):
  vendors/espressif/esp-idf/tools/cmake/scripts/expand_requirements.cmake:199 (filter_components_list)

CMake Error at vendors/espressif/esp-idf/tools/cmake/idf_functions.cmake:339 (message):
  Failed to expand component requirements
Call Stack (most recent call first):
  vendors/espressif/boards/esp32/CMakeLists.txt:536 (idf_import_components)
  CMakeLists.txt:70 (include)

-- Configuring incomplete, errors occurred!
See also "E:/FreeRTOS Device Software Configurations/coreHTTP_demo-1608663093143/FreeRTOS/build/CMakeFiles/CMakeOutput.log".
See also "E:/FreeRTOS Device Software Configurations/coreHTTP_demo-1608663093143/FreeRTOS/build/CMakeFiles/CMakeError.log".

E:\FreeRTOS Device Software Configurations\coreHTTP_demo-1608663093143\FreeRTOS>r
```

**To reproduce**
Steps to reproduce the behaviour:

1. Login and download FreeRTOS from the FreeRTOS console.
2. Create a new software configuration with

- FreeRTOS Version - 202012.00 (Latest)
- Hardware Platform - ESP32 DevKitC
- Library - coreHTTP

3. Download and unzip to extract the folder contents.
4. Open Command Prompt.
5. Change the current working directory path to ..\FreeRTOS\
6. Run the cmake command and observe the error generated. Cmake command and logs are shared above.

**Code to reproduce the bug**
N/A

**Additional context**
N/A

Thank you!
Dipen

dipen-1337lab, 2020-12-23:
Namaste Community Member,

**Update:**
The same observation/ error holds true for **https_client_demo** downloaded from the AWS IoT Console with AFR Version 202002.00.

The **cmake** command is failing when attempting to generate the build-system before running the actual **ninja** command for code compilation. Also please note, the cmake command is run **"as is"** on the demo project downloaded, w/o any file modifications.

Thanks | Regards,
Dipen

mingyue86010, 2020-12-30:
Hi @dipen-1337lab,

I followed your step and downloaded the code from the console. But it works with my local setup. 

I found there is a similar issue with what you are having now: https://github.com/aws/amazon-freertos/issues/2286

I looks  the path cannot contain spaces, and your path "E:\FreeRTOS Device Software Configurations\coreHTTP_demo-1608663093143\FreeRTOS" does contain the space. Please try to remove the space in your and see if it works.

Thank you!

Ming

dipen-1337lab, 2021-01-05:
Namaste @mingyue86010, 

Thank you for your assistance on this issue. Looks like the issue is indeed related to `spaces in path`.

Below is the console log confirming the successful generation of the binary image, when the path does not contain any spaces.

```
[55/55] Generating bootloader.bin
esptool.py v2.8
[820/829] Building C object CMakeFiles/aws_demos.dir/vendors/espressif/boards/esp32/aws_demos/application_code/main.c.obj
../vendors/espressif/boards/esp32/aws_demos/application_code/main.c: In function 'prvMiscInitialization':
../vendors/espressif/boards/esp32/aws_demos/application_code/main.c:174:13: warning: unused variable 'uartRet' [-Wunused-variable]
     int32_t uartRet;
             ^
[829/829] Generating aws_demos.bin
esptool.py v2.8

E:\FreeRTOS_Device_Software_Configurations\coreHTTP_demo-1608663093143\FreeRTOS\build>
```

Thanks | Regards,
Dipen

closed 2021-01-05 by dipen-1337lab