aws/amazon-freertos #2911: AFR & ESP32: How to implement coreHTTP (Mutual Authentication) functionality on AFR Version 202002.00?

dipen-1337lab, 2020-12-23:
Namaste Executive,

For my application, how do I implement the Mutual Authentication requirement on AFR Ver 202002.00? From my analysis, I understand that the HTTP_MUTUAL_AUTH_DEMO is implemented in AFR Ver 202012.00 and uses the coreHTTP library. This demo isn't available in prior releases.

The requirement is as below, which is same as that of the [coreHTTP functionality](https://docs.aws.amazon.com/freertos/latest/userguide/core-http-ma-demo.html#core-http-ma-demo-functionality).

1. - Connect to the HTTP server on the AWS IoT endpoint
2. - Send a POST request
3. - Receive the response
4. - Disconnect from the server

Thanks | Regards,
Dipen

dipen-1337lab, 2020-12-24:
Namaste Community Member,

So, after quite a bit of tedious work, I have finally managed to run the [RunCoreHttpMutualAuthDemo](https://github.com/aws/amazon-freertos/blob/master/demos/coreHTTP/http_demo_mutual_auth.c#L237) on AFR Version 202002.00.

Before going into the details, I'm not sure if this would be the right approach as it involves modifying a large number of files, both cmake and, C (source and header) related. It was an attempt to reverse engineer and, resolve all the errors as and when encountered, hoping that the integration would work! Fortunately, I've succeeded.

Hope the below information, helps!

### git status (Details of files modified)

```
$ git status
On branch evcsp2-364
Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   firmware/CMakeLists.txt
        modified:   firmware/dual_outlet/iot_dual_outlet.c
        modified:   firmware/dual_outlet/iot_dual_outlet.h
        modified:   firmware/include/aws_clientcredential.h
        modified:   firmware/include/iot_demo_runner.h
        modified:   libraries/3rdparty/CMakeLists.txt
        deleted:    libraries/3rdparty/tinycbor/LICENSE
        deleted:    libraries/3rdparty/tinycbor/README.md
        deleted:    libraries/3rdparty/tinycbor/cbor.h
        deleted:    libraries/3rdparty/tinycbor/cborencoder.c
        deleted:    libraries/3rdparty/tinycbor/cborencoder_close_container_checked.c
        deleted:    libraries/3rdparty/tinycbor/cborerrorstrings.c
        deleted:    libraries/3rdparty/tinycbor/cborinternal_p.h
        deleted:    libraries/3rdparty/tinycbor/cborparser.c
        deleted:    libraries/3rdparty/tinycbor/cborparser_dup_string.c
        deleted:    libraries/3rdparty/tinycbor/cborpretty.c
        deleted:    libraries/3rdparty/tinycbor/cborpretty_stdio.c
        deleted:    libraries/3rdparty/tinycbor/cborvalidation.c
        deleted:    libraries/3rdparty/tinycbor/compilersupport_p.h
        deleted:    libraries/3rdparty/tinycbor/tinycbor-version.h
        deleted:    libraries/3rdparty/tinycbor/utf8_p.h
        modified:   libraries/CMakeLists.txt
        modified:   libraries/abstractions/pkcs11/CMakeLists.txt
        modified:   libraries/abstractions/pkcs11/ReadMe.md
        modified:   libraries/abstractions/pkcs11/test/iot_test_pkcs11.c
        modified:   tools/aws_config_quick_start/configure.json
        modified:   tools/cmake/afr_module.cmake
        modified:   vendors/espressif/boards/esp32/CMakeLists.txt
        modified:   vendors/espressif/boards/esp32/components/freertos/component.mk
        modified:   vendors/espressif/boards/esp32/proj04/config_files/aws_demo_config.h

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        evcsp2-365 git diff log for all the code changes.diff
        firmware/common/
        firmware/coreHTTP/
        libraries/3rdparty/mbedtls_config/
        libraries/3rdparty/mbedtls_utils/
        libraries/3rdparty/tinycbor/src/
        libraries/abstractions/backoff_algorithm.cmake
        libraries/abstractions/backoff_algorithm/
        libraries/abstractions/pkcs11/corePKCS11/
        libraries/abstractions/pkcs11/ecc608a/
        libraries/abstractions/pkcs11/psa/
        libraries/abstractions/pkcs11/test/MBT_C_CloseSession.c
        libraries/abstractions/pkcs11/test/MBT_C_DigestFinal.c
        libraries/abstractions/pkcs11/test/MBT_C_DigestInit.c
        libraries/abstractions/pkcs11/test/MBT_C_DigestUpdate.c
        libraries/abstractions/pkcs11/test/MBT_C_Finalize.c
        libraries/abstractions/pkcs11/test/MBT_C_FindObjects.c
        libraries/abstractions/pkcs11/test/MBT_C_FindObjectsFinal.c
        libraries/abstractions/pkcs11/test/MBT_C_FindObjectsInit.c
        libraries/abstractions/pkcs11/test/MBT_C_GenerateKeyPair.c
        libraries/abstractions/pkcs11/test/MBT_C_GenerateRandom.c
[... 68 lines cut ...]
I (2053) wifi: state: assoc -> run (10)
I (2163) wifi: connected with tkkfnBYMlkv6TeKM2, channel 1, BW20, bssid = 14:ae:85:e6:45:86
I (2163) wifi: pm start, type: 1

I (2163) WIFI: SYSTEM_EVENT_STA_CONNECTED
I (2913) event: sta ip: 192.168.29.49, mask: 255.255.255.0, gw: 192.168.29.1
I (2913) WIFI: SYSTEM_EVENT_STA_GOT_IP
4 258 [iot_thread] [INFO ][DEMO][2580] Successfully initialized the demo. Network type for the demo: 1
5 258 [iot_thread] [INFO] [HTTPDemo] [http_demo_mutual_auth.c:365] 6 258 [iot_thread] Establishing a TLS session to a4i9g6t5gqi4-ats.iot.us-east-1.amazonaws.com:8443.7 258 [iot_thread]
E (3443) PKCS11: failed nvs get file size 4354 0
8 686 [iot_thread] [INFO] [HTTPDemo] [http_demo_mutual_auth.c:439] 9 686 [iot_thread] Sending HTTP POST request to a4i9g6t5gqi4-ats.iot.us-east-1.amazonaws.com/topics/topic?qos=1...10 686 [iot_thread]
11 749 [iot_thread] [INFO] [HTTPDemo] [http_demo_mutual_auth.c:464] 12 749 [iot_thread] Received HTTP response from a4i9g6t5gqi4-ats.iot.us-east-1.amazonaws.com/topics/topic?qos=1...
13 749 [iot_thread]
14 1051 [iot_thread] [INFO] [HTTPDemo] [http_demo_mutual_auth.c:311] 15 1051 [iot_thread] Demo iteration 1 was successful.16 1051 [iot_thread]
17 1051 [iot_thread] [INFO] [HTTPDemo] [http_demo_mutual_auth.c:329] 18 1051 [iot_thread] Demo completed successfully.19 1051 [iot_thread]
```

### AWS IoT Console Log ( Subscribed to All topics "#" )

The image shows that the "Hello, world" JSON payload was successfully received on the AWS IoT Console.

[image]

Thanks | Regards,
Dipen

dipen-1337lab, 2020-12-24:
Namaste Community Member,

If you are aware of a better way to integrate the coreHTTP library or, if there is another HTTP library, which can be easily integrated with the AFR Version 202002.00, please do share the details. I can surely explore it and, try out the integration.

PS. Keeping this issue Open, anticipating some useful reply(s). Will Close it soon!

Thanks | Regards,
Dipen

mingyue86010, 2020-12-30:
Hi Dipen,

From your console output it looks you get it work!

We are trying to replace the old C-SDK lib (https://github.com/aws/amazon-freertos/tree/master/libraries/c_sdk) with newer one (https://github.com/aws/aws-iot-device-sdk-embedded-C). The https lib is replaced with a mutual-auth TLS + CoreHTTP. This will give user more fixibility and control on the transport layer connection.

The new http mutual auth demo would be a good source of reference of migrate from https lib to CoreHTTP https://github.com/aws/amazon-freertos/blob/master/demos/coreHTTP/http_demo_mutual_auth.c. The RunCoreHttpMutualAuthDemo() is basically send/recv HTTP messaages over a mutual authenticated TLS connection established by the secure socket. Then it passes the network context and reigisters the corresponding callback functions to the CoreHTTP libs by defining a the transport interface.

Regards,
Ming

dipen-1337lab, 2021-01-05:
Namaste @mingyue86010, 

Thank you for all the above useful information!

Additionally from my application perspective, the HTTP send and recv APIs of the coreHTTP library are working as desired, to help me decode the received response.

Thanks | Regards,
Dipen

sukhmanm, 2021-01-06:
Hi @dipen-1337lab,

I'm glad you were able to incorporate coreHTTP successfully! Did you have any further questions related to this topic?

dipen-1337lab, 2021-01-06:
Namaste @sukhmanm, 

> If you are aware of a better way to integrate the coreHTTP library or, if there is another HTTP library, which can be easily integrated with the AFR Version 202002.00, please do share the details. I [...]

Above is the reason for keeping this issue open. Is there a 3rd party library available, which can be used in instead of the cumbersome changes, which I have carried out to integrate coreHTTP library? My primary requirement is of send() and recv() APIs for Mutual Authentication over HTTP.

Either ways, we can surely close this issue 👍 

Thanks | Regards,
Dipen

closed 2021-01-06 by dipen-1337lab

sukhmanm, 2021-01-06:
Hi @dipen-1337lab,

Have you taken a look at the [HTTPS compatibility layer](https://github.com/aws/amazon-freertos/tree/master/libraries/c_sdk/standard/https)? It provides backward compatibility from coreHTTP to the older HTTPS APIs. You can take a look at some of the [https demos in 202002.00 ](https://github.com/aws/amazon-freertos/blob/202002.00/demos/https/iot_demo_https_s3_download_sync.c) to see how those APIs are used. The v1 library is located here: https://github.com/aws/amazon-freertos/tree/202002.00/libraries/c_sdk/standard/https. Let me know if you have any particular questions about this, and please feel free to open another issue if you need further assistance. Thanks!

dipen-1337lab, 2021-01-07:
Namaste @sukhmanm,

Thank you for the above references. As suggested, I shall surely open another issue for any further assistance.

Thanks | Regards,
Dipen