aws/amazon-freertos #2913: CMakeLists.txt not found and hence unsble to flash the code

NikhilPatale, 2020-12-25:
1- The following is the error when I try to flash the board with MinGW64.

C:\FreeRTOS\vendors\espressif\esp-idf\tools>idf.py flash -B C:\FreeRTOS\Builds
Note: You are using Python 3.8.2. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Setting IDF_PATH environment variable: C:\FreeRTOS\vendors\espressif\esp-idf
Checking Python dependencies...
Python requirements from C:\FreeRTOS\vendors\espressif\esp-idf\requirements.txt are satisfied.
CMakeLists.txt not found in project directory C:\FreeRTOS\vendors\espressif\esp-idf\tools

2- I am getting a CMakeLists.txt error because apparently there is no CMakeLists.txt in the folder C:\FreeRTOS\vendors\espressif\esp-idf\tools
I have downloaded the ESP32 DevKit C from the official AWS SDK resources. The CMakeLists.txt is available in FreeRTOS folder but it is not present in the ...\esp-idf\tools folder. Due to this reason, I am not able to flash the code.

3- However the ESP flashing part: ./vendors/espressif/esp-idf/tools/idf.py erase_flash -B build-directory works and the ESP is able to reset successfully.

4- Also the monitoring part: ./vendors/espressif/esp-idf/tools/idf.py monitor -p /dev/ttyUSB1 -B build-directory does work partially too, It does not ask for a CMakeLists.

A Solution to this issue would be highly appreciated.

mingyue86010, 2020-12-28:
Hi @NikhilPatale,

From where did you ran the idf.py flash command? The IDF tool has to be run from the your project root. (As said here https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html)

There was a simialr issue asked under esp-idf repo: https://github.com/espressif/esp-idf/issues/6007

Can you verify and see if that works for you? If you still having the issue, please let me know.

Thank you!
Ming

NikhilPatale, 2020-12-28:
Hi @mingyue86010 
The path which I have used is C:\FreeRTOS\vendors\espressif\esp-idf\tools
In this case, I think this itself is the root folder right? Since idf.py is present in the tools directory. Your help on this issue will be highly appreciated since I am stuck on this error for weeks and even though the error seems dumb, I am unable to find a solution for this

mingyue86010, 2020-12-29:
Hi @NikhilPatale,

The root shoud be the FreeRTOS proejct. I checked on my local repo it works. It I run the idf.py I can see the same error you saw.

So please go back to your FreeRTOS proejct folder by 
    cd C:\FreeRTOS 
Then run:
   ./vendors/espressif/esp-idf/tools/idf.py flash -B Builds

For more instruction, you can find on https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

Thanks,
Ming

NikhilPatale, 2020-12-29:
Hi @mingyue86010 

Thank you for your help. However, I seem to be stuck on this error when I try to run it. Could you kindly let me know why am I not being able to execute the code? pasting the same from my Run terminal. 

C:\>cd freertos

C:\FreeRTOS>./vendors/espressif/esp-idf/tools/idf.py flash -B C:/Builds
'.' is not recognized as an internal or external command,
operable program or batch file.

Thanks, 
Nikhil

NikhilPatale, 2020-12-29:
Hi @mingyue86010,

Thanks for the solution! I am able to flash it successfully. Thank you for your help!!

closed 2020-12-29 by NikhilPatale