aws/amazon-freertos #2915: No support for PIC32MZ1025W104 in

CaptainMJx, 2020-12-28:
Not possible to compile in MPLAB X if processor is chosen as "PIC32MZ1025W104" which is the processor used in the development board "PIC32 WFI32E Curiosity Board" that is marked as a supported board on Amazon website.

Suggested solution:
Add support for the board so it is possible to compile AWS Freertos.

Additional information:
It seem to be missing the processor ports file, referenced in ports_p32xxxx.h.
../../../../../vendors/microchip/harmony/v2.05/framework/peripheral/ports/processor/ports_p32xxxx.h:779:6: error: #error Unknown processor! #error Unknown processor!

Thank you!

sukhmanm, 2020-12-31:
Hi @CaptainMJx,

Thank you for your request. As you had pointed out, the PIC32 WFI32E Curiosity Board is in fact listed in the AWS Partner Device Catalog, but isn't yet officially supported by Amazon FreeRTOS. However, Microchip has been developing a port for the board (based on release 202002.00) which may be useful, and can be found here: https://github.com/MicrochipTech/amazon-freertos/blob/mchpdev/vendors/microchip/boards/GettingStarted_With%20PIC32MZW1.md.

This is also the official "Getting Started" point of reference used in the board's AWS Device Catalog page. In case you haven't already, you can find additional information about the AWS specifications of the board on that page: https://devices.amazonaws.com/detail/a3G0h000007dl7rEAA/PIC32MZW1.

Please let us know if you have any further questions. Thank you.

CaptainMJx, 2021-01-01:
Hi @sukhmanm,

Thanks, that explains it. I will start with the demo software as you suggested, but do you know when the support will be added? Is it something that is expected this month or at least before end of Q1?

Many thanks,
Marcus

sukhmanm, 2021-01-04:
Hi @CaptainMJx ,

Unfortunately, we are unable to provide a timeline for official support of the board. However, the request is in our internal task queue, so please look out for future updates regarding this matter.

Thank you again for your inquiry, and please feel free to create a new issue or re-open this one for any additional questions you may have.

closed 2021-01-04 by sukhmanm