aws/amazon-freertos #2916: NETWORK ERROR encountered, for aperiodic MQTT Publish operation

dipen-1337lab, 2020-12-29:
Namaste Community Member,

Please find below the details of the issue encountered.

### Application Overview
The FreeRTOS version in use for development is V202002.00.

For my application I am leveraging the OTA (over MQTT) and MQTT demo codes. The OTA thread runs first, initiates the required MQTT connection with AWS IoT, which is then shared by the MQTT thread.

I can continuously see the OTA log on the USB Serial console, which is as below.
`136 5635 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0`

However, the MQTT Publish logs are event based and hence, aperiodic. A sample MQTT based Publish log seen on the USB Serial console is as below.
```
95 4968 [iot_thread] [INFO ][DEMO][49680] Publishing message
96 4968 [iot_thread] [INFO ][DEMO][49680] 95 bytes written.
97 4968 [iot_thread] [INFO ][MQTT][49680] (MQTT connection 0x3fff3a24) MQTT PUBLISH operation queued.
98 4968 [iot_thread] [INFO ][DEMO][49680] Published Outlet 1 message.
99 5002 [iot_thread] [INFO ][DEMO][50020] MQTT PUBLISH 0 successfully sent.
```

### Issue Description

It is observed that due to the aperiodic behaviour of the MQTT Publish operation in my application, the MQTT connection gets closed. With multiple rounds of testing, it is observed that the MQTT connection closes after around 5 minutes of inactivity.

The success and failure logs for the MQTT Publish operation are as below.

```
/***** ***** ***** ***** *****
First log Publish successful 
***** ***** ***** ***** *****/

I (48886) dual_outlet: INFO : pvOutletTwoRfidReader() RFID Card Number: b9e63029
92 4967 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
93 4968 [iot_thread] [INFO ][DEMO][49680] deviceState[0] = 1
94 4968 [iot_thread] [INFO ][DEMO][49680] deviceState[1] = 2
95 4968 [iot_thread] [INFO ][DEMO][49680] Publishing message
96 4968 [iot_thread] [INFO ][DEMO][49680] 95 bytes written.
97 4968 [iot_thread] [INFO ][MQTT][49680] (MQTT connection 0x3fff3a24) MQTT PUBLISH operation queued.
98 4968 [iot_thread] [INFO ][DEMO][49680] Published Outlet 1 message.
99 5002 [iot_thread] [INFO ][DEMO][50020] MQTT PUBLISH 0 successfully sent.

/***** ***** ***** ***** *****
Second log Publish unsuccessful 
***** ***** ***** ***** *****/

I (360886) dual_outlet: INFO : pvOutletTwoRfidReader() RFID Card Number: b9e63029
412 36167 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
413 36168 [iot_thread] [INFO ][DEMO][361680] deviceState[0] = 1
414 36168 [iot_thread] [INFO ][DEMO][361680] deviceState[1] = 2
415 36168 [iot_thread] [INFO ][DEMO][361680] Publishing message
416 36168 [iot_thread] [INFO ][DEMO][361680] 95 bytes written.
417 36168 [iot_thread] [INFO ][MQTT][361680] (MQTT connection 0x3fff3a24) MQTT PUBLISH operation queued.
418 36168 [iot_thread] [INFO ][DEMO][361680] Published Outlet 1 message.
419 36169 [iot_thread] [ERROR][NET][361690] Error -27648 while sending data.
420 36171 [iot_thread] [ERROR][DEMO][361710] MQTT PUBLISH 0 could not be sent. Error NETWORK ERROR.
```

### Additional Information

1. From the **Free Heap Size** logs, as shared below, it can be understood that the availability of the Heap memory increases, which might point to the fact that the MQTT related thread closes.

```
***** Publish successful *****
I (307765) dual_outlet: INFO : Free Heap size = 15388.
I (307765) dual_outlet: INFO : Minimum Ever Free Heap size = 1724.

***** Publish unsuccessful *****
I (322765) dual_outlet: INFO : Free Heap size = 49676.
I (322765) dual_outlet: INFO : Minimum Ever Free Heap size = 1724.
```

2. The OTA related console log continues to get displayed on the USB Serial console as below.
`413 33432 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0`

3. It looks like a timeout related issue and, a keep alive mechanism might resolve this issue.

How do I ensure that the MQTT connection remains alive as long as the firmware is running?

Thanks | Regards,
Dipen

sukhmanm, 2021-01-01:
Hi @dipen-1337lab,

Thank you for your question, and for attaching some logs. Would you be able to provide the full demo log as well? It appears the issue may be a result of how the connection is being managed between the demos, and as there seem to be quite a few missing logs printed between the two publishes, a better understanding of the entire output would be great.

It would also be helpful to know how the two demos are initialized in your implementation, and how the connection is being shared between them. In the OTA demo, the MQTT connection is created first, and then passed to the OTA Agent task by reference, [in line 251](https://github.com/aws/amazon-freertos/blob/987d9697ad4dc924bd71333c57b2f68bbe81b172/demos/ota/aws_iot_ota_update_demo.c#L251). You can then also use this connection for other MQTT operations (like publish/subscribe), or in additional tasks. This example directly modifies the OTA demo to also publish to an example topic, which may be relevant to your use case: https://github.com/aws/amazon-freertos/blob/mchp/otaapp/demos/ota/aws_iot_ota_update_demo.c. 

If you wish to run the two demos independently, and aren't doing so already, you should modify [`iot_demo_runner.h`](https://github.com/aws/amazon-freertos/blob/202002.00/demos/include/iot_demo_runner.h) to point to a single demo entry function that runs the two demos as separate tasks. Depending on your current implementation, it may be important to note that only one demo should be defined in  [`aws_demo_config.h`](https://github.com/aws/amazon-freertos/blob/987d9697ad4dc924bd71333c57b2f68bbe81b172/vendors/espressif/boards/esp32/aws_demos/config_files/aws_demo_config.h#L29-L46). 

Regarding the timeout inquiry, the MQTT library handles this by sending PINGREQ packets to the broker if the user-defined timeout has been exceeded. The keep-alive job is scheduled when `IotMqtt_Connect` is called. 

Please let me know if I can clarify any of the above. Thanks.

sukhmanm, 2021-01-06:
Hi @dipen-1337lab,

I've taken a look at the files you had posted in the forum thread. It seems the issue still lies in how the connection is being shared and accessed between the threads. A few comments:

-  As suggested in the forum thread, it would be better to use a synchronization primitive (maybe a task notification sent from the OTA task to indicate that a successful connection has been established), instead of delaying the MQTT demo by 15 seconds. It would ensure that the MQTT demo isn't accessing the shared connection prematurely. Alternatively, you could also establish the connection in the main demo task, before starting the MQTT and OTA demo threads.

- I would also suggest taking a look at the [latest OTA demo](https://github.com/aws/amazon-freertos/blob/master/demos/ota/aws_iot_ota_update_demo.c), as it contains a few updates that might be helpful in debugging. Most notably:

  - `prvNetworkDisconnectCallback` is updated with logs that indicate the reason for disconnection.
  - `_establishMqttConnection` is added to provide more info to `IotMqtt_Connect`, such as network credentials and network interface struct members previously not set by the initializer.

If these changes are feasible and you are able to try them, and provide an updated demo log, it would help greatly in isolating the issue. Let me know if there are any clarifications I can provide. Thanks.

dipen-1337lab, 2021-01-08:
Namaste @sukhmanm, 

> I've taken a look at the files you had posted in the forum thread.

Thank you for obliging and, sharing your review comments. I shall revert if I am able to integrate the suggested changes and, share the corresponding logs.

Further on this, I carried out long duration testing 3 times for about 5 hours. In the 2nd attempt, I didn't encounter any failures and, the MQTT Publish operations continued as desired periodically, at an interval of 15 seconds. In the 1st and 3rd attempts, I encountered the below error, which finally resulted in the MQTT Publish failure error.
`5440 363735 [iot_thread] [ERROR][MQTT][3637350] (MQTT connection 0x3ffe9688) Failed to receive PINGRESP within 10000 ms.`

The partial console log is shared below.

```
5415 361937 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
5416 362037 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
5417 362137 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
5418 362237 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0

I (3622685) dual_outlet: Current UNIX EPOCH Time is: 1610077001
I (3622685) dual_outlet: Equivalent GMT Date Time is: Fri Jan  8 03:36:41 2021
I (3622785) dual_outlet: Outlet Id = 1
E (3622785) dual_outlet: Failed to obtain new readings from the Energy Meter.
I (3622785) dual_outlet: INFO : Free Heap size = 14328.
I (3622785) dual_outlet: INFO : Minimum Ever Free Heap size = 1924.

5419 362337 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0

5420 362368 [iot_thread] [INFO ][DEMO][3623680] deviceState[0] = 1
5421 362368 [iot_thread] [INFO ][DEMO][3623680] deviceState[1] = 5
5422 362368 [iot_thread] [INFO ][DEMO][3623680] INFO : _publishEnergyMeterReadings: Publishing message ...
5423 362368 [iot_thread] [INFO ][DEMO][3623680] INFO : _publishEnergyMeterReadings: 86 bytes written.
5424 362368 [iot_thread] [INFO ][MQTT][3623680] (MQTT connection 0x3ffe9688) MQTT PUBLISH operation queued.
5425 362368 [iot_thread] [INFO ][DEMO][3623680] INFO : _publishEnergyMeterReadings(): successfully published Energy Meter readings for Outlet 1.
5426 362410 [iot_thread] [INFO ][DEMO][3624100] MQTT PUBLISH 3 successfully sent.

5427 362437 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
5428 362537 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
5429 362637 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
5430 362737 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
5431 362837 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
5432 362937 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
5433 363037 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
5434 363137 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
5435 363237 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
I (3632685) dual_outlet: Current UNIX EPOCH Time is: 1610077011
I (3632685) dual_outlet: Equivalent GMT Date Time is: Fri Jan  8 03:36:51 2021
5436 363337 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
5437 363437 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
5438 363537 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
5439 363637 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0

5440 363735 [iot_thread] [ERROR][MQTT][3637350] (MQTT connection 0x3ffe9688) Failed to receive PINGRESP within 10000 ms.
5441 363735 [iot_thread] [INFO ][MQTT][3637350] (MQTT connection 0x3ffe9688) Network connection closed.

5442 363737 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0

I (3637785) dual_outlet: Outlet Id = 1
E (3637785) dual_outlet: Failed to obtain new readings from the Energy Meter.
I (3637785) dual_outlet: INFO : Free Heap size = 14356.
I (3637785) dual_outlet: INFO : Minimum Ever Free Heap size = 1924.

5443 363837 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0

5444 363868 [iot_thread] [INFO ][DEMO][3638680] deviceState[0] = 1
5445 363868 [iot_thread] [INFO ][DEMO][3638680] deviceState[1] = 5
5446 363868 [iot_thread] [INFO ][DEMO][3638680] INFO : _publishEnergyMeterReadings: Publishing message ...
5447 363868 [iot_thread] [INFO ][DEMO][3638680] INFO : _publishEnergyMeterReadings: 86 bytes written.
5448 363868 [iot_thread] [WARN ][MQTT][3638680] (MQTT connection 0x3ffe9688) Attempt to use closed connection.
5449 363868 [iot_thread] [ERROR][MQTT][3638680] (MQTT connection 0x3ffe9688) New operation record cannot be created for a closed connection
5450 363868 [iot_thread] [ERROR][DEMO][3638680] _publishEnergyMeterReadings: MQTT PUBLISH 3 returned error NETWORK ERROR.
5451 363868 [iot_thread] [ERROR][DEMO][3638680] ERROR: _publishEnergyMeterReadings(): failed to publish Energy Meter readings for Outlet 1.
5452 363868 [iot_thread] [INFO ][DEMO][3638680] INFO : Retrying ... 

5453 363937 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
[... cut ...]
The console log file `consoleLog Fri Jan 8 023622 2021.log` corresponds to the failure scenario where the MQTT Publish fails as explained above.
[consoleLog Fri Jan 8 023622 2021.log](https://github.com/aws/amazon-freertos/files/5787309/consoleLog.Fri.Jan.8.023622.2021.log)

Some additional information,

1. This is a periodic MQTT Publish operation, which is also failing.
2. The MQTT Publish begins after User authentication over HTTP, which can be seen in the log files.

Requesting your inputs for the possible resolution.

Thanks | Regards,
Dipen

dipen-1337lab, 2021-01-09:
Namaste @sukhmanm and Community Members,

Another long duration testing attempt and again the MQTT Publish fails after about 2 hours of operation. The console log file is shared here for reference.
[consoleLog Fri Jan 8 210214 2021.txt](https://github.com/aws/amazon-freertos/files/5790141/consoleLog.Fri.Jan.8.210214.2021.txt)

The errors observed on the console are as below.

```
8723 701364 [iot_thread] [INFO ][DEMO][7013640] INFO : _publishEnergyMeterReadings: Publishing message ...
8724 701364 [iot_thread] [INFO ][DEMO][7013640] INFO : _publishEnergyMeterReadings: 86 bytes written.
8725 701364 [iot_thread] [INFO ][MQTT][7013640] (MQTT connection 0x3ffe8dac) MQTT PUBLISH operation queued.
8726 701364 [iot_thread] [INFO ][DEMO][7013640] INFO : _publishEnergyMeterReadings(): successfully published Energy Meter readings for Outlet 1.
8727 701365 [iot_thread] [ERROR][NET][7013650] Error -27648 while sending data.
8728 701367 [iot_thread] [ERROR][DEMO][7013670] MQTT PUBLISH 3 could not be sent. Error NETWORK ERROR.

...
...

9037 725751 [iot_thread] [ERROR][NET][7257510] Error -27648 while sending data.
9038 725751 [iot_thread] [ERROR][MQTT][7257510] (MQTT connection 0x3ffe8dac) Failed to send PINGREQ.
9039 725751 [iot_thread] [INFO ][MQTT][7257510] (MQTT connection 0x3ffe8dac) Network connection closed.

...
...

9068 728364 [iot_thread] [INFO ][DEMO][7283640] INFO : _publishEnergyMeterReadings: Publishing message ...
9069 728364 [iot_thread] [INFO ][DEMO][7283640] INFO : _publishEnergyMeterReadings: 86 bytes written.
9070 728364 [iot_thread] [WARN ][MQTT][7283640] (MQTT connection 0x3ffe8dac) Attempt to use closed connection.
9071 728364 [iot_thread] [ERROR][MQTT][7283640] (MQTT connection 0x3ffe8dac) New operation record cannot be created for a closed connection
9072 728364 [iot_thread] [ERROR][DEMO][7283640] _publishEnergyMeterReadings: MQTT PUBLISH 3 returned error NETWORK ERROR.
9073 728364 [iot_thread] [ERROR][DEMO][7283640] ERROR: _publishEnergyMeterReadings(): failed to publish Energy Meter readings for Outlet 1.
9074 728364 [iot_thread] [INFO ][DEMO][7283640] INFO : Retrying ... 
```

Thanks | Regards,
Dipen

sukhmanm, 2021-01-13:
Hi @dipen-1337lab,

Thank you for attaching the complete logs, and apologies for the delay. After taking another look at the implementation, it could very well be a timeout issue that is causing the network to disconnect. Below are a couple suggestions:

1. In order to maintain the MQTT connection for a long period of time, you may need to verify that you are maintaining the TCP/TLS connection as well. If this connection is dropped, any further MQTT operations would fail as a result. I would suggest again to make use of the [network callback](https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/mqtt/structIotMqttNetworkInfo__t.html#a7d3b3e7895b3240f9e4904191cc98402) function, which will be able to notify you of the [disconnect reason](https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/mqtt/group__mqtt__datatypes__enums.html#ga7b42cf7dbbe16a89a6a0886285790216) ([this is an example](https://github.com/aws/amazon-freertos/blob/9205e3c9468ba01f847d8dc82ac11c96454ef878/demos/ota/aws_iot_ota_update_demo.c#L202) from the OTA demo). When the network is disconnected and this callback is received, you can then re-establish the MQTT connection if you wish to continue running the demos.
2. I see that you are running a couple of threads, in addition to the two demo threads, all having the same priority. If you are using one core on ESP32, only one will be running at a time, and the delays and scheduling of each may be leading to a timeout. Additionally, using `vTaskDelay()` isn't [ideal for controlling periodic tasks](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#_CPPv410vTaskDelayK10TickType_t) due to its unpredictability (since the blocking time is specified relative to the time at which `vTaskDelay()` is called, the activity of other tasks might delay the next execution, hence why in the logs the publishes are often over 30 seconds apart). `vTaskDelayUntil()` uses absolute time instead, and might be better suited for periodic tasks. 
3. I would also suggest enabling [MQTT debug logging](https://github.com/aws/amazon-freertos/blob/987d9697ad4dc924bd71333c57b2f68bbe81b172/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h#L51) to receive more information about the state of the connection. If none of the above help to identify the problem, could you provide an updated log with these debug logs enabled? You can also specify the wait time for `PINGRESP` packets in that file.

Again, please let me know if there are additional questions or clarifications. Thanks.

dipen-1337lab, 2021-01-14:
Namaste @sukhmanm,

> apologies for the delay.

Please don't be. I'm grateful for your continued assistance. 

> If these changes are feasible and you are able to try them, and provide an updated demo log, it would help greatly in isolating the issue.

As suggested, in the FreeRTOS project folder, I replaced the existing `aws_iot_ota_update_demo.c` source file (version 202002.00) with the latest file from the [master branch](https://github.com/aws/amazon-freertos/blob/master/demos/ota/aws_iot_ota_update_demo.c). I then resolved all the compilation errors encountered.

Here's one of the log captured, which identifies the disconnection reason as `Mqtt disconnected due to Keep-alive response not received.`
[console_log_Mqtt disconnected due to Keep-alive response not received.txt](https://github.com/aws/amazon-freertos/files/5815993/console_log_Mqtt.disconnected.due.to.Keep-alive.response.not.received.txt)

> 1\. I would suggest again to make use of the [network callback](https://docs.aws.amazon.com/freertos/latest/lib-ref/c-sdk/mqtt/structIotMqttNetworkInfo__t.html#a7d3b3e7895b3240f9e4904191cc98402) function

Yes, this has now been incorporated by using the latest `aws_iot_ota_update_demo.c` source file.

> When the network is disconnected and this callback is received, you can then re-establish the MQTT connection if you wish to continue running the demos.

I can surely attempt this but would like to take this approach as a fall-back mechanism/ workaround. The Root Cause Analysis is my current attempt to understand what is it that is causing such behaviour.

> 2\. `vTaskDelayUntil()` uses absolute time instead, and might be better suited for periodic tasks.

I shall incorporate this in my firmware. Thank you!

> 3\. I would also suggest enabling [MQTT debug logging](https://github.com/aws/amazon-freertos/blob/987d9697ad4dc924bd71333c57b2f68bbe81b172/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h#L51) to receive more information about the state of the connection.

Again, this has been incorporated and, the Debug logs can be seen in the log file shared above.

> 3\. You can also specify the wait time for `PINGRESP` packets in that file.

I am using the default value of 10000 milliseconds for now.

Do let me know for any additional information from my end.

Thanks | Regards,
Dipen

dipen-1337lab, 2021-01-15:
Namaste @sukhmanm,

An update on the aperiodic MQTT Publish operation, using the latest `aws_iot_ota_update_demo.c` source file is as below.

> It is observed that due to the aperiodic behaviour of the MQTT Publish operation in my application, the MQTT connection gets closed. With multiple rounds of testing, it is observed that the MQTT connection closes after around 5 minutes of inactivity.

This observation is not reproduced anymore. My multiple test attempts do not result in any error and, the MQTT connection remains established as long as the device remains ON. I have tested for a maximum of 5 hours.

### A new finding

Further on the `Error -27648 while sending data.` error encountered, which then causes the PINGREQ also to fail, below is an interesting observation on this issue.

1\. Firmware is working properly. The PINGREQ are sent every 2 minutes and PINGRESP is received within 10 seconds.

2\. Firmware processes the Subscription Callback when the topic subscribed for, is published from the AWS IoT's [MQTT test client](https://console.aws.amazon.com/iot/home?region=us-east-1#/test).

3\. PINGREQ and PINGRESP continues as expected.

4\. The moment our application's Python code block publishes the same topic, the first MQTT Publish operation attempted by the firmware results in the `Error -27648 while sending data.` error. Also, with the connection breaking, the PINGREQ publish fails too.

### Findings

1. The JSON topic published by our application code is correctly displayed on the AWS IoT's MQTT test client as can be seen below.

[image]

2\. Interestingly, and I don't know why, the error is encountered even when the device has not subscribed to the topic.

### Python code block

Below is the code block used to Publish a message to the device.

```
CLIENT_ID = device_id
TOPIC = [device]/[outlet]/stop/action"
event_loop_group = io.EventLoopGroup(1)
host_resolver = io.DefaultHostResolver(event_loop_group)
client_bootstrap = io.ClientBootstrap(event_loop_group, host_resolver)
mqtt_connection = mqtt_connection_builder.mtls_from_path(
    endpoint=settings.AWS_MQTT_ENDPOINT,
    cert_filepath=settings.AWS_MQTT_PATH_TO_CERT,
    pri_key_filepath=settings.AWS_MQTT_PATH_TO_KEY,
    client_bootstrap=client_bootstrap,
    ca_filepath=settings.AWS_MQTT_PATH_TO_ROOT,
    client_id=device_id,
    clean_session=False,
    keep_alive_secs=6
)
print("Connecting to {} with client ID '{}'...".format(
settings.AWS_MQTT_ENDPOINT, CLIENT_ID))
# Make the connect() call
connect_future = mqtt_connection.connect()
# Future.result() waits until a result is available
connect_future.result()
print("Connected!")
# Publish message to server desired number of times.
print('Begin Publish')
message = {"action" : "0"}
mqtt_connection.publish(topic=TOPIC, payload=json.dumps(message), qos=mqtt.QoS.AT_LEAST_ONCE)
db_logger.info("Published: '" + json.dumps(message) + "' to the topic: " + TOPIC)
print('Publish End')
disconnect_future = mqtt_connection.disconnect()
disconnect_future.result()
```

### My analysis

1. A message received for the topic subscribed for, is causing the MQTT connection to break, which then results in failure of MQTT Publish operations initiated by the device.
2. Having said the above, the subscription mechanism works fine when a Publish is initiated from the MQTT test client on AWS IoT. However, the subscription mechanism generates some error when a Publish is initiated by our application code.
3. Exact line of failure on firmware side is yet to identified.
4. My multiple attempts to print on the console, the topic and payload received, were futile.

Hope this helps you in sharing some further steps to resolve this issue. Any change to be incorporated in the Python code?

Thanks | Regards,
Dipen

sukhmanm, 2021-01-15:
Hi @dipen-1337lab,

Thank you for the detailed updates - they're very helpful in understanding the situation. 

If I understand correctly, the connection error occurs on the device-side only when attempting to publish from the python application. I see in the python code that a new MQTT connection is created -- is there any chance that the client ID used for this connection is the same one used in your device application (i.e. is `device_id == clientcredentialIOT_THING_NAME`, which appears from the logs to be `esp32DevKitCBoard01`)? According to the [MQTT spec](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc385349242), multiple connections to the same endpoint must have unique client IDs, otherwise it's likely that the first connection will suddenly be dropped by the server, without notification, after the second connection has been made.

Could you confirm if this is the issue? If not, please let me know and I will take a deeper look. Thanks.

dipen-1337lab, 2021-01-15:
Namaste @sukhmanm,

> Thank you for the detailed updates - they're very helpful in understanding the situation.

Yes! I truly believe in rigorous amount of testing and detailing of the observations. This is a classic example of the scenario where the logs and observations have helped us pin point to the exact root cause. Thank you too, for acknowledging my detailed updates and, for your continued assistance on this issue.

 

> Could you confirm if this is/isn't the issue?

I confirm that this was indeed the issue i.e., the same `device_id` was being used at both the ends. Lesson learnt the hard way in the absence of not knowing the MQTT specifications.

Please find below the logs in three parts, which confirm that the MQTT connection remains established after receiving the payload for the Subscribed topic.

**First part - Subscribed Topic is received by device and viewed on console**
```
563 11497 [iot_thread] [INFO ][DEMO][114970] Incoming PUBLISH received:
Subscription topic filter: esp32DevKitCBoard01/1/stop/action
Publish QoS: 1
Publish payload: {"action": "0"}
```

**Second part - An acknowledge of the above receipt, is sent to the Server**
```
580 11504 [iot_thread] [INFO ][MQTT][115040] (MQTT connection 0x3ffef190) MQTT PUBLISH operation queued.
581 11504 [iot_thread] [INFO ][DEMO][115040] INFO : _publishTopicCharging(): queued the MQTT PUBLISH operation to publish Relay switching timestamp and charging state (0/1) for Outlet 1.
582 11505 [iot_thread] [DEBUG][MQTT][115050] Changing PUBLISH packet identifier 27 to 29.
583 11505 [iot_thread] [DEBUG][MQTT][115050] (MQTT connection 0x3ffef190, PUBLISH operation 0x3ffd57f0) Sending MQTT packet.
584 11505 [iot_thread] [DEBUG][MQTT][115050] (MQTT connection 0x3ffef190, PUBLISH operation 0x3ffd57f0) Scheduling retry 1 of 10 in 5000 ms.
585 11553 [NetRecv] [DEBUG][MQTT][115530] (MQTT connection 0x3ffef190) PUBACK in data stream.
586 11554 [NetRecv] Packet identifier 29.
587 11554 [NetRecv] [DEBUG][MQTT][115540] (MQTT connection 0x3ffef190) Searching for operation PUBLISH pending response with packet identifier 29.
588 11555 [NetRecv] [DEBUG][MQTT][115550] (MQTT connection 0x3ffef190) Found operation PUBLISH.
589 11556 [NetRecv] [DEBUG][MQTT][115560] (MQTT connection 0x3ffef190, PUBLISH operation 0x3ffd57f0) Callback scheduled.
590 11556 [iot_thread] [INFO ][DEMO][115560] MQTT PUBLISH 1 successfully sent.
```

**Third part - The Keep Alive PINGREQ and PINGRESP operations, which follow are also successful.**

```
609 12766 [iot_thread] [DEBUG][MQTT][127660] (MQTT connection 0x3ffef190) Keep-alive job started.
610 12766 [iot_thread] [DEBUG][MQTT][127660] (MQTT connection 0x3ffef190) Sending PINGREQ.
611 12767 [iot_thread] [DEBUG][MQTT][127670] (MQTT connection 0x3ffef190) PINGREQ sent. Scheduling check for PINGRESP in 10000 ms.
612 12767 [iot_thread] [DEBUG][MQTT][127670] (MQTT connection 0x3ffef190) Next keep-alive job in 10000 ms.
613 12767 [iot_thread] [INFO ][DEMO][127670] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0

614 12802 [NetRecv] [DEBUG][MQTT][128020] (MQTT connection 0x3ffef190) PINGRESP in data stream.
615 12802 [NetRecv] [DEBUG][MQTT][128020] (MQTT connection 0x3ffef190) PINGRESP successfully parsed.

626 13767 [iot_thread] [DEBUG][MQTT][137670] (MQTT connection 0x3ffef190) Keep-alive job started.
627 13767 [iot_thread] [DEBUG][MQTT][137670] (MQTT connection 0x3ffef190) Checking for PINGRESP.
628 13767 [iot_thread] [DEBUG][MQTT][137670] (MQTT connection 0x3ffef190) PINGRESP was received.
629 13767 [iot_thread] [DEBUG][MQTT][137670] (MQTT connection 0x3ffef190) Next keep-alive job in 120000 ms.
```

Once again, truly grateful for your assistance! Wishing you continued success. Stay blessed.

Thanks | Regards,
Dipen

closed 2021-01-15 by dipen-1337lab

sukhmanm, 2021-01-16:
Hi @dipen-1337lab,

I'm so glad that the issue was found and resolved! And yes, I definitely agree with you on that -- the more information available the better, so the thoroughness in your testing and observations is much appreciated!

Please don't hesitate to reach out again if you have any questions/concerns in the future. Thanks again!