aws/amazon-freertos #2917: Wifi Provisioning; are there Android and iPhone Apps available for this?

CaptainMJx, 2020-12-30:
Hi, in regards to WiFi provisioning. Are there any apps made for Android and iPhone that can do this and that are supported? Or is that something that I would need to build myself? The need would simply be to deliver the WiFi details to the AWS RTOS device in as simple as possible way. I'm not too concerned about branding just need a way to set the WiFi config.

Many thanks!
Marcus

mingyue86010, 2020-12-31:
Hi @CaptainMJx,

Yes, we do have mobile SDKs for FreeRTOS Bluetooth Devices at here:
iOS: https://github.com/aws/amazon-freertos-ble-ios-sdk 
Android: https://github.com/aws/amazon-freertos-ble-android-sdk

You need to clone the iOS/Android repo and build the App from PC by yourself. But for wifi provisioning demo, currently we only support the Espressif ESP32-DevKitC. You can find more detailed information about how to run the demo from here: https://docs.aws.amazon.com/freertos/latest/userguide/ble-demo.html#ble-demo-wifi

Hopefully, those information can help, and please let us know if you have any further question.

Thank you!

Regards,
Ming

CaptainMJx, 2020-12-31:
@mingyue86010 Thanks for that, i will review those apps. 
Happy New Year!

Regards,
Marcus

closed 2020-12-31 by CaptainMJx