aws/amazon-freertos #2918: how to read my analog temperature sensor data in AWS-freertos?

SGOSAI602, 2020-12-30:
Describe the bug
how to read my analog temperature sensor data in AWS-freertos?

System information
Hardware - ESP32-WROOM-32 D
Operating System Linux-Ubuntu
Version of FreeRTOS (run git describe --tags to find it)
Project AWS FreeRTOS shadow demo

Thank you!

sukhmanm, 2021-01-04:
HI @SGOSAI602,

Would you be able to describe your issue in more detail? As well as the FreeRTOS version you are using?

We currently have a set of libraries to abstract different I/O peripherals into a common API. This temperature sensor peripheral header (and tests) may be relevant to your use case: https://github.com/aws/amazon-freertos/blob/master/libraries/abstractions/common_io/include/iot_tsensor.h

The ESP-IDF also has example of using their built-in temperature sensor: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/temp_sensor_esp32s2.

Please let me know if this is helpful, and feel free to follow up with additional details if necessary. Thanks.

pvyawaha, 2021-01-19:
Closing this issue now, please reopen for the same topic or create a new one for other topocs.

closed 2021-01-19 by pvyawaha

mudassaramin, 2021-11-02:
Can we integrate Arduino library in AWS FreeRTOS. I am experimenting with FreeRTOS and confused to select either Arduino IDE or Mongoose OS or FreeRTOS but my main issues is I have the sensor library in Arduino IDE but not in Mongoose OS or FreeRTOS.

Any help in this regard will be highly appreciated...