aws/amazon-freertos #2923: osKernelSuspended

LVhiroki, 2021-01-06:
**Is your feature request related to a problem? Please describe.**  
A clear and concise description of what the problem is. What are the use cases of the feature to be added? 

When using STMicro's microcomputer, the created task may stop.
It is recommended to use osKernelSuspended, but osKernelSuspended is not implemented.

**Describe the solution you would like.**  
A clear and concise description of what you want to happen.

I want to implement osKernelSuspended. Or want to use an alternative

**Additional context**  
Add any other information (including rationale, requirements, or references) about the feature request here.

Thank you!

alfred2g, 2021-01-12:
Hi,
Where did you see that it is recommended to use osKernelSuspended? I don't see it implemented anywhere in amazon-freertos.
It is implemented in mbedos though.

pvyawaha, 2021-01-19:
Closing this issue now, please reopen for the same topic or create a new one for other topics.

closed 2021-01-19 by pvyawaha