aws/amazon-freertos #2927: AFR & ESP32: CMake Error (Cannot find source file) when using -DAFR_ESP_LWIP=1 option

dipen-1337lab, 2021-01-08:
**Describe the bug**
The AFR Version 202002.00 generates a CMake Error (Cannot find source file) when running the cmake command with the -`DAFR_ESP_LWIP=1` option

**System information**
- Hardware board: ESP32-DevKitC
- IDE used: N/A
- Operating System [Windows 10 x64 -- Version: 20H2 -- OS Build: 19042.685]
- Version of FreeRTOS: AFR Version 202002.00 downloaded from console.aws.amazon.com
- Project: OTA Demo

**Expected behavior**
The cmake command should run successfully.

**Screenshots or console output**
Refer the cmakeError.log file for the console output providing the error log,
[cmakeError.log](https://github.com/aws/amazon-freertos/files/5787416/cmakeError.log)

**To reproduce**
Steps to reproduce the behaviour:

1. Login and download FreeRTOS from the FreeRTOS console.
2. Create a configuration for a ESP32-DevKitC. On the Configure FreeRTOS Software configuration properties page, under Libraries, choose OTA Updates. Under Demo projects, choose to enable the OTA demo.
3. Download and unzip to extract the folder contents.
4. Open Command Prompt with Administrator privilege.
5. Change the current working directory path to ..\FreeRTOS\
6. Run the below command and observe the error generated.
`cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -DAFR_ESP_LWIP=1 -GNinja -S . -B build`

**Code to reproduce the bug**
N/A

**Additional context**
N/A

Thank you!
Dipen

alfred2g, 2021-01-12:
Hi,
I see this line in  cmakeerror.log file
```
Loading defaults file E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/build/sdkconfig.defaults...
-- IDF_VER: v3.3-163-g601a03e
fatal: not a git repository (or any of the parent directories): .git
-- Project is not inside a git repository,                         will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
```

Could you share with us all the commands you preformed to get to this point? it seems you are missing some git repository

Thanks,

dipen-1337lab, 2021-01-13:
Namaste @alfred2g,

Thank you for your reply.

> Could you share with us all the commands you preformed to get to this point? it seems you are missing some git repository

Please don't mind, but I have not understood as to what `all the commands` you are expecting. Kindly help me understand this better.

The `freertos-1609946200179` folder consists of the FreeRTOS Device Software downloaded from the AWS IoT Console. The configuration details are as below.

- Configuration - OTA Update
- Hardware platform - ESP32-DevKitC
- FreeRTOS Version - 202002.00

Could the `missing some git repository` really be an issue considering that the below `cmake` command without the `DAFR_ESP_LWIP=1` option executes properly as can be understood from the below log. Here also, we can view the `fatal: not a git repository (or any of the parent directories): .git` and other related logs.

```
E:\FreeRTOS_Device_Software_Configurations\freertos-1609946200179\FreeRTOS>cmake -DVENDOR=espressif -DBOARD=esp32_devkitc -DCOMPILER=xtensa-esp32 -GNinja -S . -B build
-- Checking Python dependencies...
Python requirements from E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf\requirements.txt are satisfied.
-- Building for target esp32
-- Component names: soc log heap freertos vfs newlib esp_ringbuf driver tcpip_adapter esp_event partition_table app_update spi_flash mbedtls micro-ecc bootloader_support efuse xtensa-debug-module app_trace ethernet nvs_flash pthread smartconfig_ack wpa_supplicant espcoredump esp32 cxx amazon-freertos-common secure_sockets freertos_plus_tcp bootloader nimble bt console esp_adc_cal esptool_py expat wear_levelling sdmmc fatfs freemodbus lwip nghttp openssl spiffs ulp
-- Component paths: E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/soc;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/log;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/heap;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/boards/esp32/components/freertos;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/vfs;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/newlib;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/esp_ringbuf;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/driver;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/boards/esp32/components/freertos_tcpip/tcpip_adapter;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/esp_event;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/partition_table;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/app_update;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/spi_flash;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/boards/esp32/components/mbedtls;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/micro-ecc;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/bootloader_support;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/efuse;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/xtensa-debug-module;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/app_trace;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/boards/esp32/components/freertos_tcpip/ethernet;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/nvs_flash;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/pthread;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/boards/esp32/components/freertos_tcpip/smartconfig_ack;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/wpa_supplicant;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/espcoredump;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/esp32;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/esp-idf/components/cxx;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code/amazon-freertos-common;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/vendors/espressif/boards/esp32/components/secure_sockets;E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/ [... cut ...]
=========================================================================

-- Configuring done
-- Generating done
-- Build files have been written to: E:/FreeRTOS_Device_Software_Configurations/freertos-1609946200179/FreeRTOS/build

E:\FreeRTOS_Device_Software_Configurations\freertos-1609946200179\FreeRTOS>
```

### Root Cause Analysis

From the error log, it is understood that the expected `secure_sockets` folder in the path `FreeRTOS\vendors\espressif\boards\esp32\ports` is missing. Hence the error!

Thanks | Regards,
Dipen

alfred2g, 2021-01-19:
Hi @dipen-1337lab 
```
Please don't mind, but I have not understood as to what all the commands you are expecting. Kindly help me understand this better.
```
Ah never mind, I thought you did a git clone.

Do you have secure sockets in the path?
```
FreeRTOS/vendors/espressif/boards/components/secure_sockets/
```
or in 
```
/libraries/abstractions/secure_sockets
```

dipen-1337lab, 2021-01-20:
Namaste @alfred2g,

 

> FreeRTOS/vendors/espressif/boards/components/secure_sockets/

The above path does not exist in the FreeRTOS folder structure.

The `secure_sockets` folder does exist at the below paths of the FreeRTOS directory.

- `FreeRTOS\libraries\abstractions\` and,
- `FreeRTOS\vendors\espressif\boards\esp32\components`

Thanks | Regards,
Dipen

dachalco, 2021-01-27:
Hi @dipen-1337lab 

How are you downloading the code?
Are you perhaps using the Online Configuration Wizard OCW?

dipen-1337lab, 2021-01-27:
Namaste @dachalco, 

> To reproduce

Yes, I did use the OCW to create a new configuration and, download the code with a specific demo selected. Please refer the **To reproduce** section, which is part of the very first comment when opening this issue.

Thanks | Regards,
Dipen

dachalco, 2021-01-27:
Got it, thank you. Brought this up to the team and there should be fix soon.

aggarw13, 2021-02-01:
Namaste @dipen-1337lab, I am able to reproduce the issue after downloading the 202002.00 version code from FreeRTOS console.
The cause of the issue is that the code downloaded from FreeRTOS console only supports the default enabled TCP/IP stack for the Espressif board, and the 202002.00 version used FreeRTOS+TCP stack by default. Thus, it is missing the `/vendors/espressif/boards/esp32/ports/secure_sockets/lwip/iot_secure_sockets.c` file in the download package. 

We had switched to using LWIP as the default stack for Espressif board only from 202007.00 version release onwards.

If you would like to use the LWIP stack for the Espressif board on the 202002.00 version, the only option would be to clone the [202002.00](https://github.com/aws/amazon-freertos/tree/202002.00) version code directly from GitHub. We are sorry about the inconvenience, and hope that helps.

closed 2021-02-02 by dipen-1337lab