aws/amazon-freertos #2929: ESP32 & AFR: OTA Update failure (IngestResult_t error -8)

dipen-1337lab, 2021-01-08:
Namaste Community Member,

My application runs the following threads created using `Iot_CreateDetachedThread()` API.

1. pvEsp32GetTimeOverSntp()
2. pvOtaUpdateTask()
3. pvMqttTask()
4. pvEsp32AndEnergyMeterCommunication()

and few other threads. All the threads are created using the default `IOT_THREAD_DEFAULT_PRIORITY` and `IOT_THREAD_DEFAULT_STACK_SIZE` parameters.

When an OTA Update is initiated via Jobs in AWS IoT, the firmware update initiates successfully. However, the update fails with multiple errors seen on the console log. The complete log file is shared here.
[otaUpdateFailure.log](https://github.com/aws/amazon-freertos/files/5789240/otaUpdateFailure.log)

Below are the couple of console log messages, which give some information on the failure.

```
500 20453 [OTA Agent Task] [prvProcessDataMessage] Aborting due to IngestResult_t error -8
I (204595) esp_ota_ops: [1] aflags/seq:0xffffffff/0x0, pflags/seq:0x2/0x1
501 20463 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] Msg: {"status":"FAILED","statusDetails":{"reason":"0x27000000: 0xfffffff8"}}

504 20492 [NetRecv] [ERROR][MQTT][204920] (MQTT connection 0x3fff0688) Failed to allocate buffer of length 4217 for incoming packet type 48.
505 20513 [iot_thread] Error: No OTA data buffers available.
```

The `IngestResult_t` value is identified as below.
`eIngest_Result_BadData = -8,            /* The data block from the server was malformed. */`

**Questions**
1) Is the OTA update failing because of memory unavailability? Below is the most recent log captured for the available Free Heap size, which is followed by the logs associated with OTA failure. 
`I (202635) dual_outlet: INFO : Free Heap size = 21208.`

2) If yes, what is the minimum amount of Heap memory required for OTA update to complete successfully on the ESP32DevKitC board?

3) Can the execution of multiple threads cause the OTA update to fail?

4) Is it possible to increase the `IOT_THREAD_PRIORITY` of the OTA thread at run time once the request for an OTA Update is identified by the firmware? More importantly, is this recommended or is the priority change required?

Thanks | Regards,
Dipen

pvyawaha, 2021-01-11:
Hello,

Thank you for reporting this issue. Can you please share which version of the library you are using and any changes in the OTA config? 

It seems the memory allocation is failing as sufficient heap is not available for the incoming packet resulting in ingest error in OTA library ( of 4K block size ) and then while allocating buffers for incoming packet ( ~ 4 KB ).  The heap used by OTA library depends on the OTA configuration and can be reduced by using a small block size. Can you please try using 1 KB block size and check if heap is always applicable for application task as well as OTA task? 

Regarding the thread priority , the OTA Agent thread should be of higher priority otherwise suspended for queue messages. You can configure the priority based on other application tasks created so that it does not starve using the OTA configuration.

dipen-1337lab, 2021-01-12:
Namaste @pvyawaha,

Thank you for your reply.

> Can you please share which version of the library you are using and any changes in the OTA config?

I've updated the version information in my original post above. My application is developed using **FreeRTOS V202002.00** and I'm using the OTA demo code as is with all the default values of the OTA configuration.

More details on the implementation and the source code is available [here](https://forums.freertos.org/t/running-mqtt-demo-periodically-with-ota-update-demo-causes-firmware-to-restart/10611/10?u=dipen12). Request you to review the files and, the conversation in this thread for any relevant information relating to this current issue.

I shall attempt your below two suggestions.

- using 1KB block size
- configuring OTA thread with a higher priority.

Thanks | Regards,
Dipen

dipen-1337lab, 2021-01-12:
Namaste @pvyawaha,

> Can you please try using 1 KB block size and check if heap is always applicable for application task as well as OTA task?

As suggested, I modified [this](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/aws_ota_agent_config.h#L44) `#define` to 10UL for using 1KB block size for OTA update.

Thank you so much for suggesting this, as it does resolve the `IngestResult_t error -8` error and, the entire firmware now seems to be getting downloaded onto the device. Further however, I encounter the `IngestResult_t error -2` error. The corresponding console log error is `E (285772) ota_pal: signature verification failed`.

Kindly suggest what I may be missing, which is generating this error.

```
17462 28364 [OTA Agent Task] [prvIngestDataBlock] Received file block 1120, size 1024
17463 28364 [OTA Agent Task] [prvIngestDataBlock] Remaining: 5
17464 28364 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
17465 28368 [NetRecv] [DEBUG][MQTT][283680] (MQTT connection 0x3ffef2c4) PUBLISH in data stream.
17466 28368 [NetRecv] Retain bit is 0.
17467 28369 [NetRecv] QoS is 0.
17468 28369 [NetRecv] DUP is 0.
17469 28369 [NetRecv] Topic name length 94: $aws/things/esp32DevKitCBoard01/streams/AFR_OTA-6d5c5b46-29a6-4062-9a73-c6b0b86014ae/data/cbor
17470 28370 [NetRecv] Payload length 1050.
17471 28371 [NetRecv] [DEBUG][MQTT][283710] (MQTT connection 0x3ffef2c4) Reference count changed from 2 to 3.
17472 28372 [iot_thread] [DEBUG][MQTT][283720] (MQTT connection 0x3ffef2c4) Reference count changed from 3 to 2.
17473 28372 [OTA Agent Task] [prvIngestDataBlock] Received file block 1121, size 1024
17474 28372 [OTA Agent Task] [prvIngestDataBlock] Remaining: 4
17475 28372 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
17476 28376 [NetRecv] [DEBUG][MQTT][283760] (MQTT connection 0x3ffef2c4) PUBLISH in data stream.
17477 28376 [NetRecv] Retain bit is 0.
17478 28377 [NetRecv] QoS is 0.
17479 28377 [NetRecv] DUP is 0.
17480 28377 [NetRecv] Topic name length 94: $aws/things/esp32DevKitCBoard01/streams/AFR_OTA-6d5c5b46-29a6-4062-9a73-c6b0b86014ae/data/cbor
17481 28381 [NetRecv] Payload length 1050.
17482 28381 [NetRecv] [DEBUG][MQTT][283810] (MQTT connection 0x3ffef2c4) Reference count changed from 2 to 3.
17483 28382 [OTA Agent Task] [prvIngestDataBlock] Received file block 1122, size 1024
17484 28382 [OTA Agent Task] [prvIngestDataBlock] Remaining: 3
17485 28382 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
17486 28382 [iot_thread] [DEBUG][MQTT][283820] (MQTT connection 0x3ffef2c4) Reference count changed from 3 to 2.
17487 28386 [NetRecv] [DEBUG][MQTT][283850] (MQTT connection 0x3ffef2c4) PUBLISH in data stream.
17488 28386 [NetRecv] Retain bit is 0.
17489 28386 [NetRecv] QoS is 0.
17490 28387 [NetRecv] DUP is 0.
17491 28387 [NetRecv] Topic name length 94: $aws/things/esp32DevKitCBoard01/streams/AFR_OTA-6d5c5b46-29a6-4062-9a73-c6b0b86014ae/data/cbor
17492 28388 [NetRecv] Payload length 1050.
17493 28389 [NetRecv] [DEBUG][MQTT][283890] (MQTT connection 0x3ffef2c4) Reference count changed from 2 to 3.
17494 28390 [iot_thread] [DEBUG][MQTT][283900] (MQTT connection 0x3ffef2c4) Reference count changed from 3 to 2.
17495 28390 [OTA Agent Task] [prvIngestDataBlock] Received file block 1123, size 1024
17496 28390 [OTA Agent Task] [prvIngestDataBlock] Remaining: 2
17497 28390 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
17498 28394 [NetRecv] [DEBUG][MQTT][283940] (MQTT connection 0x3ffef2c4) PUBLISH in data stream.
17499 28394 [NetRecv] Retain bit is 0.
17500 28394 [NetRecv] QoS is 0.
17501 28395 [NetRecv] DUP is 0.
17502 28395 [NetRecv] Topic name length 94: $aws/things/esp32DevKitCBoard01/streams/AFR_OTA-6d5c5b46-29a6-4062-9a73-c6b0b86014ae/data/cbor
17503 28396 [NetRecv] Payload length 1050.
17504 28397 [NetRecv] [DEBUG][MQTT][283970] (MQTT connection 0x3ffef2c4) Reference count changed from 2 to 3.
17505 28398 [iot_thread] [DEBUG][MQTT][283980] (MQTT connection 0x3ffef2c4) Reference count changed from 3 to 2.
17506 28398 [OTA Agent Task] [prvIngestDataBlock] Received file block 1124, size 1024
17507 28398 [OTA Agent Task] [prvIngestDataBlock] Remaining: 1
17508 28398 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock]
17509 28402 [NetRecv] [DEBUG][MQTT][284020] (MQTT connection 0x3ffef2c4) PUBLISH in data stream.
17510 28402 [NetRecv] Retain bit is 0.
17511 28403 [NetRecv] QoS is 0.
17512 28403 [NetRecv] DUP is 0.
17513 28403 [NetRecv] Topic name length 94: $aws/things/esp32DevKitCBoard01/streams/AFR_OTA-6d5c5b46-29a6-4062-9a73-c6b0b86014ae/data/cbor
17514 28404 [NetRecv] Payload length 682.
17515 28405 [NetRecv] [DEBUG][MQTT][284050] (MQTT connection 0x3ffef2c4) Reference count changed from 2 to 3.
I (284092) ota_pal: No such certificate file: /. Using aws_ota_codesigner_certificate.h.

17516 28406 [iot_thread] [DEBUG][MQTT][284060] (MQTT connection 0x3ffef2c4) Reference count changed from 3 to 2.
[... cut ...]
17617 30474 [iot_thread] [INFO ][DEMO][304740] State: WaitingForJob  Received: 1129   Queued: 0   Processed: 0   Dropped: 0

17618 30574 [iot_thread] [INFO ][DEMO][305740] State: WaitingForJob  Received: 1129   Queued: 0   Processed: 0   Dropped: 0

17619 30674 [iot_thread] [INFO ][DEMO][306740] State: WaitingForJob  Received: 1129   Queued: 0   Processed: 0   Dropped: 0

17620 30774 [iot_thread] [INFO ][DEMO][307740] State: WaitingForJob  Received: 1129   Queued: 0   Processed: 0   Dropped: 0
```

Thanks | Regards,
Dipen

pvyawaha, 2021-01-14:
Hello,

Thank you for sharing update. For the signature verification error you are getting , can you check if you have correct code signer certificate in aws_ota_codesigner_certificate.h as per the code signing profile used when creating the OTA job.

Please refer here for more details regarding the steps - https://docs.aws.amazon.com/freertos/latest/userguide/ota-code-sign-cert-esp.html

dipen-1337lab, 2021-01-17:
Namaste @pvyawaha,

The `E (285772) ota_pal: signature verification failed` error was indeed because of the incorrect code signer certificate within the `aws_ota_codesigner_certificate.h` file, which didn't match the code signing profile used when creating the OTA job.

I confirm that the OTA update was successful after using the appropriate code signing certificate. Below is the console log, which confirms the successful OTA update.
`54 910 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] Msg: {"status":"SUCCEEDED","statusDetails":{"reason":"accepted v0.9.3"}}
`

Thank you for your assistance on both the `IngestResult_t` errors encountered by me!

Thanks | Regards,
Dipen

closed 2021-01-17 by dipen-1337lab