aws/amazon-freertos #2931: Compile and Falsh issues with ESP-IDF 4.2

aggarg, 2021-01-11:
Hi, 

I am currently trying to work with this commit and I'm running into some issues... 

I'm using an ESP32-D0WD-V3 chip on a custom board connected to a ESP-Prog board for programming. I'm developing on a Windows 10 PC and I've installed [esp-idf tools v2.3.](https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/get-started/windows-setup.html#esp-idf-tools-installer)

In my project, I am using amazon-freertos as a submodule and when I tried to build with `cmake  -DVENDOR=espressif -DBOARD=esp32_plus_ecc608a_devkitc -DCOMPILER=xtensa-esp32 -DCMAKE_BUILD_TYPE=Debug -DCMAKE_TOOLCHAIN_FILE=./amazon-freertos/tools/cmake/toolchains/xtensa-esp32.cmake -GNinja -S . -B build` 

I was getting the error : 
```
CMake Error at amazon-freertos/vendors/espressif/boards/esp32/CMakeLists.txt:416 (add_executable):
  add_executable cannot create target "my-app" because another target
  with the same name already exists.  The existing target is an executable
  created in source directory "C:/<proj_directory>".  See documentation
  for policy CMP0002 for more details.
Call Stack (most recent call first):
  amazon-freertos/CMakeLists.txt:70 (include)
```

It appears the changes in vendors\espressif\boards\esp32\CMakeLists.txt broke the build. By reverting this file to commit cc5530d5f79f7f515f770a91f2eff62cf41a785c I  was able to continue on in the build, but I ran into a different error.

**Question 1:**
After running `.\amazon-freertos\vendors\espressif\esp-idf\tools\idf.py -p COM4 -b 921600 erase_flash flash monitor` there is a error in the esptool script output below. Any ideas on how to fix this building and flash issue? Let me know if you need any more information. 
```
Executing action: flash
Running ninja in directory <proj_directory>\build
Executing "ninja flash"...
[0/2] Re-checking globbed directories...
[4/9] Generating binary image from built executable
FAILED: .bin_timestamp
cmd.exe /C "cd /D <proj_directory>\build && python <proj_directory>/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 4MB --elf-sha256-offset 0xb0 -o <proj_directory>/build/.bin / && "C:\Program Files\CMake\bin\cmake.exe" -E echo "Generated <proj_directory>/build/.bin" && "C:\Program Files\CMake\bin\cmake.exe" -E md5sum <proj_directory>/build/.bin > <proj_directory>/build/.bin_timestamp"
esptool.py v3.0
Traceback (most recent call last):
  File "<proj_directory>/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py", line 3969, in <module>
    _main()
  File "<proj_directory>/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py", line 3962, in _main
    main()
  File "<proj_directory>/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py", line 3631, in main
    operation_func(args)
  File "<proj_directory>/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py", line 3079, in elf2image
    e = ELFFile(args.input)
  File "<proj_directory>/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py", line 2542, in __init__
    with open(self.name, 'rb') as f:
PermissionError: [Errno 13] Permission denied: '/'
```

**Question 2:**
Another issue, I've downloaded the latest xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64 compiler tools but the version is 8.4. Is there any way to get the proper xtensa-esp32-elf-gcc 8.2 version you reference?
```
xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
```

_Originally posted by @octopusbunch in https://github.com/aws/amazon-freertos/issues/2893#issuecomment-757097431_

mingyue86010, 2021-01-11:
Hi @octopusbunch,

Can you firstly check the following things which are some common problems that people runs into an error when using IDF:

1. Did you clean/remove the previous build artifacts before starting a new build?
2. Did you have IDF_PATH / IDF_TOOLS_PATH environment variable set? If so, try unset thoes variable. Otherwise, the build tool may use the esp-idf tools v2.3 instread.
3. Did your project path contains any "space"? That cause the IDF a build problem on Windows.

For the xtensa-esp32-elf-gcc version, I think the 8.4.0 should be fine. That is what I'm using and also ESP current IDF v4.2 instruction suggest: https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-guides/tools/idf-tools.html

If you still have the problem, please let me know. I can further take a look into it.

Thank you!
Ming

octopusbunch, 2021-01-11:
Hi Ming,

Thanks for getting back to me and for the suggestions. 
1. I deleted the build artifacts 
2. I unset IDF_TOOLS and IDF_TOOLS_PATH... and this shows up in the build log: 
```
CMake Warning at amazon-freertos/vendors/espressif/esp-idf/tools/cmake/idf.cmake:24 (message):
  IDF_PATH environment variable not found.  Setting IDF_PATH to
  '<proj_directory>/amazon-freertos/vendors/espressif/esp-idf'.
Call Stack (most recent call first):
  amazon-freertos/vendors/espressif/boards/esp32/CMakeLists.txt:38 (include)
  amazon-freertos/CMakeLists.txt:70 (include)
```
3. My project directory doesn't have any spaces in it
 
The flash procedure still fails with the same error message as before below. Please let me know if there's any other information you need. 
```
-- Build files have been written to: <proj_directory>/build
Running ninja in directory <proj_directory>\build
Executing "ninja flash"...
[0/2] Re-checking globbed directories..
FAILED: .bin_timestamp
cmd.exe /C "cd /D <proj_directory>\build && "C:\Program Files\Python38\python.exe" <proj_directory>/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 40m --flash_size 4MB --elf-sha256-offset 0xb0 -o <proj_directory>/build/.bin / && "C:\Program Files\CMake\bin\cmake.exe" -E echo "Generated <proj_directory>/build/.bin" && "C:\Program Files\CMake\bin\cmake.exe" -E md5sum <proj_directory>/build/.bin > <proj_directory>/build/.bin_timestamp"
esptool.py v3.0
Traceback (most recent call last):
  File "<proj_directory>/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py", line 3969, in <module>
    _main()
  File "<proj_directory>/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py", line 3962, in _main
    main()
  File "<proj_directory>/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py", line 3631, in main
    operation_func(args)
  File "<proj_directory>/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py", line 3079, in elf2image
    e = ELFFile(args.input)
  File "<proj_directory>/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py", line 2542, in __init__
    with open(self.name, 'rb') as f:
PermissionError: [Errno 13] Permission denied: '/'
[3/12] Generating ../../partition_table/partition-table.bin
```

shubhamkulkarni97, 2021-01-12:
Hi @octopusbunch,

Are you trying to build aws_demos/aws_tests in amazon-freertos or using AFR in external project?

If you are using amazon-freertos in external project, please refer to [feature/idf_v4.2](https://github.com/shubhamkulkarni97/amazon-freertos-examples/tree/feature/idf_v4.2) branch in amazon-freertos-examples repo.

I tried building aws_demos on Windows machine, it works fine.
If you have issues building aws_demos/aws_tests in amazon-freertos, please try following steps:

* `cd vendors/espressif/esp-idf`
* `install.bat`
* `export.bat`
* `cd <amazon-freertos-root-dir>`
* Clear IDF_PATH environment variable
* `set AFR_TOOLCHAIN_PATH=<path-to-xtensa-toolchain>`
* `set ESPPORT=<ESP32-COM-port>`
* `cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B build -GNinja`
* `cmake --build build --target flash`

Thanks,
Shubham

mingyue86010, 2021-01-13:
Hi @octopusbunch,

The warning you saw is normal. After you unset the IDF_PATH, the cmake script will set it to the IDF location in the AFR repo.

```
CMake Warning at amazon-freertos/vendors/espressif/esp-idf/tools/cmake/idf.cmake:24 (message):
  IDF_PATH environment variable not found.  Setting IDF_PATH to
  '<proj_directory>/amazon-freertos/vendors/espressif/esp-idf'.
Call Stack (most recent call first):
  amazon-freertos/vendors/espressif/boards/esp32/CMakeLists.txt:38 (include)
  amazon-freertos/CMakeLists.txt:70 (include)
```

BTW, does @shubhamkulkarni97 's answer resolved your issue?

Thanks,
Ming

octopusbunch, 2021-01-13:
Hi mingyue86010,

Thank you for the clarification. 

@shubhamkulkarni97, I am using amazon-freertos as an external project so I am basing my project after the [feature/idf_v4.2](https://github.com/shubhamkulkarni97/amazon-freertos-examples/tree/feature/idf_v4.2) repo you posted and I got past the Permission denied error I was having before. Now I am just fixing some other errors but the changes in that repository helped me, Thank you!

mingyue86010, 2021-01-13:
Hi @octopusbunch,

Great! I'm glad to hear that it resolve your issues. So can I just close this issue for now? If you have any further question, please feel free to open a new issue to us.

Thank you!
Ming

octopusbunch, 2021-01-13:
OK that's fine. Thank you and I will open an issue if I need help again.

closed 2021-01-13 by mingyue86010