aws/amazon-freertos #2932: http post without private key and client certificate

surya-teja-p, 2021-01-13:
Hi,

I am trying to perform a http post request with only root CA and without private key and client certificate on ESP32 device, but when I perform request it is expecting a private key to establish TLS connection and the request is failing, Is there any way to perform http request with out private key and client certificate?

And when I am trying to add esp_http_client.h and other headers like esp_tls.h it is not being identified, I have already added tls and https modules to target_link_libraries  need your suggestion in this issues also.

Thanks 
Surya

lundinc2, 2021-01-13:
Hi @surya-teja-p,

It is possible with coreHTTP, and there is an example of a TLS connection without mutual auth in the C-SDK repository here: https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/main/demos/http/http_demo_basic_tls. Can you clarify what version of Amazon FreeRTOS you are using? The latest release will have demo examples using coreHTTP.

Are you having an issue using the code without client credentials, or is the server rejecting your request? The server will dictate if you need to provide client credentials, and mbedtls will ignore the credentials if they are not needed.

Thanks,

Carl

surya-teja-p, 2021-01-18:
Hi @lundinc2,
I am using **202007.00 July 2020** version it does not have coreHTTP library, instead it has https library under standard in c_sdk using the mentioned ones is it possible to make a https post request without private key

The server does not require any private key other then ROOT CA which I am sending, I have also verified it using POSTMAN.

Thanks 
surya

sarenameas, 2021-01-19:
Please try setting the certificate and private key fields to NULL.
https://github.com/aws/amazon-freertos/blob/202007.00/libraries/c_sdk/standard/https/include/types/iot_https_types.h#L751
https://github.com/aws/amazon-freertos/blob/202007.00/libraries/c_sdk/standard/https/include/types/iot_https_types.h#L754

surya-teja-p, 2021-01-27:
Hi @sarenameas,

Sorry for the delay,
below are the logs you asked. I have tried to perform post with my custom URL with postman and it worked fine with only ROOT-CA. In my code I have set private key and client cert to NULL as you mentioned, but have set ROOT-CA to a valid one.

0 739 [iot_thread] ERROR: Private key not found. 1 739 [iot_thread] TLS_Connect fail (0x7d4, URL)
2 739 [iot_thread] [ERROR][NET][7390] Failed to establish new connection. Socket status: -1.
3 740 [iot_thread] [ERROR][HTTPS Client][7400] Failed to connect to the server at URL on port 443 with error: 4
4 740 [iot_thread] [ERROR][HTTPS Client][7400] Error in IotHttpsClient_Connect(). Error code 203.
5 740 [iot_thread] [ERROR][HTTPS Client][7400] connHandle was NULL.
6 740 [iot_thread] [ERROR][HTTPS Client][7400] IotHttpsClient_SendSync() failed.

Thanks
Surya

surya-teja-p, 2021-01-29:
Hi @sarenameas 

Waiting for your suggestion here about how to proceed, I am working on esp-32, If it is not possible with amazon-freertos will I be able to perform the operation using esp-idf http calls, and when I am adding http header files of esp the compiler is not able to find the given headers, can you suggest any way to overcome this issue to perform HTTPS post request?

Thanks
Surya

surya-teja-p, 2021-02-02:
Hi @sarenameas

If there is no support for https call in server authentication mode, can I get info how to include esp_http_client.h file into my code, when I try to include the header file compiler is showing : "No file or directory exists" error, can you suggest any way to fix it.

Thanks
Surya

aggarw13, 2021-02-02:
Hi @surya-teja-p, we have provided a patch in [this post](https://github.com/aws/amazon-freertos/issues/2956#issuecomment-771278810) for supporting TLS connection in server authentication mode. The patch includes changes for the [`iot_tls.c`](https://github.com/aws/amazon-freertos/blob/202007.00/libraries/freertos_plus/standard/tls/src/iot_tls.c) which is the file containing logic for establishing TLS session with a server. After applying the patch in the file, you will be able to use the [HTTPS library](https://github.com/aws/amazon-freertos/tree/202007.00/libraries/c_sdk/standard/https) to establish server authenticated connection with a server.

surya-teja-p, 2021-02-04:
Hi @aggarw13 

Thanks for your help, it is working.

Thanks
Surya

closed 2021-02-04 by surya-teja-p