aws/amazon-freertos #2933: when I am trying to send the shadow update to the cloud using the following funciton, 40% of the time I get the error "Failed to send Shadow update error MQTT LIBRARY ERROR".

kumarfirmware, 2021-01-13:
Hi Team,

I am using AWS FreeRTOS version 202007 and ESP32.

when I am trying to send the shadow update to the cloud using the following funciton, 40% of the time I get the error "Failed to send Shadow update error MQTT LIBRARY ERROR".

Some times I get the error, "Failed to set shadow callback, error MQTT LIBRARY ERROR".

Some times I get the error, Wait complete with result NETWORK ERROR, Wait complete with result TIMEOUT.

Can you please let me know what could be the reason for these errors 40% of the time and what could be the reason it works for 60% of the time?

What is the significance of these errors?

When will these errors occur? What will be the root cause of these errors?

_sendShadowUpdates(IotMqttConnection_t mqttConnection,
                               const char * const pThingName,
                               size_t thingNameLength );

static int _setShadowCallbacks(IotMqttConnection_t mqttConnection,
                                const char * pThingName,
                                size_t thingNameLength );

updateStatus = AwsIotShadow_TimedUpdate( mqttConnection,
                                             &updateDocument,
                                             AWS_IOT_SHADOW_FLAG_KEEP_SUBSCRIPTIONS,
                                             5000 );

I have a stable wifi connection and when these erros occur, I check the ping of the iP address and everything is good.

To over come this when ever shadow fails, I try to disconnect MQTT connection and connect again and it is same story, it works some times and it does not work sometimes.

Thanks.

Please find the log as follows.

400 26017 [OTA_Update_task] [INFO ][MQTT][260170] (MQTT connection 0x3ffde03c) SUBSCRIBE operation scheduled.
401 26017 [OTA_Update_task] [INFO ][MQTT][260170] (MQTT connection 0x3ffde03c, SUBSCRIBE operation 0x3ffddf78) Waiting for operation completion.
402 26070 [OTA_Update_task] [INFO ][MQTT][260700] (MQTT connection 0x3ffde03c, SUBSCRIBE operation 0x3ffddf78) Wait complete with result SUCCESS.
403 26075 [OTA_Update_task] [INFO ][MQTT][260750] (MQTT connection 0x3ffde03c) SUBSCRIBE operation scheduled.
404 26075 [OTA_Update_task] [INFO ][MQTT][260750] (MQTT connection 0x3ffde03c, SUBSCRIBE operation 0x3ffddf78) Waiting for operation completion.
405 26103 [OTA_Update_task] [INFO ][MQTT][261030] (MQTT connection 0x3ffde03c, SUBSCRIBE operation 0x3ffddf78) Wait complete with result SUCCESS.
406 26103 [OTA_Update_task] [INFO ][MQTT][261030] (MQTT connection 0x3ffde03c) SUBSCRIBE operation scheduled.
407 26103 [OTA_Update_task] [INFO ][MQTT][261030] (MQTT connection 0x3ffde03c, SUBSCRIBE operation 0x3ffddf78) Waiting for operation completion.
408 26603 [OTA_Update_task] [INFO ][MQTT][266030] (MQTT connection 0x3ffde03c, SUBSCRIBE operation 0x3ffddf78) Wait complete with result TIMEOUT.
409 26603 [OTA_Update_task] [INFO ][MQTT][266030] (MQTT connection 0x3ffde03c) UNSUBSCRIBE operation scheduled.
Sendshawdowupdate function not succesful, reconnect again
410 26603 [OTA_Update_task] [INFO ][MQTT][266030] (MQTT connection 0x3ffde03c, UNSUBSCRIBE operation 0x3ffddf78) Waiting for operation completion.
411 26604 [iot_thread] [ERROR][NET][266040] Error -27648 while sending data.
412 26604 [OTA_Update_task] [INFO ][MQTT][266040] (MQTT connection 0x3ffde03c, UNSUBSCRIBE operation 0x3ffddf78) Wait complete with result NETWORK ERROR.
413 26604 [OTA_Update_task] [ERROR][EOS][266040] Failed to send Shadow update error MQTT LIBRARY ERROR
414 26704 [OTA_Update_task] MQTT connection got disconnected
415 26704 [OTA_Update_task] [OTA_AgentShutdown] Nothing to do: Already in state [Stopped]
416 26704 [OTA_Update_task] [INFO ][MQTT][267040] (MQTT connection 0x3ffde03c) Disconnecting connection.
417 26704 [OTA_Update_task] [INFO ][MQTT][267040] (MQTT connection 0x3ffde03c, DISCONNECT operation 0x3ffbd97c) Waiting for operation completion.
418 26705 [iot_thread] [ERROR][NET][267050] Error -27648 while sending data.
419 26705 [OTA_Update_task] [INFO ][MQTT][267050] (MQTT connection 0x3ffde03c, DISCONNECT operation 0x3ffbd97c) Wait complete with result NETWORK ERROR.
420 26705 [OTA_Update_task] [WARN ][MQTT][267050] (MQTT connection 0x3ffde03c) DISCONNECT not sent, error NETWORK ERROR.
421 26705 [OTA_Update_task] prvNetworkDisconnectCallback function called
422 26705 [OTA_Update_task] [INFO ][MQTT][267050] (MQTT connection 0x3ffde03c) Network connection closed.

leegeth, 2021-01-15:
@kumarfirmware

Thanks for bringing this issue to our attention. 
By looking at the logs you shared, it seems like a timeout happened while trying to subscribe to a Shadow MQTT topic and waiting for getting an acknowledgement back from the network. There are network errors immediately following it
`411 26604 [iot_thread] [ERROR][NET][266040] Error -27648 while sending data.`

The error code appears to be from TLS(Transport Layer Security) layer and it could be due to some network issues. How stable is the network you are using? Will you be able to try another demo(MQTT demo) on the same network and see if you hit a similar issue? If the issue persists, will it be possible for you to do a network packet capture and share it with us to better understand if there are any network related issues you are running into?

About the Shadow limits, AWS documentation [here](https://docs.aws.amazon.com/general/latest/gr/iot-core.html#limits_iot) mentions about the limit of Shadow document to be 8KB. 1300 bytes are well within that limit. So the size may not be an issue.

kumarfirmware, 2021-01-15:
Hi Leegeth,

Thanks for your immediate response. This is a production stopping issue, need your urgent attention and support.

"There are network errors immediately following it
411 26604 [iot_thread] [ERROR][NET][266040] Error -27648 while sending data."

As I mentioned above since shadow failed to publish, I disconnected MQTT connection and reconnected again. That is the reason you are seeing this message.

OTA seems to work properly multiple times with out any issue. Why will shadow has problems 40% of the time?

can we capture this encrypted secure communication between ESP and Cloud in Wireshark? please clarify?

Please find my setup as follows.

Setup:-

Device is connected to the corporate network Wi-Fi. Very High speed network. No chance of unstable network.
Also ESP does not loose ping at all. Ping is very stable. Device is physically present in middle east and configured to US-East region and cloud is configured to US-East region.

Please let me know the next steps how to debug and how to proceed as soon as possible. This is stopping the production.

Thanks
Kumar.

kumarfirmware, 2021-01-19:
Hi Leegeth,

Hope you had an excellent weekend.

Can you please help me in this as soon as possible.

Thanks,
Kumar.

leegeth, 2021-01-19:
@kumarfirmware,

Thanks for your patience and apologies for the delay in response.  Thanks for adding more details to the issue you are facing.
In order to understand more about the added logic for handling a failure, I have got a few questions.
1. On what event do you attempt to reconnect to MQTT broker? Is it while detecting the timeout while waiting for the SUBSCRIBE operation completion? Or is it if you face any error?
2. Can you also help with a code snippet or the sequence of APIs that you use for attempting to reconnect?
3. After attempting to reconnect,  I see from the description of the issue that it is not always a success. Are there logs from a success case that you can share?
4. What is the MQTT keep alive timeout value you are using for Shadow application?

For capturing the network packets using Wireshark from ESP32, you can connect ESP32 to a Wifi hotspot from a laptop and run Wireshark on the laptop.  Wireshark provides support to decrypt the TLS messages as described [here](https://wiki.wireshark.org/TLS#TLS_Decryption). I haven't tried this configuration for decryption yet on Wireshark. I will also give it a try and update you. For a start, if you can try to capture the packets without decryption to see if there are any TCP level issues happening when you are facing this issue.  Can you please also share the logs with us so that we can also take a look? Please also note the timestamp at which the issue is happening.

The error MQTT LIBRARY ERROR means that a Shadow operation failed because of failure in MQTT library. Just for an example, please refer to the logs pasted below.

```
411 26604 [iot_thread] [ERROR][NET][266040] Error -27648 while sending data.
412 26604 [OTA_Update_task] [INFO ][MQTT][266040] (MQTT connection 0x3ffde03c, UNSUBSCRIBE operation 0x3ffddf78) Wait complete with result NETWORK ERROR.
413 26604 [OTA_Update_task] [ERROR][EOS][266040] Failed to send Shadow update error MQTT LIBRARY ERROR
```
There is a network error and MQTT library fails to send an UNSUBSCRIBE packet. This error is propagated to the Shadow library as MQTT LIBRARY ERROR. Hope that clarifies your question.

In addition, can you please explain to me, what you meant by not loosing ping? Did you mean the MQTT ping requests to the broker? How did you verify that the pings are not loosing?

kumarfirmware, 2021-01-19:
Hi Leegeth,

Thanks for your immediate response. 

1. Added logic for failure is a blind logic, when ever shadow returns me failure to send shadow, I disconnect MQTT connection, 
           I call     AwsIotShadow_Cleanup(); and I re-initialise AwsIotShadow_Init(0); and try to do MQTT connection and send shadow again.

2. MQTT keep alive time out is 30 seconds.

3. I will connect ESP32 to my laptop and capture Wireshark log and send it to you tomorrow morning.

4. "How did you verify that the pings are not loosing?" this ping is not MQTT ping. This ping is pinging IP address of ESP32 from command prompt and checking whether we get response or not.

I will provide logs to you tomorrow as soon as possible. Hope you crack the issue from logs.

Thanks,
Kumar.

leegeth, 2021-01-19:
@kumarfirmware,

Thanks

kumarfirmware, 2021-01-25:
Hi Leegeth,

Hope you had an excellent weekend.

I could capture the Wireshark logs. But the Wireshark logs contain TCP packets and TLS 1.2 packets.

How should we decode TLS 1.2 packets in Wireshark. Did you try to decode TLS 1.2 packets from your end in the Wireshark?

Can you please let me know this as soon as possible?

Thanks,
Kumar.

leegeth, 2021-01-27:
@kumarfirmware,

Glad to hear that you could capture TCP packets using Wireshark. Can you please share the file with current capture and let us know at what time you had seen the issue with Shadow failures? I tried to decrypt by following the Wireshark documentation and I couldn't successfully decrypt the packets. I will try it again using different options mentioned in the documentation and let you know. However, If you could reproduce the issue with TCP packets captured, I can start taking a look at that and see if there is any issues that I can observe from that. Will you be able to share the packet capture with us so that we can take a look?

Thanks

leegeth, 2021-01-28:
@kumarfirmware,

Posting an update related to decrypting the TLS messages in Wireshark. I have tried a few approaches mentioned in the Wireshark documentation, but couldn't successfully decrypt messages. Both the session key log file approach and the RSA private key file approach didn't work for decrypting messages. I believe the reasons are
1. If the laptop is a hotspot and packet capture is done from it, it is acting as man in the middle. This is preventing us from creating the session key log file.
2. RSA keys are not supported with ECDHE cipher suite(documented in the [Wireshark wiki](https://wiki.wireshark.org/TLS#TLS_Decryption)) and that is the cipher suite picked in my tests.

While keep looking at that configuration, I believe it will be better for us to take a look at what is already available. If you have the packet capture with issue reproduced, sharing that and the corresponding device logs will help debug the issue. Can you please share that with us, once available?

kumarfirmware, 2021-02-22:
Hi Leegeth,

Hope you had an excellent weekend.

With out decrypting the messages in Wireshark, how can we solve this issue. We need to find a way to decrypt packets in Wireshark, otherwise we will never be able to solve this problem. This problem is very critical.

Are there any other ways of solving this issue? kindly 

Can you please let me know this as soon as possible?

Thanks,
Kumar.

leegeth, 2021-02-26:
@kumarfirmware,

By using the TCP dump, we can see if the server terminates the connection or not. Can you please attach Wireshark logs if you have so that we can take a look?
In addition, let us also try to analyze the issue with some other possible methods.
Can you please attach full device logs after increasing the log verbosity to IOT_LOG_DEBUG [here](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h#L46-L52) when this issue is happening?
There is one more detail about the task priorities that can help as well.  Can you please let us know the task priorities that you are using for the below tasks.
1. Shadow application (For example the priority can be seen [here ](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/aws_demo_config.h#L55)for the demo in this repo).
2. Task pool tasks (For example the priority can be seen [here ](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h#L58)for the demo in this repo).
3.  Network receive task (For example the priority can be seen [here ](https://github.com/aws/amazon-freertos/blob/master/demos/include/iot_config_common.h#L159)for the demo in this repo).

yanjos-dev, 2021-03-15:
I am closing this issue due to inactivity. Please reopen the issue if you would like any additional help.